wmbusmeters/wmbusmeters #434: MBUS (PN-EN 13757)  suported?

djmalin, 2022-01-02:
Hello @weetmuts , this is a heat allocator Wireless MBUS (PN-EN 13757) B Meters
is it possible to read data from it?

log telegram:
(meter) MainWater3: meter detection could not find driver for id: 74393723 mfct: (BMP) BMETERS Polska, Poland (0x9b0) type: Heat Cost Allocator (0x08) ver: 0x33
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

2D44B00923373974330878|0F33130002AAB387B10FC8FC035ECA03047DDCD503F6DBAEDC420C420C420CE30B4204|+654
2D44B00923373974330878|0F34130002D743C2359A56ED8A4B5E468B6C54F48A13533B4A078307830783A682078B|+896
2D44B00923373974330878|0F36130002A1C1D2B0CAC73D041BC9F603BCDBE40303DCEBDCB70BB70BB70BD60BB703|+1379
2D44B00923373974330878|0F3713000254D9CBC9D3BA58A742B21FA7D1AE89A75AB032B65EA05EA05EA0BF9E5EA8|+1620
2D44B00923373974330878|0F381300022C94DCA3840D67D2010C90D2F2F86AD259F991F94DC94DC94DC930C94DD1|+1861
2D44B00923373974330878|0F39130002C94594360C57FF8A6957B88A7A53A28AF152B949F582F582F5821883F58A|+2102
2D44B00923373974330878|0F3A130002E4F751E94990CECB6C92A5CB4B9413CCF4939C94E8C3E8C3E8C345C4E8CB|+2343
2D44B00923373974330878|0F3B13000282ABBB91A3F248DF32EAEFDEC1F619DFAAF642FEAEE6AEE6AEE6CFE6AEDE|+2584
2D44B00923373974330878|0F3C130002D5D167EDAFC4DC887DC4E3885DC2F588D6C00EC0A290A290A2900391A288|+2825
2D44B00923373974330878|0F3D130002F9D1E3EB7BC5908AB9C3678A19C1C18832C15AC126912691269187912689|+3065
2D44B00923373974330878|0F3E1300023F6B5D5EB533F217873519186731EF17D02F1831CC1FCC1FCC1FB91FCC17|+3306
2D44B00923373974330878|0F3F130002126FCC77D420573D4220203DA2247A3D692621245D365D365D36C0345D3E|+3546
2D44B00923373974330878|0F401300023AA895C0FDED0AF22FE7E1F18FE347F2A8D9E0E2B4E9B4E9B4E901EAB4F1|+3786
2D44B00923373974330878|0F41130002D8B2A3BEBBED50F2F8E627F2D9D981F272E31AE366F366F366F3C7E966FB|+4027
2D44B00923373974330878|0F42130002294FA6626E4B291E414B221EA045381E973F4F3F630F630F630FC2156317|+4267
2D44B00923373974330878|0F4313000284CC87B10FC8FC0374CA03047DDCD503F6DBAEDC420C420C420CE30B4204|+4506
2D44B00923373974330878|0F441300025443C2359A56ED8A555E468B6C54F48A13533B4A078307830783A682078B|+4746
2D44B00923373974330878|0F48130002B996DCA3840D67D2FF0B90D2F2F86AD259F991F94DC94DC94DC930C94DD1|+232
2D44B00923373974330878|0F49130002494E94360C57FF8A6757B88A7A53A28AF152B949F582F582F5821883F58A|+471
2D44B00923373974330878|0F4A130002640251E94990CECB7592A5CB4B9413CCF4939C94E8C3E8C3E8C345C4E8CB|+710
2D44B00923373974330878|0F4B13000203A3BB91A3F248DF1FEAEFDEC1F619DFAAF642FEAEE6AEE6AEE6CFE6AEDE|+949
2D44B00923373974330878|0F4C13000258D567EDAFC4DC8893C4E3885DC2F588D6C00EC0A290A290A2900391A288|+1188
2D44B00923373974330878|0F4D1300026CD6E3EB7BC5908AB7C3678A19C1C18832C15AC126912691269187912689|+1427
2D44B00923373974330878|0F4E1300026B645D5EB533F217893519186731EF17D02F1831CC1FCC1FCC1FB91FCC17|+1666
2D44B00923373974330878|0F4F130002857ECC77D420573DF01F203DA2247A3D692621245D365D365D36C0345D3E|+1904
2D44B00923373974330878|0F5013000264B195C0FDED0AF231E7E1F18FE347F2A8D9E0E2B4E9B4E9B4E901EAB4F1|+2142
2D44B00923373974330878|0F511300024FB0A3BEBBED50F2F7E627F2D9D981F272E31AE366F366F366F3C7E966FB|+2381
2D44B00923373974330878|0F69130002D51B94360C57FF8A6657B88A7A53A28AF152B949F582F582F5821883F58A|+452
2D44B00923373974330878|0F6C130002910567EDAFC4DC8881C4E3885DC2F588D6C00EC0A290A290A2900391A288|+105
2D44B00923373974330878|0F741300020A11C2359A56ED8A585E468B6C54F48A13533B4A078307830783A682078B|+624
2D44B00923373974330878|0F751300028BDEBEE8D68F41CC1C91EACBC89310CCAF93F794ABC3ABC3ABC3CAC3ABCB|+858
2D44B00923373974330878|0F76130002E494D2B0CAC73D0428C9F603BCDBE40303DCEBDCB70BB70BB70BD60BB703|+1092
2D44B00923373974330878|0F77130002A4F6CBC9D3BA58A7EDB11FA7D1AE89A75AB032B65EA05EA05EA0BF9E5EA8|+1326
2D44B00923373974330878|0F7813000290C2DCA3840D67D2FE0B90D2F2F86AD259F991F94DC94DC94DC930C94DD1|+1560
2D44B00923373974330878|0F79130002842494360C57FF8A6657B88A7A53A28AF152B949F582F582F5821883F58A|+1793
2D44B00923373974330878|0F7D13000250FCE3EB7BC5908AB5C3678A19C1C18832C15AC126912691269187912689|+199
2D44B00923373974330878|0F7E130002568D5D5EB533F2177B3519186731EF17D02F1831CC1FCC1FCC1FB91FCC17|+432
2D44B00923373974330878|0F7F1300023048CC77D420573DEE1F203DA2247A3D692621245D365D365D36C0345D3E|+665
2D44B00923373974330878|0F80130002758F95C0FDED0AF283E7E1F18FE347F2A8D9E0E2B4E9B4E9B4E901EAB4F1|+898
2D44B00923373974330878|0F811300027E96A3BEBBED50F2F5E627F2D9D981F272E31AE366F366F366F3C7E966FB|+1131
2D44B00923373974330878|0F821300029F74A6626E4B291E544B221EA045381E973F4F3F630F630F630FC2156317|+1363
2D44B00923373974330878|0F8313000202A787B10FC8FC0361CA03047DDCD503F6DBAEDC420C420C420CE30B4204|+1596
2D44B00923373974330878|0F84130002D318C2359A56ED8A585E468B6C54F48A13533B4A078307830783A682078B|+1829
2D44B00923373974330878|0F871300023C06CBC9D3BA58A7EDB11FA7D1AE89A75AB032B65EA05EA05EA0BF9E5EA8|+98
2D44B00923373974330878|0F8813000228A9DCA3840D67D2FE0B90D2F2F86AD259F991F94DC94DC94DC930C94DD1|+332
2D44B00923373974330878|0F891300026C2494360C57FF8A6657B88A7A53A28AF152B949F582F582F5821883F58A|+565
2D44B00923373974330878|0F8A130002B8E351E94990CECB7792A5CB4B9413CCF4939C94E8C3E8C3E8C345C4E8CB|+798
2D44B00923373974330878|0F8B13000269B9BB91A3F248DF1DEAEFDEC1F619DFAAF642FEAEE6AEE6AEE6CFE6AEDE|+1032
2D44B00923373974330878|0F8C130002F0F567EDAFC4DC8881C4E3885DC2F588D6C00EC0A290A290A2900391A288|+1266
2D44B00923373974330878|0F8D13000218EFE3EB7BC5908AB5C3678A19C1C18832C15AC126912691269187912689|+1499
2D44B00923373974330878|0F8E130002DD855D5EB533F2177B3519186731EF17D02F1831CC1FCC1FCC1FB91FCC17|+1733
2D44B00923373974330878|0F8F130002A54FCC77D420573DEE1F203DA2247A3D692621245D365D365D36C0345D3E|+1967
2D44B00923373974330878|0F901300022F9795C0FDED0AF283E7E1F18FE347F2A8D9E0E2B4E9B4E9B4E901EAB4F1|+2202
2D44B00923373974330878|0F91130002C39DA3BEBBED50F2F5E627F2D9D981F272E31AE366F366F366F3C7E966FB|+2436
2D44B00923373974330878|0F92130002EB71A6626E4B291E544B221EA045381E973F4F3F630F630F630FC2156317|+2670
2D44B00923373974330878|0F9513000203EEBEE8D68F41CC1C91EACBC89310CCAF93F794ABC3ABC3ABC3CAC3ABCB|+45
2D44B00923373974330878|0F961300025C9FD2B0CAC73D0428C9F603BCDBE40303DCEBDCB70BB70BB70BD60BB703|+280
2D44B00923373974330878|0F971300022B0DCBC9D3BA58A7EDB11FA7D1AE89A75AB032B65EA05EA05EA0BF9E5EA8|+515
2D44B00923373974330878|0F98130002F5AFDCA3840D67D2FE0B90D2F2F86AD259F991F94DC94DC94DC930C94DD1|+750
2D44B00923373974330878|0F99130002072294360C57FF8A6657B88A7A53A28AF152B949F582F582F5821883F58A|+985
2D44B00923373974330878|0F9A13000251EC51E94990CECB7792A5CB4B9413CCF4939C94E8C3E8C3E8C345C4E8CB|+1220
2D44B00923373974330878|0F9B130002F5BEBB91A3F248DF1DEAEFDEC1F619DFAAF642FEAEE6AEE6AEE6CFE6AEDE|+1456
2D44B00923373974330878|0F9C130002132E67EDAFC4DC8881C4E3885DC2F588D6C00EC0A290A290A2900391A288|+1691
2D44B00923373974330878|0F9D130002B12DE3EB7BC5908AB5C3678A19C1C18832C15AC126912691269187912689|+1927
2D44B00923373974330878|0F9E130002B59D5D5EB533F2177B3519186731EF17D02F1831CC1FCC1FCC1FB91FCC17|+2163
2D44B00923373974330878|0FA1130002E97EA3BEBBED50F2F5E627F2D9D981F272E31AE366F366F366F3C7E966FB|+59
2D44B00923373974330878|0FA213000296A3A6626E4B291E544B221EA045381E973F4F3F630F630F630FC2156317|+295
2D44B00923373974330878|0FA3130002297087B10FC8FC0361CA03047DDCD503F6DBAEDC420C420C420CE30B4204|+532
2D44B00923373974330878|0F6313000203C51DB10FC8FC0369EECC0317D680DB7DDCF6DBD503420C420CE30B4204|+46
2D44B00923373974330878|0F6713000248E8B1C9D3BA58A744C688A7BBBCA4A7D1AE5AB089A75EA05EA0BF9E5EA8|+989
2D44B00923373974330878|0F6B13000261BB619BA3F248DF330E18DFCBF4D4EFC1F6AAF619DFAEE6AEE6CFE6AEDE|+1931
2D44B00923373974330878|0F6C130002470CBDEBAFC4DC887FB0EC88B7C260C75DC2D6C0F588A290A2900391A288|+2166
2D44B00923373974330878|0F6D1300023E0DF9EB7BC5908AABAFC08893C30CCE19C132C1C1882691269187912689|+2401
2D44B00923373974330878|0F6E130002E77BC75CB533F21785414218AD2966386731D02FEF17CC1FCC1FB91FCC17|+2636

djmalin, 2022-01-03:
hello @DomAtHome , You can a help me? add this heat allocator?

weetmuts, 2022-01-11:
The telegram contains a proprietary encoding. What is the meter type/model name?

djmalin, 2022-01-12:
@weetmuts 

`Received telegram from: 74393723
manufacturer: (BMP) BMETERS Polska, Poland (0x9b0)
type: Heat Cost Allocator (0x08)
ver: 0x33
device: rtlwmbus[00000001]
rssi: 157 dBm
driver: unknown!`

weetmuts, 2022-01-13:
Yes, that is the output from wmbusneters, but what does it say on the meter itself. :-)

ptorrent, 2022-05-05:
any news about the implementation of this driver ?

ptorrent, 2022-05-05:
maybe should I force the driver in the meter config file ?

weetmuts, 2022-05-05:
Howdy @djmalin and @ptorrent !

It seems like there are two versions of the Hydroclima RFM HCA out there.
The old version 0x33 and the newer 0x53. I have added a hydroclima driver for 0x53 
(https://github.com/weetmuts/wmbusmeters/issues/534)
because it contains a wmbus compliant HCA value and I happen to have some docs for it.

However the 0x33 I have no idea. @djmalin if you can pick out some
value that correlates to your readings on the display, then we could perhaps
reverse engineer the bytes.

It seems the newer meter likes using two bytes at a time so 0x23 0x01 translates to 0x123 to 291 translates to 29.1
Perhaps this one also does.

heinzek, 2022-09-29:
Hi,

This is from display my BMeters Hydroclima:
C 670.0    - energy consumption in current period 
U 1067.9 - energy consumption in previous period
tC 20.8 - average ambient temperature ('C) in current period
tU 21.1 - previous average ambient temperature ('C) in previous period
2d 31.12 - end period date DD.MM
Ad 29.09 - current date DD.MM
Hb 595d - control code

telegram: 2d44B009837541743308780f55330000ae76102dB72941082c1a2608c8190000c819c8190000000000001f000008

using driver hydroclima:

Auto driver  : not found!
Best driver  : hydroclima 28/63
Using driver : hydroclima 28/63
000   : 2d length (45 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b009 dll-mfct (BMP)
004   : 83754174 dll-id (74417583)
008   : 33 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
011 C?: 0F manufacturer specific data 55330000AE76102DB72941082C1A2608C8190000C819C8190000000000001F000008
012 C!: *** 5533 last 1 month uc 1314.100000   ---  number of measurements (13141) +1 on every telegram
014 C!: *** 0000 last 2 month uc 0.000000
016 C!: *** AE76 last 3 month uc 3038.200000  ---  current time ( every telegram add 120s). I dont know how to decode this time
018 C!: *** 102D last 4 month uc 1153.600000  ---  This maybe is date to time or time is 2 byte long
020 C!: *** B729 last 5 month uc 1067.900000  ---  energy consumption in previous period
022 C!: *** 4108 last 6 month uc 211.300000  ---  previous average ambient temperature ('C) in previous period (21,13 'C) 
024 C!: *** 2C1A last 7 month uc 670.000000  ---  energy consumption in current period 
026 C!: *** 2608 last 8 month uc 208.600000  ---  average ambient temperature ('C) in current period (20,86 'C)
028 C!: *** C819 last 9 month uc 660.000000
030 C!: *** 0000 last 10 month uc 0.000000
032 C!: *** C819 last 11 month uc 660.000000
034 C!: *** C819 last 12 month uc 660.000000
036 C!: *** 0000 date case opened 0
038 C!: *** 0000 date start month 0

https://wmbusmeters.org/analyze/2d44B009837541743308780f55330000ae76102dB72941082c1a2608c8190000c819c8190000000000001f000008

ps. Sorry for my bad english

ismailcelikkan, 2022-12-31:
2144B0092A46B7002A08B2130000513F692D00000000000000000D09DE083B090000000000008882A5

closed 2023-04-06 by github-actions[bot]