wmbusmeters/wmbusmeters #435: Support for wired mbus

awor, 2022-01-03:
Hello weetmuts

If I understand correctly, your fine project only works with wireless mbus. Is this correct? If so, do you have any plans to support wired mbus in the future?

I would like to integrate the following devices:
https://gwf.ch/wp-content/uploads/2018/11/BGZcoder_MP_AE5_EPd30115.pdf
https://mgs-owl.de/mediafiles/Sonstiges/Deltamess/Montageanleitung/Bedien_10242_103243_10244_10245_10246_10247_10248_10249_10250_10251_10252_10253.pdf

Both would be via wired mbus

Best regards

weetmuts, 2022-01-03:
Cool, actually I have implemented almost everything to do plain wired mbus, the test meter is this one:
https://www.piigab.com/en/product/tradad-m-bus-temperatur-och-hygromatare/

To poll it, and assuming it still sits on address 0 and uses 2400 bps, you run this command:
wmbusmeters MYBUS=/dev/ttyUSB0 MyTemp piigth:MYBUS:2400 0 NOKEY

The serial to mbus converter is assumed to be on /dev/ttyUSB0.
What kind of serial <-> mbus converter do you have or do you intend to use?

awor, 2022-01-03:
I currently only have the 3 meters (gas, energy consumption, water) and am investigating how I could hook them up with Home Assistant. I don't have any converters yet. Is there one that you could recommend for use with your solution?

weetmuts, 2022-01-03:
So mbus is pretty nifty. Two wires from the controller sends power to the attached meters. Over these
power lines it also sends serial data (even parity 8 bits 1 stop bit 2400 bps) to the meters and receives serial data from a selected meter. Only one meter can transmit at a time. Therefore the controller polls the meters. (Of course some meters do not care for the power since they might have their own power (electricity meters) also the meters can use different speeds.)

The controller signals 1/0 using changes in voltage and the meters send 1/0 back to the controller using changes in current consumption. https://m-bus.com/documentation-wired/05-data-link-layer

So you need a device that translates serial data from a normal serial port (rs232) to m-bus two wire with power and voltage/current signaling. You can buy this one: https://www.ebay.com/itm/133335990968
or this one: https://www.piigab.com/en/product/m-bus-micro-master/ (a tad bit expensive though)
or build your own: http://domoticx.com/m-bus-hardware-components/

Industrial versions of the m-bus converter that is supposed to handle hundreds of meters can be very expensive since you need good control over the power, current and voltage to manage the load of all the meters while still maintaining a stable voltage for signalling and to be able to find the meter responses in all the noise coming from changes in different meters changes in how they consume the provided power.

awor, 2022-01-03:
Many thanks for your input! Do you think the master version of this might work?
https://www.aliexpress.com/item/4000787535232.html
Would a PRI 1B 512MB run your solution + fedora or should I rather put it on my NUC? I would get a better RPI if I could, but they are sold out until at least April...

weetmuts, 2022-01-03:
It might! However, I do not know the kind of usb-2-serial chip that is used. I have had problems
with another wmbus dongle from china where the serial communication just stops working....

atze09, 2022-04-21:
> Many thanks for your input! Do you think the master version of this might work? https://www.aliexpress.com/item/4000787535232.html

Did you try the master version?
I have ordered the slave version and i will test it during the next days.

weetmuts, 2022-04-21:
@atze09 Please note that:

master = bus controller.
slave = meter attached to bus.

So with a slave version of the device from aliexpress, you can build a meter...
If you want to query your meters, you need a master version, ie bus controller.

weetmuts, 2022-04-21:
The aliexpress interface is a bit weird, I think you select master/client using the color setting of the product. :-)

atze09, 2022-04-21:
> The aliexpress interface is a bit weird, I think you select master/slave using the color setting of the product. :-)

You are right, i checked my aliexpress and ordered the master version, so this should work.
I will test it during the next days! 
Thank you.

atze09, 2022-04-21:
Yes i have 4 heat counting devices, 3 Zenner Zelsius C5 ISF M-BUS and one Sensus PolluStat M-Bus hope those will work.

atze09, 2022-04-24:
Do you think those will work?
-3x Zenner Zelsius C5 ISF (heat meter)
-1x Sensus PolluStat (heat meter)
-1x Sensus water meter
For testing i added both Sensus to the chinese M-BUS device but home assistant wmbusmeter does not show me any telegrams.

This is my log:
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Generating MQTT configuration ... 
Broker 192.168.17.7 will be used.
Running wmbusmeters ...
Started config mbus on /dev/ttyUSB0
No meters configured. Printing id:s of all telegrams heard!`

weetmuts, 2022-04-25:
Before you try to integrate with HASS, we need to get it working from the command line.
Can you download and build the latest version of wmbusmeters.
Then try this:
My sure that at least one of your c5isf meters is connected to the mbus controller on /dev/ttyUSB0.
Then figure out the id number of this meter, typically 8 numerical digits (eg 10000284) 
Then run, but with your meter id:

wmbusmeters --format=json --logtelegrams MAIN=/dev/ttyUSB0:mbus Tempo piigth:MAIN:2400 10000284 NOKEY

You need to use the piigth driver for the moment, but as soon as we have seen that the meter responds, then 
I can fix the c5isf driver for mbus as well.

atze09, 2022-04-25:
Thanks for your awesome support,.can I also try it with the snapd version? Or do I need to compile on my raspberryOS?

atze09, 2022-04-25:
Can you give me a hint how to extract the 8 numerical digit meter id?

atze09, 2022-04-26:
> -3x Zenner Zelsius C5 ISF (heat meter)
> -1x Sensus PolluStat (heat meter)
> -1x Sensus water meter

Seems the ID are the last 8 digits of the printed number on the meter.

weetmuts, 2022-04-26:
Great, I am fixing a bug in the code, which is the cause why you do not get a response right now.
However it is good to connect a single meter at first, because we do not yet know the exact id,mfct,type an version.
Can you check if the 8 digits are acommpanied with 4 digits, 2 digits and 2 digits. (mfct,version,type). 
We might need to scan the mbus and then you can only have a single meter connected, or everyone will talk at nobody can be heard.

atze09, 2022-04-26:
> Great, I am fixing a bug in the code, which is the cause why you do not get a response right now. However it is good to connect a single meter at first, because we do not yet know the exact [...]

What would be the command line for scanning?
I have checjed the digits but it does not match:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=de style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

  | ID
-- | --
Zenner Zelsius C5   ISF (heat meter) Kamin | 32008754
Zenner Zelsius C5   ISF (heat meter) Heizkreis 1 | 10857629
Zenner Zelsius C5   ISF (heat meter) Heizkreis 2 | 32008755
Sensus PolluStat   (heat meter) heat pump | 00324285
Sensus water meter | 51005158?

</div>

</body>

</html>

weetmuts, 2022-04-27:
Fix pushed. It accidentally used hardcoded values for mfct/version/type (matching my mbus device), now it uses wildcard values for those, so only the id matters for secondary addressing. This works fine as long as no meter shares the same id, they usually dont. Please build the latest try again.

weetmuts, 2022-04-27:
Hopefully the meters all listen to 2400 bps, if not, then you you have to test  300, 600, 1200, 2400, 4800 or 9600

weetmuts, 2022-04-27:
Changed the default poll interval to 10 minutes. Instead of every 2 seconds.... :-)
You can change it with --pollinterval=5s when debugging and something greater like 10m
1h or  so when you got stable readings.

weetmuts, 2022-04-27:
When you got a successfull telegram logged (telegram=|...|) then post it here please!

atze09, 2022-04-27:
Thanks for information, i compiled before the pollinterval commit.

But i do a reseponse on any of the bps, the telegrams should appear, right under the started config?

pi@raspberrypi2:~/wmbusmeters $ $ wmbusmeters -pjson --format=json --logtelegrams  MAIN=/dev/ttyUSB0:mbus Tempo c5isf:MAIN:2400:mbus 10857629 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0

Wondering maybe the chinese send me the slave version of their controller, or any other issue.
But both cables are attached and c5isf gives me the ID in it's menu.
i am confused

weetmuts, 2022-04-27:
Ok, can you clone and built this tool: https://github.com/rscada/libmbus
When it is compiled, then you can run the scan command:
..../libmbus/bin/mbus-serial-scan -b 2400 /dev/ttyUSB0
and
..../libmbus/libmbus/bin/mbus-serial-scan-secondary  -b 2400 /dev/ttyUSB0

Try different 2400 bauds as well.

weetmuts, 2022-04-27:
Ah, I found another bug. The bps cannot yet be set from the meter, it has to be set on the device.
wmbusmeters --pollinterval=5s --ppjson --format=json --logtelegrams MAIN=/dev/ttyUSB0:mbus:2400 Tempo c5isf:MAIN:mbus 10857629 NOKEY

Now try different bps 300,2400, etc.

weetmuts, 2022-04-27:
Note that the ordering is important on the device: ...USB0:mbus:2400 works but ...USB0:2400:mbus does not. 
I will fix this.

atze09, 2022-04-28:
> Ok, can you clone and built this tool: https://github.com/rscada/libmbus When it is compiled, then you can run the scan command: ..../libmbus/bin/mbus-serial-scan -b 2400 /dev/ttyUSB0 and ..../libmbus/libmbus/bin/mbus-serial-scan-secondary -b 2400 /dev/ttyUSB0
> 
> Try different 2400 bauds as well.

I have tried and the output was:

pi@raspberrypi2:~/libmbus/bin $ mbus-serial-scan -b 2400 /dev/ttyUSB0
Found a M-Bus device at address 0
^C
pi@raspberrypi2:~/libmbus/bin $ mbus-serial-scan-secondary -b 2400 /dev/ttyUSB0
Found a device on secondary address 10857629496A8804 [using address mask 1FFFFFFFFFFFFFFF]

So seems the adapter found the meter.
pi@raspberrypi2:~/libmbus/bin $ wmbusmeters --ppjson --format=json --logtelegrams MAIN=/dev/ttyUSB0:mbus:2400 Tempo c5isf:MAIN:mbus 10857629 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0

Does not response, (i used without pollinterval because i compiled before the fix.

weetmuts, 2022-04-28:
Ok, run scan-secondary with -d to get debug output.
Run wmbusmeters --debug to get debug output.
Post them here.

atze09, 2022-04-28:
just for overview <html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=de style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

Type | ID | address | secondary address | using address mask | comment
-- | -- | -- | -- | -- | --
Zenner Zelsius C5   ISF (heat meter) Kamin | 32008754 | 0 | 32008754496A8804 | 3FFFFFFFFFFFFFFF |  
Zenner Zelsius C5   ISF (heat meter) Heizkreis 1 | 10857629 | 0 | 10857629496A8804 | 1FFFFFFFFFFFFFFF |  
Zenner Zelsius C5   ISF (heat meter) Heizkreis 2 | 32008755 | 0 | 32008755496A8804 | 3FFFFFFFFFFFFFFF |  
Sensus PolluStat   (heat meter) heat pump | 00324285? | 0 | 20480057C5140004 | 2FFFFFFFFFFFFFFF |  
Sensus water meter | 51005158 | 0 | 51005158496A0007 | 5FFFFFFFFFFFFFFF |  

</div>

</body>

</html>

atze09, 2022-04-28:
> 1FFFFFFFFFFFFFFF

pi@raspberrypi2:~/libmbus/bin $ mbus-serial-scan-secondary -d -b 2400 /dev/ttyUSB0 
init_slaves: debug: sending init frame #1
[2022-04-28 06:14:17Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2022-04-28 06:14:17Z] SEND (005): 10 40 FF 3F 16
[2022-04-28 06:14:17Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 0F FF FF FF FF CA 16
[2022-04-28 06:14:18Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 0F FF FF FF FF CA 16
[2022-04-28 06:14:18Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 1F FF FF FF FF DA 16
[2022-04-28 06:14:18Z] RECV (001): E5
[2022-04-28 06:14:19Z] SEND (005): 10 5B FD 58 16
[2022-04-28 06:14:19Z] RECV (119): 68 71 71 68 08 00 72 29 76 85 10 49 6A 88 04 41 00 00 00 0C 78 29 76 85 10 04 06 93 00 00 00 82 04 6C C1 21 C2 84 00 6C FF FF 84 04 06 00 00 00 00 C4 84 00 06 00 00 00 80 82 0A 6C C1 24 84 0A 06 00 00 00 00 04 13 C3 42 01 00 02 59 DA 0B 02 5D F9 09 02 61 E1 01 04 2D 25 00 00 00 04 3B A6 02 00 00 04 6D 08 07 DC 24 04 26 7D 01 00 00 02 FD 17 00 00 1F 71 16
Found a device on secondary address 10857629496A8804 [using address mask 1FFFFFFFFFFFFFFF]
[2022-04-28 06:14:20Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 2F FF FF FF FF EA 16
[2022-04-28 06:14:20Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 2F FF FF FF FF EA 16
[2022-04-28 06:14:20Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 3F FF FF FF FF FA 16
[2022-04-28 06:14:21Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 3F FF FF FF FF FA 16
[2022-04-28 06:14:21Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 4F FF FF FF FF 0A 16
[2022-04-28 06:14:22Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 4F FF FF FF FF 0A 16
[2022-04-28 06:14:22Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 5F FF FF FF FF 1A 16
[2022-04-28 06:14:22Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 5F FF FF FF FF 1A 16
[2022-04-28 06:14:23Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 6F FF FF FF FF 2A 16
[2022-04-28 06:14:23Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 6F FF FF FF FF 2A 16
[2022-04-28 06:14:24Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 7F FF FF FF FF 3A 16
[2022-04-28 06:14:24Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 7F FF FF FF FF 3A 16
[2022-04-28 06:14:24Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 8F FF FF FF FF 4A 16
[2022-04-28 06:14:25Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 8F FF FF FF FF 4A 16
[2022-04-28 06:14:25Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 9F FF FF FF FF 5A 16
[2022-04-28 06:14:25Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 9F FF FF FF FF 5A 16

And:
pi@raspberrypi2:~/libmbus/bin $ wmbusmeters --ppjson --format=json --logtelegrams --debug MAIN=/dev/ttyUSB0:mbus:2400 Tempo c5isf:MAIN:mbus 10857629 NOKEY
(wmbusmeters) version: 1.7.0-45-ga234cf0
(config) using device: MAIN=/dev/ttyUSB0:mbus:2400 
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus:2400
Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars(3) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received binary "E5"
(mbus) checkMBUSFrame
 "E5"
(mbus) received E5 single byte frame.
(meter) no meter handled  checking 1 templates.
(wmbus) parser expected more data! (844)
(wmbus) telegram from  ignored by all configured meters!
(mbus) checkMBUSFrame
 ""
(mbus) less than 6 bytes, partial frame
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 mbus
(serialtty) closed /dev/ttyUSB0 (mbus)
(wmbus) deleted mbus
(serial) closing 1 devices

weetmuts, 2022-04-28:
Oh, that explains it. It does not send the REQ_UD2 message to the bus.
Can you build the latest version on github? I think is fixed already.

atze09, 2022-04-28:
> Oh, that explains it. It does not send the REQ_UD2 message to the bus. Can you build the latest version on github? I think is fixed already.

pi@raspberrypi2:~/wmbusmeters $ wmbusmeters --ppjson --format=json --logtelegrams  MAIN=/dev/ttyUSB0:mbus:2400 Tempo c5isf:MAIN:mbus 10857629 NOKEYStarted config MAIN=mbus on /dev/ttyUSB0
telegram=|6871716808007229768510496A880443000000_0C782976851004069700000082046CC121C284006CFFFF84040600000000C484000600000080820A6CC124840A06000000000413E44601000259750A025D0D0A02616800042D07000000043B86020000046D2C08DC2404267F01000002FD1700001FBA16|+3
{
    "media":"heat",
    "meter":"c5isf",
    "name":"Tempo",
    "id":"10857629",
    "total_energy_consumption_kwh":151,
    "total_volume_m3":83.684,
    "status":"OK",
    "prev_1_month":"",
    "prev_2_month":"",
    "prev_3_month":"",
    "prev_4_month":"",
    "prev_5_month":"",
    "prev_6_month":"",
    "prev_7_month":"",
    "prev_8_month":"",
    "prev_9_month":"",
    "prev_10_month":"",
    "prev_11_month":"",
    "prev_12_month":"",
    "prev_13_month":"",
    "prev_14_month":"",
    "prev_1_month_kwh":0,
    "prev_2_month_kwh":0,
    "prev_3_month_kwh":0,
    "prev_4_month_kwh":0,
    "prev_5_month_kwh":0,
    "prev_6_month_kwh":0,
    "prev_7_month_kwh":0,
    "prev_8_month_kwh":0,
    "prev_9_month_kwh":0,
    "prev_10_month_kwh":0,
    "prev_11_month_kwh":0,
    "prev_12_month_kwh":0,
    "prev_13_month_kwh":0,
    "prev_14_month_kwh":0,
    "prev_1_month_m3":0,
    "prev_2_month_m3":0,
    "prev_3_month_m3":0,
    "prev_4_month_m3":0,
    "prev_5_month_m3":0,
    "prev_6_month_m3":0,
    "prev_7_month_m3":0,
    "prev_8_month_m3":0,
    "prev_9_month_m3":0,
    "prev_10_month_m3":0,
    "prev_11_month_m3":0,
    "prev_12_month_m3":0,
    "prev_13_month_m3":0,
    "prev_14_month_m3":0,
    "due_energy_consumption_kwh":0,
    "due_date":"2022-01-01",
    "volume_flow_m3h":0.646,
    "power_kw":0.7,
    "total_energy_consumption_last_month_kwh":0,
    "last_month_date":"",
    "max_power_last_month_kw":0,
    "flow_temperature_c":26.77,
    "return_temperature_c":25.73,
    "timestamp":"2022-04-28T07:50:23Z"

Seems to work, i will try also the other sensus meters.
Is the HASSIO integration also updated?

atze09, 2022-04-28:
The Sensus Pollucom is new and gives me the follwing output:

pi@raspberrypi2:~ $ wmbusmeters --ppjson --format=json --logtelegrams MAIN=/dev/ttyUSB0:mbus:2400 Tempo c5isf:MAIN:mbus 20480057 NOKEYStarted config MAIN=mbus on /dev/ttyUSB0
(meter) Tempo: meter detection did not match the selected driver unknown! correct driver is: sensostar
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
telegram=|68B3B36808007257004820C514000464100000_047839803801040600000000041300000000042B00000000142B00000000043B00000000143B00000000025B1000025F10000261F0FF02235C00046D3828DC24440600000000441300000000426C000001FD171003FD0C05000084200600000000C420060000000084300600000000C430060000000084401300000000C44013000000008480401300000000C48040130000000084C0401300000000C4C0401300000000AE16|+4
{
    "media":"heat",
    "meter":"c5isf",
    "name":"Tempo",
    "id":"20480057",
    "total_energy_consumption_kwh":0,
    "total_volume_m3":0,
    "status":"",
    "prev_1_month":"",
    "prev_2_month":"",
    "prev_3_month":"",
    "prev_4_month":"",
    "prev_5_month":"",
    "prev_6_month":"",
    "prev_7_month":"",
    "prev_8_month":"",
    "prev_9_month":"",
    "prev_10_month":"",
    "prev_11_month":"",
    "prev_12_month":"",
    "prev_13_month":"",
    "prev_14_month":"",
    "prev_1_month_kwh":0,
    "prev_2_month_kwh":0,
    "prev_3_month_kwh":0,
    "prev_4_month_kwh":0,
    "prev_5_month_kwh":0,
    "prev_6_month_kwh":0,
    "prev_7_month_kwh":0,
    "prev_8_month_kwh":0,
    "prev_9_month_kwh":0,
    "prev_10_month_kwh":0,
    "prev_11_month_kwh":0,
    "prev_12_month_kwh":0,
    "prev_13_month_kwh":0,
    "prev_14_month_kwh":0,
    "prev_1_month_m3":0,
    "prev_2_month_m3":0,
    "prev_3_month_m3":0,
    "prev_4_month_m3":0,
    "prev_5_month_m3":0,
    "prev_6_month_m3":0,
    "prev_7_month_m3":0,
    "prev_8_month_m3":0,
    "prev_9_month_m3":0,
    "prev_10_month_m3":0,
    "prev_11_month_m3":0,
    "prev_12_month_m3":0,
    "prev_13_month_m3":0,
    "prev_14_month_m3":0,
    "due_energy_consumption_kwh":0,
    "due_date":"",
    "volume_flow_m3h":0,
    "power_kw":0,
    "total_energy_consumption_last_month_kwh":0,
    "last_month_date":"",
    "max_power_last_month_kw":0,
    "flow_temperature_c":16,
    "return_temperature_c":16,
    "timestamp":"2022-04-28T07:57:18Z"

Seems to work also with c5isf preset

weetmuts, 2022-04-28:
Great! Yes, please try the other meters and post the telegram=|...| for them as well.
All integrations with HASSIO etc should work as before.

atze09, 2022-04-28:
The sensur water meter:

pi@raspberrypi2:~ $ wmbusmeters --ppjson --format=json --logtelegrams MAIN=/dev/ttyUSB0:mbus:2400 Tempo iperl:MAIN:mbus 51005158 NOKEYStarted config MAIN=mbus on /dev/ttyUSB0
telegram=|6874746808007258510051496A000703000000_0C7858510051046D0B09DC240413000000000C943C000000004413FFFFFFFF426CFFFF840113FFFFFFFF82016CFFFFC40113FFFFFFFFC2016CFFFF840213FFFFFFFF82026CFFFF043B0000000004223A24000004260000000034220000000002FD1700001FA316|+3
{
    "media":"water",
    "meter":"iperl",
    "name":"Tempo",
    "id":"51005158",
    "total_m3":0,
    "max_flow_m3h":0,
    "timestamp":"2022-04-28T08:21:41Z"
}

atze09, 2022-04-28:
> iperl

For home assistant, i get this log:
config:
data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=/dev/ttyUSB0:mbus
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - name=HK1 driver=c5isf id=10857629
  - name=HK2 driver=c5isf id=32008755
  - name=Kamin driver=c5isf id=32008754
  - name=Heatpump driver=c5isf id=20480057
  - name=WW2 driver=iperl id=51005158

Output:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Adding meter-0002 ...
Adding meter-0003 ...
Adding meter-0004 ...
Adding meter-0005 ...
Generating MQTT configuration ... 
Broker 192.168.17.7 will be used.
Running wmbusmeters ...
Not a valid meter id nor a valid meter match expression ""
Started config mbus on /dev/ttyUSB0
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 00000000
          manufacturer: (@@@) Unknown (0x00)
                  type: Other (0x00)
                   ver: 0x00
                driver: unknown!
telegram=|_|+0

weetmuts, 2022-04-28:
Change device=... to
device=MAIN=/dev/ttyUSB0:mbus:2400
and for each meter:
driver=c5isf:MAIN:mbus

weetmuts, 2022-04-28:
Use sensostar instead of iperl.

atze09, 2022-04-28:
data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=MAIN=/dev/ttyUSB0:mbus:2400
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - name=HK1 driver=c5isf:MAIN:mbus id=10857629
  - name=HK2 driver=c5isf:MAIN:mbus id=32008755
  - name=Kamin driver=c5isf:MAIN:mbus id=32008754
  - name=Heatpump driver=c5isf:MAIN:mbus id=20480057
  - name=WW2 driver=sensorstar:MAIN:mbus  id=51005158

Found invalid meter name "HK2 driver=c5isf:MAIN:mbus id=32008755" in meter config file, must not contain a ':', skipping meter.

atze09, 2022-04-28:
> must not contain a ':',

seems driver can must not contain a ':',

atze09, 2022-04-28:
> Use sensostar instead of iperl.

I did, but in home assistant i get the connection refused error.
I read in another issue that this should have been solved?
Registering meters ...
Adding meter-0001 ...
Adding meter-0002 ...
Adding meter-0003 ...
Adding meter-0004 ...
Adding meter-0005 ...
Generating MQTT configuration ... 
Broker 192.168.17.7 will be used.
Running wmbusmeters ...
Started config MAIN=mbus on /dev/ttyUSB0
telegram=|6871716808007229768510496A880448000000_0C782976851004069900000082046CC121C284006CFFFF84040600000000C484000600000080820A6CC124840A06000000000413414E01000259130A025DC70902614C00042D05000000043B7B020000046D270BDC2404268201000002FD1700001F5516|+5
Connection error: Connection Refused: not authorised.
Error: The connection was refused.
(shell) /bin/sh exited with non-zero return code: 5
telegram=|6871716808007229768510496A880449000000_0C782976851004069900000082046CC121C284006CFFFF84040600000000C484000600000080820A6CC124840A06000000000413434E010002590B0A025DC80902614300042D04000000043B7B020000046D270BDC2404268201000002FD1700001F4716|+8
Connection error: Connection Refused: not authorised.
Error: The connection was refused.
(shell) /bin/sh exited with non-zero return code: 5

atze09, 2022-04-28:
> Use sensostar instead of iperl.

Do you mean for the water meter or the Sensus heat meter?

My setting now works for the c5isf heat meter, but not for the Sensus Pollustat heat meter:
data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=MAIN=/dev/ttyUSB0:mbus:2400
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=HK1
    driver=c5isf:MAIN:mbus
    id=10857629
    name=HK2
    driver=c5isf:MAIN:mbus
    id=32008755
    name=Kamin
    driver=c5isf:MAIN:mbus
    id=32008754
    name=Heatpump
    driver=c5isf:MAIN:mbus
    id=20480057

This is the output, but sensostar only has the attributes for water sesor and not all the heating attributes.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker 192.168.17.7 will be used.
Running wmbusmeters ...
Started config MAIN=mbus on /dev/ttyUSB0
(meter) Heatpump: meter detection did not match the selected driver unknown! correct driver is: sensostar
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
telegram=|68B3B36808007257004820C51400046C100000_047839803801040600000000041300000000042B00000000142B00000000043B00000000143B00000000025B1400025F15000261DAFF02235C00046D2C2DDC24440600000000441300000000426C000001FD171003FD0C05000084200600000000C420060000000084300600000000C430060000000084401300000000C44013000000008480401300000000C48040130000000084C0401300000000C4C0401300000000A216|+3

weetmuts, 2022-05-01:
I have fixed a couple of things with the polling. It might fail to acquire the next wired mbus meter because it did not wait for the response before sending the next query. Please test the latest version.

atze09, 2022-05-01:
Awesome Fredrik, thank you very much!
Multiple Sensors are now working.
But now i have a question to the Sensus Pollustat, here you propose the sensostar template poll.
But this is alsow a heat meter, same as the Zenner C5isf, so what i currently get in mqtt is:
[image]

But what is missing are all the other stats from c5isf (flow temperatures and volume flow?)
[image]

Are those missing because of the error my device has "ERROR(0010)" or is this a configuration issue?
Shall i contact the company Sensus for detailed information?

atze09, 2022-05-04:
> Yes, there is a lot in your sensostar telegram. I have converted the old style driver meter_sensostar.cc to the new style driver driver_sensostar.cc as you can see the new driver is much easier to read.
> 
> Build the latest and do
> 
> ```
> wmbusmeters --analyze 68B3B36808007257004820c51400046c100000047839803801040600000000041300000000042B00000000142B00000000043B00000000143B00000000025B1400025f15000261daff02235c00046d2c2ddc24440600000000441300000000426c000001fd171003fd0c05000084200600000000c420060000000084300600000000c430060000000084401300000000c44013000000008480401300000000c48040130000000084c0401300000000c4c0401300000000a216
> ```
> 
> I have added the values you need, but there is a lot more, see if you can add some numeric fields yourself!

Thank you
I am already in home assistant and don't want to switch back to my test RPi , so i think i cannot work with the command above (or is there a way to also work with this command in HASSIO?)

As i just saw in the isseue #46 ? @erniberni  showed the result of libmbus:

Maybe it is possible to have the same attributes as c5isf, if available?
I am currently building the latest release and check if this is sufficient for me.
Thank you for your awesome support

weetmuts, 2022-05-04:
This is what the driver currently extracts. 
C? means that the field is not extract, but is available in the telegram.
C1 means that a match has been found and the content is available in the json.
```
000   : 68 start
001   : b3 length (179 bytes)
002   : b3 length again (179 bytes)
003   : 68 start
004   : 08 dll-c (RSP_UD2)
005   : 00 dll-a primary (0)
006   : 72 tpl-ci (long header)
007   : 57004820 tpl-id (20480057)
011   : c514 tpl-mfct (EFE)
013   : 00 tpl-version
014   : 04 tpl-type (Heat meter)
015   : 6c tpl-acc-field
016   : 10 tpl-sts-field (TEMPORARY_ERROR)
017   : 0000 tpl-cfg 0000 ( )
019   : 04 dif (32 Bit Integer/Binary Instantaneous value)
020   : 78 vif (Fabrication no)
021 C?: 39803801
025   : 04 dif (32 Bit Integer/Binary Instantaneous value)
026   : 06 vif (Energy kWh)
027 C!: 00000000 ("total_kwh":0)
031   : 04 dif (32 Bit Integer/Binary Instantaneous value)
032   : 13 vif (Volume l)
033 C!: 00000000 ("total_water_m3":0)
037   : 04 dif (32 Bit Integer/Binary Instantaneous value)
038   : 2B vif (Power W)
039 C?: 00000000
043   : 14 dif (32 Bit Integer/Binary Maximum value)
044   : 2B vif (Power W)
045 C?: 00000000
049   : 04 dif (32 Bit Integer/Binary Instantaneous value)
050   : 3B vif (Volume flow l/h)
051 C!: 00000000 ("flow_water_m3h":0)
055   : 14 dif (32 Bit Integer/Binary Maximum value)
056   : 3B vif (Volume flow l/h)
057 C!: 00000000 ("flow_water_max_m3h":0)
061   : 02 dif (16 Bit Integer/Binary Instantaneous value)
062   : 5B vif (Flow temperature °C)
063 C!: 1400 ("forward_c":20)
065   : 02 dif (16 Bit Integer/Binary Instantaneous value)
066   : 5F vif (Return temperature °C)
067 C!: 1500 ("return_c":21)
069   : 02 dif (16 Bit Integer/Binary Instantaneous value)
070   : 61 vif (Temperature difference 10⁻² K/°C)
071 C!: DAFF ("difference_c":-0.38)
073   : 02 dif (16 Bit Integer/Binary Instantaneous value)
074   : 23 vif (On time days)
075 C?: 5C00
077   : 04 dif (32 Bit Integer/Binary Instantaneous value)
078   : 6D vif (Date and time type)
079 C!: 2C2DDC24 ("meter_timestamp":"2022-04-28 13:44")
083   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
084   : 06 vif (Energy kWh)
085 C?: 00000000
089   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
090   : 13 vif (Volume l)
091 C?: 00000000
095   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
096   : 6C vif (Date type G)
097 C?: 0000
099   : 01 dif (8 Bit Integer/Binary Instantaneous value)
100   : FD vif (Second extension FD of VIF-codes)
101   : 17 vife (Error flags (binary))
102 C!: 10 ("current_status":"UNKNOWN_ERROR_FLAGS(0x10)")
103   : 03 dif (24 Bit Integer/Binary Instantaneous value)
104   : FD vif (Second extension FD of VIF-codes)
105   : 0C vife (Model/Version)
106 C?: 050000
109   : 84 dif (32 Bit Integer/Binary Instantaneous value)
110   : 20 dife (subunit=0 tariff=2 storagenr=0)
111   : 06 vif (Energy kWh)
112 C?: 00000000
116   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
117   : 20 dife (subunit=0 tariff=2 storagenr=1)
118   : 06 vif (Energy kWh)
119 C?: 00000000
123   : 84 dif (32 Bit Integer/Binary Instantaneous value)
124   : 30 dife (subunit=0 tariff=3 storagenr=0)
125   : 06 vif (Energy kWh)
126 C?: 00000000
130   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
131   : 30 dife (subunit=0 tariff=3 storagenr=1)
132   : 06 vif (Energy kWh)
133 C?: 00000000
137   : 84 dif (32 Bit Integer/Binary Instantaneous value)
138   : 40 dife (subunit=1 tariff=0 storagenr=0)
139   : 13 vif (Volume l)
140 C?: 00000000
144   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
145   : 40 dife (subunit=1 tariff=0 storagenr=1)
146   : 13 vif (Volume l)
147 C?: 00000000
151   : 84 dif (32 Bit Integer/Binary Instantaneous value)
152   : 80 dife (subunit=0 tariff=0 storagenr=0)
153   : 40 dife (subunit=2 tariff=0 storagenr=0)
154   : 13 vif (Volume l)
155 C?: 00000000
159   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
160   : 80 dife (subunit=0 tariff=0 storagenr=1)
161   : 40 dife (subunit=2 tariff=0 storagenr=1)
162   : 13 vif (Volume l)
163 C?: 00000000
167   : 84 dif (32 Bit Integer/Binary Instantaneous value)
168   : C0 dife (subunit=1 tariff=0 storagenr=0)
169   : 40 dife (subunit=3 tariff=0 storagenr=0)
170   : 13 vif (Volume l)
171 C?: 00000000
175   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
176   : C0 dife (subunit=1 tariff=0 storagenr=1)
177   : 40 dife (subunit=3 tariff=0 storagenr=1)
178   : 13 vif (Volume l)
179 C?: 00000000
183   : A2 dif (16 Bit Integer/Binary Minimum value)
184   : 16 dife (subunit=0 tariff=1 storagenr=12)

{
    "media":"heat",
    "meter":"sensostar",
    "name":"",
    "id":"20480057",
    "meter_timestamp":"2022-04-28 13:44",
    "total_kwh":0,
    "flow_water_m3h":0,
    "flow_water_max_m3h":0,
    "forward_c":20,
    "return_c":21,
    "difference_c":-0.38,
    "total_water_m3":0,
    "current_status":"UNKNOWN_ERROR_FLAGS(0x10)",
    "timestamp":"2022-05-04T18:39:47Z"
}
```

atze09, 2022-05-04:
Awesome, I wanted to calculate the power on my own in Hassio,but there is a power telegram.
Is it possible to add 045 power instant velue (W) then I would be very Happy.
Thanks in advance

weetmuts, 2022-05-04:
Ok, I added power_kw and power_max_kw 
If you can figure out the time period over which the max is calculated, please let me know.
That is typical of the knowledge I want to put into the meter driver, in this case the explanation.
Is it max over 24 hour rolling? max over last day, last hour, etc....

There is an error condition on your meter. If you can figure out what it means, then we can know one of
the status bits.

awor, 2022-07-30:
Hi @weetmuts 

I had originally started this issue back in Jan. Now I am finally getting to it :-) Thanks to the progress you made with @atze09 I managed to get this far:

root@wmbus:/home/libmbus/bin# ./mbus-serial-scan-secondary -d -b 2400 /dev/ttyUSB0
init_slaves: debug: sending init frame #1
[2022-07-30 08:05:22Z] SEND (005): 10 40 FD 3D 16
init_slaves: debug: sending init frame #2
[2022-07-30 08:05:23Z] SEND (005): 10 40 FF 3F 16
[2022-07-30 08:05:23Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 0F FF FF FF FF CA 16
[2022-07-30 08:05:23Z] RECV (001): E5
[2022-07-30 08:05:23Z] SEND (005): 10 5B FD 58 16
[2022-07-30 08:05:24Z] RECV (148): 68 8E 8E 68 08 01 72 24 99 30 00 E6 1E 08 04 02 00 00 00 04 06 81 46 06 00 04 14 88 F2 35 00 84 40 14 26 E0 26 00 02 5B 30 00 02 5F 2B 00 02 62 38 00 04 22 46 88 02 00 04 26 46 88 02 00 04 3B 00 00 00 00 04 2C 00 00 00 00 04 6D 28 0D DE 27 84 40 6E 00 00 00 00 84 80 40 6E 00 00 00 00 C4 80 00 06 B1 45 06 00 C4 80 00 14 3B E4 35 00 C4 C0 00 14 2F D2 26 00 C4 C0 00 6E 00 00 00 00 C4 80 40 6E 00 00 00 00 C2 80 00 6C C1 27 1F 00 00 00 00 09 16
Found a device on secondary address 00309924E61E0804 [using address mask 0FFFFFFFFFFFFFFF]
[2022-07-30 08:05:24Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 1F FF FF FF FF DA 16
[2022-07-30 08:05:25Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 1F FF FF FF FF DA 16
[2022-07-30 08:05:25Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 2F FF FF FF FF EA 16
[2022-07-30 08:05:26Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 2F FF FF FF FF EA 16
[2022-07-30 08:05:26Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 3F FF FF FF FF FA 16
[2022-07-30 08:05:26Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 3F FF FF FF FF FA 16
[2022-07-30 08:05:27Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 4F FF FF FF FF 0A 16
[2022-07-30 08:05:27Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 4F FF FF FF FF 0A 16
[2022-07-30 08:05:28Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 5F FF FF FF FF 1A 16
[2022-07-30 08:05:28Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 5F FF FF FF FF 1A 16
[2022-07-30 08:05:28Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 6F FF FF FF FF 2A 16
[2022-07-30 08:05:29Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 6F FF FF FF FF 2A 16
[2022-07-30 08:05:29Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 7F FF FF FF FF 3A 16
[2022-07-30 08:05:29Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 7F FF FF FF FF 3A 16
[2022-07-30 08:05:30Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 8F FF FF FF FF 4A 16
[2022-07-30 08:05:30Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 8F FF FF FF FF 4A 16
[2022-07-30 08:05:31Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 9F FF FF FF FF 5A 16
[2022-07-30 08:05:31Z] SEND (017): 68 0B 0B 68 73 FD 52 FF FF FF 9F FF FF FF FF 5A 16

wmbusmeters --ppjson --format=json --logtelegrams --debug MAIN=/dev/ttyUSB0:mbus:2400 Tempo c5isf:MAIN:mbus 00309924 NOKEY
(wmbusmeters) version: 1.8.0-2-gd2918c5
(config) using device: MAIN=/dev/ttyUSB0:mbus:2400
(config) number of meters: 1
(meter) Tempo 00309924 needs polling but has no pollinterval set!
(meter) created Tempo c5isf  encrypted
(lookup) with file/hex "/dev/ttyUSB0" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB0:mbus:2400
Started config MAIN=mbus on /dev/ttyUSB0
(mbus) on /dev/ttyUSB0
(serial) before config: /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars(3) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B2400) input() output() control(CLOCAL|CREAD|CSIZE|PARENB) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (mbus)
Sending NKE to mbus MAIN
(serial /dev/ttyUSB0) sent "1040004016"
(serial) EVENT thread interrupted
(main) regular reset of mbus  on /dev/ttyUSB0 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 mbus
(serialtty) closed /dev/ttyUSB0 (mbus)
(serial) TIMER thread interrupted
(serial) event loop stopped!
(wmbus) deleted mbus
(serial) closing 1 devices

I just used the same driver that atze09 had used, but that's probably wrong. The device I am trying to connect with is an Enercal F2. It is very popular in the German speaking part of Europe. If you google it, you'll find many hits. Can you please tell me what I should be doing to get my Enercal F2 to talk with your solution? Thanks in advance!

awor, 2022-07-30:
Ah, I carefully re-read everything in this issue and now tried this:

root@wmbus:/home/libmbus/bin# wmbusmeters --format=json --logtelegrams  --pollinterval=5 MAIN=/dev/ttyUSB0:mbus Tempo piigth:MAIN:2400 00309924 NOKEY
Started config MAIN=mbus on /dev/ttyUSB0
(meter) Tempo: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 00000000 mfct: (@@@) Unknown (0x00) type: Other (0x00) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|688E8E6808017224993000E61E080405000000_040681460600041488F2350084401426E02600025B2600025F220002622A00042247880200042647880200043B00000000042C00000000046D2D0EDE2784406E000000008480406E00000000C4800006B1450600C48000143BE43500C4C000142FD22600C4C0006E00000000C480406E00000000C280006CC1271F00000000F316|+4
{"media":"heat","meter":"piigth","name":"Tempo","id":"00309924","operating_time_h":46.099722,"on_time_h":46.099722,"meter_datetime":"2022-07-30 14:45","temperature_c":null,"average_temperature_1h_c":null,"average_temperature_24h_c":null,"relative_humidity_rh":null,"relative_humidity_1h_rh":null,"relative_humidity_24h_rh":null,"timestamp":"2022-07-30T09:10:02Z","device":"MAIN","rssi_dbm":0}
telegram=|688E8E6808017224993000E61E080406000000_040681460600041488F2350084401426E02600025B2600025F220002622A00042247880200042647880200043B00000000042C00000000046D2D0EDE2784406E000000008480406E00000000C4800006B1450600C48000143BE43500C4C000142FD22600C4C0006E00000000C480406E00000000C280006CC1271F00000000F416|+11
{"media":"heat","meter":"piigth","name":"Tempo","id":"00309924","operating_time_h":46.099722,"on_time_h":46.099722,"meter_datetime":"2022-07-30 14:45","temperature_c":null,"average_temperature_1h_c":null,"average_temperature_24h_c":null,"relative_humidity_rh":null,"relative_humidity_1h_rh":null,"relative_humidity_24h_rh":null,"timestamp":"2022-07-30T09:10:09Z","device":"MAIN","rssi_dbm":0}
^C
root@wmbus:/home/libmbus/bin#

...so I suppose my Enercal F2 is not supported?

qwert4u, 2023-04-26:
hello,
I use home assistant and have installed wmbusmeters as an addon. For this I bought this device as a master.
but i can't see any telegrams...

https://www.aliexpress.com/item/4000787535232.html

my settings:

device= MAIN=/dev/ttyUSB0:mbus:2400
logtelegrams = true

counter:
driver=hydrus
Taste=51728910E66D83F8
name=WE1 KW Bad

log:
[13:45:43] INFO: {"device": "MAIN=/dev/ttyUSB0:mbus:2400", "donotprobe":"/dev/ttyAMA0", "format": "json", "logfile":"/dev/stdout", "loglevel": "normal", "logtelegrams":true, "shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[13:45:43] INFO: CONFIG_METERS ...
[13:45:43] INFO: [{"driver": "hydrus", "key": "51728910E66D83F8", "name": "WE1 KW Bad"}]
[13:45:43] INFO: synchronize wmbusmeters configuration ...
[13:45:43] INFO: Register meter ...
[13:45:43] INFO: Adding meter-0001 ...
[13:45:43] INFO: Generate MQTT configuration ... 
[13:45:45] INFO: broker core-mosquitto is used.
[13:45:46] INFO: 
MQTT discovery ...
[13:45:47] INFO: Adding meter: WE1 KW Bad ...
[13:45:47] INFO: File /config/wmbusmeters/etc/mqtt_discovery/hydrus.json not found.
[13:45:47] INFO: MQTT Discovery cleanup...
[13:45:48] INFO: removed topic: homeassistant/sensor/wmbusmeters/12345678_current_power_consumption_kw/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_current_power_consumption_kw/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_device_date_time/config
[13:45:48] INFO: Remove topic: homeassistant/sensor/wmbusmeters/12345678_rssi_dbm/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_timestamp/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_total_energy_consumption_kwh/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_total_energy_generation_kwh/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_voltage_at_phase_1_v/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_voltage_at_phase_2_v/config
[13:45:48] INFO: remove topic: homeassistant/sensor/wmbusmeters/12345678_voltage_at_phase_3_v/config
[13:45:48] INFO: Starting the web configuration service.
[13:45:48] INFO: Running wmbusmeters ...
No valid meter ID and no valid meter match expression "".
Starting the configuration MAIN=mbus on /dev/ttyUSB0.
No meters configured. Print id:s of all telegrams heard!
[13:45:50] INFO: Start Nginx...
2023/04/26 13:45:50 [notice] 84#84: using "epoll" event method
2023/04/26 13:45:50 [notice] 84#84: nginx/1.22.1
2023/04/26 13:45:50 [notice] 84#84: OS: Linux 6.1.21-v8
2023/04/26 13:45:50 [notice] 84#84: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/04/26 13:45:50 [notice] 84#84: start worker processes
2023/04/26 13:45:50 [notice] 84#84: start worker process 269

qwert4u, 2023-04-26:
under windows the adapter works
[image]

qwert4u, 2023-04-26:
it works with the settings:
[image]

stepinskir, 2023-05-10:
i have IME CONTO D4 electricity meter, trying to get it working with wmbusmeters but no much luck... have some kind of communication using piigth:MAIN:mbus driver here's a log:

```
(serial) received binary "0000"
(mbus) checkMBUSFrame
 "1E00780000"
(mbus) less than 6 bytes, partial frame
(serial) received binary "78"
(mbus) checkMBUSFrame
 "1E0078000078"
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) protocol error ""
(serial) received binary ""
(mbus) checkMBUSFrame
 ""
(mbus) less than 6 bytes, partial frame
(serial) received binary "00"
(mbus) checkMBUSFrame
 "00"
(mbus) less than 6 bytes, partial frame
(serial) received binary ""
(mbus) checkMBUSFrame
 "00"
(mbus) less than 6 bytes, partial frame
(serial) received binary "78"
(mbus) checkMBUSFrame
 "0078"
(mbus) less than 6 bytes, partial frame
(serial) received binary "00"
(mbus) checkMBUSFrame
 "007800"
(mbus) less than 6 bytes, partial frame
(serial) received binary "8078"
(mbus) checkMBUSFrame
 "0078008078"
(mbus) less than 6 bytes, partial frame
(serial) received binary "1E3C0FFE00"
(mbus) checkMBUSFrame
 "00780080781E3C0FFE00"
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) protocol error ""
(serial) received binary "000000"
(mbus) checkMBUSFrame
 "000000"
(mbus) less than 6 bytes, partial frame
(serial) received binary "8000F878"
(mbus) checkMBUSFrame
 "0000008000F878"
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) protocol error ""
(serial) received binary "1EC0"
(mbus) checkMBUSFrame
 "1EC0"
(mbus) less than 6 bytes, partial frame
(serial) received binary "80F878"
(mbus) checkMBUSFrame
 "1EC080F878"
(mbus) less than 6 bytes, partial frame
(serial) received binary "1EC0"
(mbus) checkMBUSFrame
 "1EC080F8781EC0"
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) protocol error ""
(serial) received binary "00"
(mbus) checkMBUSFrame
 "00"
(mbus) less than 6 bytes, partial frame
(serial) received binary "00F80000"
(mbus) checkMBUSFrame
 "0000F80000"
(mbus) less than 6 bytes, partial frame
(serial) received binary "000000"
(mbus) checkMBUSFrame
 "0000F80000000000"
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) protocol error ""
(serial) received binary "00"
(mbus) checkMBUSFrame
 "00"
(mbus) less than 6 bytes, partial frame
(serial) received binary "78"
(mbus) checkMBUSFrame
 "0078"
(mbus) less than 6 bytes, partial frame
(serial) received binary "00781EFC"
(mbus) checkMBUSFrame
 "007800781EFC"
(mbus) no 0x68 byte found, clearing buffer.
(mbus) protocol error in message received!
(mbus) protocol error ""
(serial) received binary "78"
(mbus) checkMBUSFrame
 "78"
(mbus) less than 6 bytes, partial frame
(serial) received binary "00"
(mbus) checkMBUSFrame
 "7800"
(mbus) less than 6 bytes, partial frame
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
(meter) Meter_at_pump 00000002 did not send a response!
(meter) secondary addressing bus MAIN to address 00000002
(serial /dev/ttyUSB0) sent "680B0B6873FD5202000000FFFFFFFFC016"
(serial) EVENT thread interrupted
(meter) polling Meter_at_pump 00000002 (secondary) with req ud2 bus MAIN
(serial /dev/ttyUSB0) sent "105BFD5816"
(serial) EVENT thread interrupted
```

documentation for mbus protocol for these meters can be found here:
[](https://www.imesys.fr/protocole-pdf/PR112.pdf)

any help appreciated...

weetmuts, 2023-05-12:
@stepinskir Wmbusmeters currently only support long secondary addresses for chosing the meter.
It seems you use a very short address 2? This is probably a primary address.

There is an issue where support for primary addresses will be fixed. https://github.com/wmbusmeters/wmbusmeters/issues/669 
Please join that issue.

stepinskir, 2023-05-12:
Hi, secondary address can be changed and it is set to 0000002 by myself...

weetmuts, 2023-05-12:
Then, it does not seem like it is responding to the eq ud2 query.
Have to talked to the meter using another tool?

weetmuts, 2023-05-12:
A sorry, you wrote that you sed the windows tool. Hmmmm.

weetmuts, 2023-05-12:
Can you test setting a secondary address like 12345678 ?

weetmuts, 2023-05-12:
I can see a difference wmbusmeters sends for secondary addressing:
 "680B0B6873FD5202000000FFFFFFFFC016"
 
But the example in the documentation gives:
 "680B0B6853FD5202000000A52514028D16"

The final bytes are masked with FF in the wmbusmeters send which is fine.
However the c-field byte sent by wmbusmeters is 0x73
however their example says 0x53.

Do you know what this difference could mean?

stepinskir, 2023-05-12:
srry i don't...

[81 more replies]