wmbusmeters/wmbusmeters #436: [Question] Support for R mode (e.g. for istameter radio net 3 module)

ustaerk, 2022-01-04:
Hi all,

my meters are equipped with ista radio net 3 modules (part no. 19320). From reading the [manual](https://www.ista.com/fileadmin/twt_customer/countries/content/Switzerland/Broschueren/Funk/Produktbroschuere_Funksystem_symphonic-sensor-net.pdf#page=40) (excerpt below, German only I'm afraid) I come to believe it is working in R mode, i.e. sending only when triggered:

- Frequency 868 Mhz, i.e. not N or F mode
- Bidirectional, i.e. must be one of S2, C2, T2 or R
- The manual mentions sending data only on request, which is consistent with me being unable to see any of my meters on the radio
- I am thus concluding it is most likely operating in R mode

Two questions to the community:

1. Would you agree that those devices are likely operating in R mode?
2. How would you go about getting to the meter data?

Cheers, Uli

Screenshot from the manual:
[image]

weetmuts, 2022-01-05:
It might be R mode. Wmbusmeters has almost all the infrastructure in place to do bidirectionl C2,S2,T2,R communication in place if you have a dongle like im871a which support all of those. 

You need a dongle (im871a recommended) and the key to your meter and some documentation on how to wake it up.
Then we can experiment with sending telegrams and see if we can get a response and eventually implement full polling support in wmbusmeters.

robo-kob, 2022-01-29:
@ustaerk 
Ulrich do you have a solution? I would like to get information from heatcostcounter ista doprimo 3 and also radio net 3 for hot and cold water. Which kind of wmbus dongle you are using? My CUL only get Techem messages from my neibours ;) 

THX Rob

Hankey71, 2022-02-05:
Hello, I just tested vmbusmeters with an RTL2832U DVB stick to read my ISTA counter. I am trying to communicate with the module added to the counter. There is a button on the module, when I press for 5 seconds, the module flashes red for about 30 seconds. I tried "wmbusmeters --debug --t1 rtlwmbus" and "wmbusmeters --debug --c1 rtlwmbus" and it doesn't work. I find no signal.
So I imagine that you have to go through other dongles and other frequencies... If you have any solutions, I'm a taker!

Hankey71, 2022-02-05:
[image]
My module on the counter

jacekjaros, 2022-02-08:
Ista manual in English: https://www.ista.com/fileadmin/twt_customer/countries/content/Tutorial/Documents/ista_symphonic.pdf
Water meter in the picture above is on page 42.

Have the same experience - rtl dongle catch no frames (running for > 10h)

jacekjaros, 2022-02-08:
@weetmuts
could you confirm if rtl-sdr (in my case R820T2) can support R mode in the future release or hardware limitation don't allow that?

weetmuts, 2022-02-12:
@jacekjaros Sorry, to active R mode you have to transmit wmbus signals to the meter. The rtl-sdr dongles can only listen. So it will never work. 

@TychoWerner You can try a cheap rtl-sdr dongle first. The HCA meters are usually not R meters. What brand is it?

weetmuts, 2022-02-12:
@TychoWerner The RTL2832U  dongles seem to be pretty similar all of them. The aluminum rtl-sdr branded dongles are better quality though and can receive over longer distances. https://www.rtl-sdr.com/buy-rtl-sdr-dvb-t-dongles/

However when I read the technical specification of your HCA meter, it says bidirectional and encrypted.
So you will probably not be able to read it with an sdr-rtl also you need the key.

TychoWerner, 2022-02-12:
@weetmuts Is it possible to brute force such a key? Willing to let it run to try and get it. How are those keys normally gotten?

weetmuts, 2022-02-12:
@TychoWerner No, it is 128 bit AES keys. Assuming they follow wmbus protocol standard. We

weetmuts, 2022-02-12:
@TychoWerner Either you buy the meter yourself, or in the case of water meters, you can ask the water company for they key. Often they will provide it, but not always.

TychoWerner, 2022-02-12:
> @TychoWerner Either you buy the meter yourself, or in the case of water meters, you can ask the water company for they key. Often they will provide it, but not always. 

@weetmuts are they keys the same for the same model? They are installed by my landlord company

weetmuts, 2022-02-13:
A proper setup has a unique key per meter. Then there is everything in between a single key for all meters. A single key that is known and no key at all. 

But why do you want to listen to your landlords HCA? The HCA:s reporting is just the temperature difference between the rooms air temperature and the radiators temperature. Used to spread the total cost of heating the house over the tenants. You can almost get the same measurement by feeling the radiator with your hand. ;-)

TychoWerner, 2022-02-13:
> A proper setup has a unique key per meter. Then there is everything in between a single key for all meters. A single key that is known and no key at all.
>
> But why do you want to listen to your [...]

I wish to measure the amount of units, that it also displays on the screen. So I can keep track of how much I've used compared to last year with each temperature. They create the yearly invoice so I hope to see where the costs went.

weetmuts, 2022-02-13:
Aha! Yes that makes sense. Well, I suggest first buying a cheap rtsldr and listen to the radio traffic with rtl_433 for a while to see what is being sent and when (ie let it run for a month or so). This is fun anyway. If you see wmbus traffic from the meter we can analyze it to see if it looks like bidirectional or unidirectional. Then you can ask the landlord if the telegrams are encrypted.

TychoWerner, 2022-02-13:
> Aha! Yes that makes sense. Well, I suggest first buying a cheap rtsldr and listen to the radio traffic with rtl_433 for a while to see what is being sent and when (ie let it run for a month or so). [...]

@weetmuts 
Is this rtl sdr any good? https://www.aliexpress.com/item/1005002527097442.html
It has "quick" 10 day shipping so I would be able to start sooner.
Any other things I need?
 I got a Intel NUC with Proxmox so I can spin up a Linux box to run the rtl_433 tool

weetmuts, 2022-02-13:
@TychoWerner How would I know? :-) Looks similar to all other rtl-sdr dongles.

TychoWerner, 2022-02-26:
> Aha! Yes that makes sense. Well, I suggest first buying a cheap rtsldr and listen to the radio traffic with rtl_433 for a while to see what is being sent and when (ie let it run for a month or so). [...]

What command would you recommend for rtl_433, I've been trying 
`rtl_433 -S unknown`
and I got some responses in a few minutes:
`time      : 2022-02-25 18:37:07
model     : Conrad-S3318P ID        : 207
Channel   : 1            Battery   : 1             Temperature: 42.10 F      Humidity  : 67 %          Button    : 0
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-02-25 18:37:26
model     : Nexus-T      House Code: 196
Channel   : 1            Battery   : 0             Temperature: 7.30 C
 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _`

weetmuts, 2022-02-26:
Ok, now you have to listen at the proper frequency and for a long time. You have to capture two messages, one sent to the meter to trigger readout and one from the meter when it is being read. This might take some time since they probably only scan the meters once per month.

According to one wmbus dongle documentation, there are ten channels for R2, 868.03, 868.09, 868.15, 868.21, 868.27, 868.33, 868.39, 868.45, 868.51, 868.57

@merbanan How many channels will be caught by rtl_433 if we run it like this?
rtl_433 -f 868.27M  

Also I assume it is the standard wmbus format, so the normal wmbus trigger should work?

TychoWerner, 2022-02-27:
> Ok, now you have to listen at the proper frequency and for a long time. You have to capture two messages, one sent to the meter to trigger readout and one from the meter when it is being read. This [...]

Should the WMBus addon in HomeAssistant be able to capture required data? Got the default settings except device set to auto and MQTT is setup. 

Log so far:
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
Running wmbusmeters ...
Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!`

merbanan, 2022-02-27:
rtl_433 has by default a 1MS rate when listening on 868.27M. It translate roughly to +- 500kHz of range with probably 600kHz wide range of good reception. I have never measured this exactly. You can if you want increase the sample rate which should also widen the frequency range. Beside this there are some new options in newer version of rtl-sdr that can fine tune the reception also.

In totalt it is very hard to say something more concrete then that by placing the tuned frequency between the min and max channel and increasing the sample rate will eventually cover 2 different channels. In this case we have .57 and .03. With a channel spacing of 0.06 (+-0.03) we get a total range of 0.6MHz. Placing the tuning frequency in the middle should will unfortunately not work because of the center spike so you need to increase the sample rate to 1600ks and then offset tune to 868.7MHz. This should then cover the needed range in the bottom range half.

TychoWerner, 2022-02-28:
> Should the WMBus addon in HomeAssistant be able to capture required data? Got the default settings except device set to auto and MQTT is setup.
>
> Log so far: `[s6-init] making user provided files [...]

Seems the addon is able to capture some data thru the antenna so that's a good sign:
`Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
(wmbus) Unknown tpl-ci-field 80
Received telegram from: 84546279
          manufacturer: (ISK) Iskraemeco, Slovenia (0x266b)
                  type: Electricity meter (0x02)
                   ver: 0x50
                device: rtlwmbus[00000001]
                  rssi: 20 dBm
                driver: unknown!
(wmbus) Unknown tpl-ci-field 6c
Received telegram from: 84546279
          manufacturer: (ISK) Iskraemeco, Slovenia (0x266b)
                  type: Electricity meter (0x02)
                   ver: 0x50
                device: rtlwmbus[00000001]
                  rssi: 21 dBm
                driver: unknown!
(wmbus) Unknown tpl-ci-field 80
Received telegram from: 84546279
          manufacturer: (ISK) Iskraemeco, Slovenia (0x266b)
                  type: Electricity meter (0x02)
                   ver: 0x50
                device: rtlwmbus[00000001]
                  rssi: 21 dBm
                driver: unknown!`

slu125, 2022-10-05:
Sorry to bump this up, but did anybody make any progress with the R mode? I have an optosonic u 3 radio net from Ista and would love to get the readings of my heat meter as live as possible. https://www.ista.com/fileadmin/twt_customer/countries/content/Arab/Documents/Optosonic_u_3_radio_net.pdf

[image]

hackex, 2022-12-28:
> Sorry to bump this up, but did anybody make any progress with the R mode? I have an optosonic u 3 radio net from Ista and would love to get the readings of my heat meter as live as possible. [...]

I am also looking for a way to read the values from an ISTA Heizkostenverteiler on my apartment's heating radiators.

danielkucera, 2022-12-28:
There is an attempt here: https://wiki.munichmakerlab.de/wiki/Smart_Meter_Hacking
But without any success so far.

schildbach, 2023-01-21:
I wonder if it's possible to get inbetween the IR interface, between the radio and the meter/calculator. Would that link be AES encrypted as well?

merbanan, 2023-01-21:
It should not be unless the payload over IR is wmbus. But you need to test to be sure.

pbasista, 2023-05-12:
I would like to point out that the [Ista symphonic sensor net radio system](https://www.ista.com/fileadmin/twt_customer/countries/content/Tutorial/Documents/ista_symphonic.pdf) prospectus, linked to earlier by @jacekjaros, states, e.g. on page 7, that the transfer rate of the Ista radio net devices is ~90 kbit/s. According to the [Telit Wireless M-Bus 2013 Part 4 User Guide](https://www.iot.com.tr/uploads/pdf/Telit_Wireless_M-bus_2013_Part4_User_Guide_r14.pdf), page 11, the transfer rate of wM-Bus devices in R2 mode is 4.8 kbit/s.

It is possible that Ista radio net devices use the wM-Bus in R2 mode while idle. Then, after they detect a wake-up frame, they might perform some light signaling over the R2 mode and switch to some other, higher-speed wM-Bus mode for the actual data transfer.

But it seems also plausible to me that they might, in theory, use a completely different protocol other than wM-Bus.

hanusek, 2024-02-15:
Do you have a manual for ista meter?
Ista meter have a button. Would you happen to know why?

Tobiaskquest, 2024-08-04:
Hello everyone.
What command can I use to send a telegram to a meter (wakeup) via Terminal/wmbusmeters?
Thanks in advance for the answer.

weetmuts, 2024-08-04:
@Tobiaskquest what meter? which wmbus-mode? which frequency? which protocol? :-)

Tobiaskquest, 2024-08-04:
Example for this water meter ISTA or perhaps Sensus.
WMBUS mode maybe T,C,S,R. (Which of these can be used for sending telegrams... T2,C2,S2 or even R2?)
For example, with this ISTA Radio Net 3 I measured a frequency of 868.87 with my frequency counter when I pressed the button there.
Unfortunately I don't know the exact protocol yet but maybe a standard protocol for WMBUS.
Perhaps you could just show me a few commands as an example so that I at least have an idea of ​​what something like that might look like. I would like to buy one of these ISTA Radio Net 3 modules (to open it) soon and try out a little bit what kind of test equipment/software I can use to get which information. Baud rate will be 9600.
Thank you very much

weetmuts, 2024-08-05:
@Tobiaskquest Your question is a bit difficult to answer. Most wmbus meters just transmit regularly. No need to wake them up. Start listening and wait until you hear the telegram.
It can be as often as every 15 seconds or once per day.  Sometimes they do not transmit on weekends or at night.

There is support in wmbusmeters to do C2,T2,S2 ie polling of meters using hardware dongles like im871a and amb8365 that can transmit telegrams. However it has not yet been tested/fully implemented because no-one has requested it yet. This code could be used to wake-up a meter as well, if you know what kind of telegram should be used.

Tobiaskquest, 2024-08-06:
Thank you.
I have the Amber 8465-m Stick and a CULW868 and i buy a im871a too.
Do i need two USB Dongles for polling ?
Is it possible to send in R2 with wmbusmeters?
Thanks again.

PS: Does amyone know of other Tools/Software to send telegrams in T2,C2,S2,R2 ? 
Thanks

Tobiaskquest, 2024-08-18:
Question. I bought 2 Radio Net 3 modules. Are any of you @weetmuts, @merbanan interested in me sending one of you a module as a gift to test? It won't be parameterized yet though. I think so.

For information about what I've done so far:
What I've tested with the built-in modules is with a frequency measuring device. After a few seconds it shows a fixed 868.75 MHz and also fluctuates at around 169.5 MHz and other Frequecys. As if it's changing frequencies.
I've also tried it with rtl_433 and "-S unknown" command. I've converted a few files from .cu8 to .sr and looked at them in PulseView. I can see Datas but unfortunately I can't do anything with it. But I also have too much interference from lots of other devices, smoke detectors, etc. I've also tried an RTL Nooelec Nesdr Stick and GQRX on Fedora 40. I have nothing at 169 + - MHz, but when I try 868.33 MHz and press the button on the water meter module, after about 8 seconds I get a beep and a reading. The manufacturer also has a protocol description, but I think it's only for MBus cables. I don't know if anyone here can derive anything from it? Here is the link. Unfortunately only in German.

ista
https://www.ista.com › FunkPDF
Protokoll der M-Bus-Schnittstelle für das modul mbus

Tobiaskquest, 2024-08-20:
I know this is about wmbusmeters, but also about ISTA Radio Net 3 module, or should I reopen this under RTL_433 as a question to @merbanan ?

I noticed something strange, when I use the following command for rtl433, >rtl_433 -f 868.33M -s 180k -g "" -Y auto< and intentionally set the sample rate below 1000k, then every few seconds this appears irregularly:
Invalid sample rate: 180000 Hz
[sdr_set_sample_rate] Failed to set sample rate.
[pulse_slicer_pcm] sample rate too low for protocol 185 "Honeywell CM921 Wireless Programmable Room Thermostat"
[pulse_slicer_pcm] sample rate too low for protocol 185 "Honeywell CM921 Wireless Programmable Room Thermostat"
[pulse_slicer_pcm] sample rate too low for protocol 185 "Honeywell CM921 Wireless Programmable Room Thermostat"
[pulse_slicer_pcm] sample rate too low for protocol 185 "Honeywell CM921 Wireless Programmable Room Thermostat"
[pulse_slicer_pcm] sample rate too low for protocol 185 "Honeywell CM921 Wireless Programmable Room Thermostat"

If I now press the button on the Radio Net 3 module until the red LED flashes, then whenever the LED flashes,
a new line appears immediately with "[pulse_slicer_pcm] sample rate too low for protocol 185 "Honeywell CM921 Wireless Programmable Room Thermostat". When the LED stops, the line output also stops. Strange, right? What could that be?
Can the Radio Net 3 signal trigger other transmitters? I don't have a Honeywell CM921 Wireless Programmable Room Thermostat in the house either.

merbanan, 2024-08-20:
Invalid sample rate: 180000 Hz, use a sample rate that does not give that error.

Tobiaskquest, 2024-08-20:
OK. I'll test them all. It might take a while.

BinaryBrain86, 2025-01-24:
Hi @all,
did anyone managed to read frames from optosonic u 3 radio net connected with this Ista ultego 3 smart?

[image]

[image]