wmbusmeters/wmbusmeters #437: How to force the use of a particular driver via a meter file in wmbusmeters.d? fhkvdataiii and not fhkvdataiv

MaxPeal, 2022-01-04:
How to force the use of a particular driver via a meter file in wmbusmeters.d?

MaxPeal, 2022-01-04:
or can i trigger the automatic switching manually? 
https://github.com/weetmuts/wmbusmeters/issues/206#issuecomment-755607430

weetmuts, 2022-01-05:
Add:
driver=fhkvdataiii 
as explained in the README. :-)

weetmuts, 2022-01-05:
But you are right, there is a problem here. The fix/workaround in #206 no longer works since wmbusmeters will instantiate a single driver c++ object for the first telegram, and if that is the wrong type, then the c++ object will stay the wrong type. Previously wmbusmeters switch c++ object every time a new telegram arrived, but no longer.

Hmmm.

weetmuts, 2022-01-13:
Can you please confirm if adding driver=fhkvdataiii to your meter file solves the problem?

closed 2022-02-26 by weetmuts

nanokatz, 2023-11-04:
I think I have the same problem. I can only see C1 messages for the iv protocol. Even setting the rtl to be only in T mode I still cannot see a meter with the IDs from the meters.