wmbusmeters/wmbusmeters #438: Detailed logging

raszuk, 2022-01-05:
Hi, 

When no meters are configured wmbus log contains very useful meter descriptions for new discovered meters:

Received telegram from: 16167040
          manufacturer: (BMT) BMETERS, Italy (0x9b4)
                  type: Water meter (0x07)
                   ver: 0x13
                device: rtlwmbus[77771111153705700]
                  rssi: 47 dBm
                driver: hydrodigit
telegram=||4E44B4094070161613077A3A0040054CBC456D03B5E874044AE2C191292F2896D15BD03F405EB9738731DC6D50AB0EEA16B19CCEEA01C505E7F808E3947353928B385E89ABC976DAF01E83BC21F1AD|+2513

However when at least one meter is setup the output written into log file turns very cryptic for new not configured meters:

CONFIGURED: 

(wmbusmeters) started meter 3 (lok10-cw 16167040 auto)
telegram=|4E44B4094070161613077A3B0040052F2F|0C1343180000046D0F31C5210F152E0B0000009A00009A0000A10000A10000A10000A10000A10000A70000A70000A70000B80000B80000002F2F2F2F2F2F|+52
(wmbusmeters) started meter 4 (lok35cw 16167001 auto)
telegram=|4E44B4090170161613077A370840052F2F|0C1350410000046D2B31C5210F172E192425080000009F01009F01009F01009F01009F01009F01009F01009F01009F01009F01009F01009F0100002F2F2F|+56

NOT CONFIGURED METER: 

telegram=|4E44010616094021420C7A340040252F2F|0C0E000000000C13000000004C0E00000000426CBF2C0B3B0000000B2D0000000A5A88010A5E7201046D3430C52102FD1700002F2F2F2F2F2F2F2F2F2F2F|+66

Is there any switch to allow to write to log file the original format listing all details ? That way it is easy to discover new meters ... 

Many thx - Robert

BIBOLV, 2022-01-10:
Hi,
Set loglevel to debug and you will get all the details in logfile.

BIBOLV, 2023-01-01:
Closing due to inactivity.

closed 2023-01-01 by BIBOLV