wmbusmeters/wmbusmeters #439: Missing driver for Techem Radio converter

raszuk, 2022-01-05:
Hi, 

The Techem Meters for cold or hot water are nicely decoded as mkradio4 for cold and mkradio3 for hot water. 

But when we see radio converter driver is unknown: 

Received telegram from: 73813705
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
      Concerning meter: 71323304
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Water meter (0x07)
                   ver: 0xfe
                device: rtlwmbus[77771111153705700]
                  rssi: 13 dBm
                driver: unknown!
telegram=||3644685005378173953772043332712423FE073A002025DEB790B789A103146E061A402AA97B9F20C8D371DBD74F8EE6A5D56D6B727FA4|+2269

Received telegram from: 63613886
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Warm water (0x62)
                   ver: 0x74
                device: rtlwmbus[77771111153705700]
                  rssi: 25 dBm
                driver: mkradio3
telegram=||2F446850863861637462A2069F2B1F015002000000000002020202020302020302020201020103020302030202030301|+2271

Received telegram from: 73813705
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Cold water (0x72)
                   ver: 0x95
                device: rtlwmbus[77771111153705700]
                  rssi: 8 dBm
                driver: mkradio4
telegram=||2F446850053781739572A2069F2BFF08500203000000030C0C1115161513161715171F151A161816151513120D121312|+2274

Is this known and perhaps someone has solved this already ? 

Many thx, Robert

BIBOLV, 2022-01-10:
Hi,
For me it looks like converter is reporting with same ID as one of the meters and IMO it should not be like that as ID`s must be unique. Strange.

weetmuts, 2022-01-13:
That is odd indeed. Do you have two physical meters with the id numbers: 63613886 and 73813705?
What do they look like? Can you post a picture here?

raszuk, 2022-01-13:
The telegrams in question come from a different building so I can not make a picture of it. The reason I am asking is that they have very good radios and I am going to evaluate installation of about 300 of them in my building. 

Btw another interesting things I noticed in the log when grepping for "ignored" is that sometimes two of those meters come as tuple. Not sure if this is expected or not.  Of course ignored means not configured in meter file, but normally I see only one meter number in each line. 

robert@nuc1eo:/var/log/wmbusmeters/meter_readings$ grep ignored ../wmbusmeters.log | sort -u
(wmbus) telegram from 000650c2 ignored by all configured meters!
(wmbus) telegram from 0009b0a5 ignored by all configured meters!
(wmbus) telegram from 000c15a4 ignored by all configured meters!
(wmbus) telegram from 000cc5df ignored by all configured meters!
(wmbus) telegram from 000d1ad3 ignored by all configured meters!
(wmbus) telegram from 000d6321 ignored by all configured meters!
(wmbus) telegram from 000e5223 ignored by all configured meters!
(wmbus) telegram from 000edd0f ignored by all configured meters!
(wmbus) telegram from 000f7fef ignored by all configured meters!
(wmbus) telegram from 00331729 ignored by all configured meters!
(wmbus) telegram from 03952680 ignored by all configured meters!
(wmbus) telegram from 63613881 ignored by all configured meters!
(wmbus) telegram from 63613884 ignored by all configured meters!
(wmbus) telegram from 63613886 ignored by all configured meters!
(wmbus) telegram from 63613985 ignored by all configured meters!
(wmbus) telegram from 73810689,59571501 ignored by all configured meters!
(wmbus) telegram from 73810689 ignored by all configured meters!
(wmbus) telegram from 73813701,71323291 ignored by all configured meters!
(wmbus) telegram from 73813701 ignored by all configured meters!
(wmbus) telegram from 73813704 ignored by all configured meters!
(wmbus) telegram from 73813705,71323304 ignored by all configured meters!
(wmbus) telegram from 73813705 ignored by all configured meters!
(wmbus) telegram from 73813831,71323227 ignored by all configured meters!
(wmbus) telegram from 73813831 ignored by all configured meters!
(wmbus) telegram from 73813836,71323239 ignored by all configured meters!
(wmbus) telegram from 73813836 ignored by all configured meters!
(wmbus) telegram from 73813849,71323249 ignored by all configured meters!
(wmbus) telegram from 73813849 ignored by all configured meters!
(wmbus) telegram from 73813850,71323064 ignored by all configured meters!
(wmbus) telegram from 73813850 ignored by all configured meters!
(wmbus) telegram from 73813859,71323037 ignored by all configured meters!
(wmbus) telegram from 73813859 ignored by all configured meters!
(wmbus) telegram from 73813871,59344879 ignored by all configured meters!
(wmbus) telegram from 73813871 ignored by all configured meters!
(wmbus) telegram from 73813877,59344877 ignored by all configured meters!
(wmbus) telegram from 73813877 ignored by all configured meters!
(wmbus) telegram from 73813882,59344875 ignored by all configured meters!
(wmbus) telegram from 73813882 ignored by all configured meters!
(wmbus) telegram from 73813885,59344885 ignored by all configured meters!
(wmbus) telegram from 73813885 ignored by all configured meters!
(wmbus) telegram from 73813890,59344878 ignored by all configured meters!
(wmbus) telegram from 73813890 ignored by all configured meters!
(wmbus) telegram from 73813901,71323251 ignored by all configured meters!
(wmbus) telegram from 73813901 ignored by all configured meters!
(wmbus) telegram from 73813955,59368310 ignored by all configured meters!
(wmbus) telegram from 73813955 ignored by all configured meters!
(wmbus) telegram from 73815158,59369394 ignored by all configured meters!
(wmbus) telegram from 73815158 ignored by all configured meters!
(wmbus) telegram from 73815448,59302612 ignored by all configured meters!
(wmbus) telegram from 73815448 ignored by all configured meters!
(wmbus) telegram from 73815459,59302619 ignored by all configured meters!
(wmbus) telegram from 73815459 ignored by all configured meters!
(wmbus) telegram from 74417313,59740938 ignored by all configured meters!
(wmbus) telegram from 74417313 ignored by all configured meters!
robert@nuc1eo:/var/log/wmbusmeters/meter_readings$

weetmuts, 2022-01-13:
You can have up to three ids in a wmbus telegram representing data link, relay and application. 
wmbusmeters can match on any of these, so "telegram from 73815459,59302619 ignored by all configured meters!"
means that a telegram with two ids has been ignored.

weetmuts, 2022-01-13:
Can you log --logtelegrams for an couple of hours or so and send me the log.

raszuk, 2022-01-13:
I am logging it already :)

***@***.***:/etc$ more wmbusmeters.conf
loglevel=debug
#loglevel=normal
#device=cul:868.95M:t1
device=rtlwmbus:868.95M:t1
#device=auto:t1
#device=/dev/ttyUSB0:38400
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
#meterfilesaction=overwrite
meterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
***@***.***:/etc$

Here is a pointer to recent logfile with debug level:

https://drive.google.com/file/d/1Mb3993MEUQzMnm9y0T53UTdSHpnmg97z/view?usp=sharing

Thx,
R.

merbanan, 2022-02-06:
This is most likely a repeater. There should be a bit about that somewhere.

closed 2023-02-19 by github-actions[bot]