wmbusmeters/wmbusmeters #440: Add support for AMB8665-M

pascaltippelt, 2022-01-07:
Hi!

I do own a AMB8465 and a AMB8665-M wireress m-Bus Stick. As far as I can see from the public available datasheets, their way to communicate with a pc are equal. The difference seems to be the existence of a SMA antenna on the AMB8665-M and some internal changes.

As the software-interface is identical between AMB8626, AMB8465 and AMB8665, could you please add support for all those devices? Or maybe allow to force-detect a tty to a device? Currently if I do /dev/ttyUSB2:amb8465:c1,t1 my device is not recognised.

Best regards

Pascal

weetmuts, 2022-01-11:
Can you run "wmbusmeters --debug /dev/ttyUSB2" and post the log here?
This will list all communication with the dongle and we can see how it responds when wmbusmeters probes it.
Do you have a link to some documentation, can you see what the default serial speed is?

pascaltippelt, 2022-01-11:
Hi!

Thank you for your reply. Here some more information:

```
lsusb
Bus 002 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
```
Just in case you needed this. Now the log:

```
wmbusmeters --debug /dev/ttyUSB0:amb8465:c1
(wmbusmeters) version: 1.4.1-62-g1f42183
(config) using device: /dev/ttyUSB0:amb8465:c1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0"
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(amb8465) sent 6 bytes OK
(amb8465) reading response... 1
(serial) received binary "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFF"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb8465) reading response... 2
(serial) received binary "FF000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE6"
(amb8465) proprely decoded ConfigAMB8465 response. Content: id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) found response at offset 0
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(amb8465) detect id=00000000 media=00 version=00 c_field=44 auto_rssi=00
(amb8465) are you there? yes 00000000
(main) opening /dev/ttyUSB0:amb8465:c1
Started config amb8465[00000000] on /dev/ttyUSB0 listening on c1
(amb8465) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (amb8465)
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF8A7A00780080"
(amb8465) checkAMB8465Frame "FF8A7A00780080"
(amb8465) not enough bytes yet, partial command response 7 126.
(serial) received binary "710200000000FFFFFA00FFFFFFFFFF"
(amb8465) chunk time 18 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 22 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFF0C32000214"
(amb8465) chunk time 13 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C32000214"
(amb8465) not enough bytes yet, partial command response 38 126.
(serial) received binary "00FFFFFFFFFF010004000000FFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200021400FFFFFFFFFF010004000000FFFFFF"
(amb8465) not enough bytes yet, partial command response 53 126.
(serial) received binary "01440000000000000000FFFF0B060100"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100"
(amb8465) not enough bytes yet, partial command response 69 126.
(serial) received binary "FFFFFFFFFF00030000FFFFFFFFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0C3200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B060100FFFFFFFFFF00030000FFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 84 126.
[... 19 lines cut ...]
(amb8465) checkAMB8465Frame "FF8401007A"
(amb8465) received full command frame
(amb8465) set link mode completed
(amb8465) set link mode response "0100"
(wmbus) no alarm (expected activity) for amb8465
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 amb8465
(serialtty) closed /dev/ttyUSB0 (amb8465)
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted amb8465
(serial) closing 1 device
```

I hope this helps!

Best regards

Pascal

weetmuts, 2022-01-13:
Looks good to me. It sends commands to the dongle and gets the expected answers. I think amb8665 speaks the same protocol. When you run it like this, does it not work for you?

Can you also try without the amb8465 spec as well, like this:
wmbusmeters --debug /dev/ttyUSB2

pascaltippelt, 2022-01-13:
```
wmbusmeters --debug /dev/ttyUSB0
Wmbus devices found but no meters supplied. You must supply which link modes to listen to. Eg. auto:c1
```
```
wmbusmeters --debug /dev/ttyUSB0:auto:c1
Not a valid device "/dev/ttyUSB0:auto:c1"
```

But...
```
wmbusmeters /dev/ttyUSB0:amb8465:c1,t1
Started config amb8465[00000000] on /dev/ttyUSB0 listening on c1,t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: XXXXXXXX
          manufacturer: (BFW) BFW Buero fuer Waermemesstechnik OHG, Germany (0x8d7)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x02
                device: amb8465[00000000]
                  rssi: -57 dBm
                driver: bfw240radio
```
... works perfectly. So I think it is only a "problem" in detecting the AMB8665 / AMB8665-M as usable device.

weetmuts, 2022-01-13:
Ah, try:
wmbusmeters --debug /dev/ttyUSB0:c1,t1

weetmuts, 2022-01-13:
And you can try:
wmbusmeters auto:c1,t1

pascaltippelt, 2022-01-16:
> Ah, try:
> wmbusmeters --debug /dev/ttyUSB0:c1,t1

Works nicely.

> And you can try: wmbusmeters auto:c1,t1

Works too

GekoCH, 2022-08-03:
hmm I also have the AMB8665-M but can't get it running.
`wmbusmeters --debug /dev/ttyUSB0:c1,t1`
```
(wmbusmeters) version: 1.8.0-7-g78990c4
(config) using device: /dev/ttyUSB0:c1,t1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" unknown
You have to specify the expected device type for the tty /dev/ttyUSB0
(serial) stopping manager
```
The device is recognized on USB0

```
$wmbusmeters --verbose /dev/ttyUSB0:amb8465:c1,t1
returns:
(wmbusmeters) version: 1.8.0-7-g78990c4
(config) using device: /dev/ttyUSB0:amb8465:c1,t1
(config) number of meters: 0
(serial) tty /dev/ttyUSB0 can be accessed
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
Expected amb8465 on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
Notice! Other wmbusmeters (pid 745) is running and it might hog any wmbus devices.
No meters configured. Printing id:s of all telegrams heard!

```

weetmuts, 2022-08-04:
Do you see the warning:
```
Notice! Other wmbusmeters (pid 745) is running and it might hog any wmbus devices.
No meters configured. Printing id:s of all telegrams heard!
```
Is the daemon running? Stop the daemon: `sudo systemctl stop wmbusmeters` 
then kill 745. Try again.

GekoCH, 2022-08-04:
thx but still:
```
(wmbusmeters) version: 1.8.0-7-g78990c4
(config) using device: /dev/ttyUSB0:amb8465:c1,t1
(config) number of meters: 0
(serial) tty /dev/ttyUSB0 can be accessed
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
Expected amb8465 on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

```

weetmuts, 2022-08-04:
Can you do:
```
wmbusmeters --debug /dev/ttyUSB0:amb8465:c1,t1
```
and post the result here.

GekoCH, 2022-08-04:
```
(wmbusmeters) version: 1.8.0-7-g78990c4
(config) using device: /dev/ttyUSB0:amb8465:c1,t1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" amb8465
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
Expected amb8465 on /dev/ttyUSB0 but did not find it! Ignoring tty!
(serial) EVENT thread interrupted
(verbose) ignoring device c1,t1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1876 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1873
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1877 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
```

weetmuts, 2022-08-04:
Seems like there is nothing on this usb port that responds to the probe
message.

Unplug replug the amber dongle. check dmesg to see which usb port it gets
connected to. Try different bps rates, if the dongle has been changed from
the default speed of 9600 bps.

GekoCH, 2022-08-04:
ok I will try using another baudrate. 
Is this correct:
`wmbusmeters --debug /dev/ttyUSB0:115200:c1`

weetmuts, 2022-08-04:
Yep, that is where you set the serial speed.

weetmuts, 2022-08-28:
Ping @GekoCH

MikeTheTux, 2022-09-25:
Hi!

I do own a AMB8465 and a AMB8665-M wireress m-Bus Stick. Both run into the same problem:

```
/wmbusmeters # ./wmbusmeters  --debug  /dev/ttyUSB3:amb8465:t1
(wmbusmeters) version: 1.9.0
(config) using device: /dev/ttyUSB3:amb8465:t1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB3" amb8465
(serial) check if /dev/ttyUSB3 can be accessed
(serial) tty /dev/ttyUSB3 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB3 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(107,229,47,99,251,27,127,25) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB3 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(107,229,47,99,16) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB3 fd 3 (detect amb8465/amb3665)
(serial) received binary ""
(amb8465/3665) sending 6 bytes attempt 0
(serial /dev/ttyUSB3) sent "FF0A02008077"
(amb8465/3665) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465/3665) reading response... 1
(serial) received binary "FF8A7A00780080710200010000FFFFFF00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF0108"
(amb8465) not enough data for decoding ConfigAMB8465!
(amb3665) not the right header decoding ConfigAMB3665!
(amb8465/3665) reading response... 2
(serial) received binary "0000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF19"
(amb8465) proprely decoded ConfigAMB8465 response. Content: uart_cmd_out_enable=01 id=00000000 media=00 version=00 c_field=44 auto_rssi=01 b1_add_disable=01
(amb8465) found response at offset 0
(serialtty) closed /dev/ttyUSB3 (detect amb8465/amb3665)
(amb8465) detect uart_cmd_out_enable=01 id=00000000 media=00 version=00 c_field=44 auto_rssi=01 b1_add_disable=01
(amb8465) are you there? yes 00000000
(main) opening /dev/ttyUSB3:amb8465:t1
Started config amb8465[00000000] on /dev/ttyUSB3 listening on t1
(amb8465) on /dev/ttyUSB3
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB3 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(144,229,47,99,251,27,127,25) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB3 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(144,229,47,99,16) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB3 fd 3 (amb8465)
(amb8465) get config
(serial /dev/ttyUSB3) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF8A7A00"
(amb8465) checkAMB8465Frame "FF8A7A00"
(amb8465) not enough bytes yet, partial command response 4 126.
(serial) received binary "780080710200010000FFFFFF00FFFF"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFF00FFFF"
(amb8465) not enough bytes yet, partial command response 19 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFF00"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFF00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00"
(amb8465) not enough bytes yet, partial command response 34 126.
(serial) received binary "3200021400FFFFFFFFFF010004000000"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFF00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000"
(amb8465) not enough bytes yet, partial command response 50 126.
(serial) received binary "FFFFFF01440000000000000000FFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFF00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF"
(amb8465) not enough bytes yet, partial command response 65 126.
(serial) received binary "0B040100FFFFFFFFFF01080000FFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFF00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF01080000FFFFFF"
(amb8465) not enough bytes yet, partial command response 81 126.
(serial) received binary "FFFFFFFF0000FFFFFFFFFFFFFFFFFF"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF8A7A00780080710200010000FFFFFF00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF01080000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 96 126.
(serial) received binary "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[... 495 lines cut ...]
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB3 amb8465
(serialtty) closed /dev/ttyUSB3 (amb8465)
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted amb8465
(serial) closing 1 devices
/wmbusmeters #
```

I guess these are the critical lines:
```
(amb8465) unhandled device message 3
(amb8465) unknown response "4F44A5115317635370077AFF004005971D2833ED21416977E1DBA91660D1453BCF14F96ED309651DE0F40F0F8065FD612CC04EADC19244B19EC3B24F3E98863950F04F412294C445260EA6904D4C3B61"
^C(serial) stopping manager
```

wmbusmeters is running witin a docker.
I would expect WMbus Frames from a Diehl HYDRUS (hydrus) and a Siemens TD-3511.

Thanks for your support
mike

weetmuts, 2023-02-15:
@MikeTheTux 

>I guess these are the critical lines:
>(amb8465) unhandled device message 3
>(amb8465) unknown response "4F44A5115317635370077AFF004005971D2833ED21416977E1DBA91660D1453BCF14F96ED309651DE0F40F0F8065FD612CC04EADC19244B19EC3B24F3E98863950F04F412294C445260EA6904D4C3B61"
>(serial) stopping manager

Yes, so what is happening here is that wmbusmeters has sent a command to the dongle, but the response
is a just received telegram!  Wmbusmeters gives up. It should be possible to detect if it is an incoming telegram and discard it and continue waiting for the command response (or resend the command request if it was dropped by the dongle when it was receiving the telegram).

MikeTheTux, 2024-03-09:
The problem occurs always. I never got my Amber sticks working. Changed to Imst as the only possible workaround...

Tom1000, 2024-05-23:
Coming back to the original subject of this Issue (support for **AMB8665-M**):
I can confirm that this device (don't know the exact firmware, bought it 05/2024) works out-of-the-box with wmbusmeters.

With wmbusmeters it identifies as amb8465 and works fine in auto mode. I just tested it with version `1.13.1-3` in t1 and c1 and "t1,c1" combi mode. In all modes it receives telegrams when an antenna is connected properly.