wmbusmeters/wmbusmeters #443: Add support for SENSUS (S)COUT

cedsib, 2022-01-09:
Hello,

I set up all the project but i'm not detecting my Watermeter (which is a  Sensus Scout, see the picture), seems to be [this model](https://www.batiproduits.com/fiche/produits/systeme-de-radio-releve-des-compteurs-d-eau-en-t-p69110158.html), communicating in 868Mhz.

Here is the picture :

[image]

I can recieve data from another Watermeter (I thought it was mine at first), but with a total different data (143m3 of water vs 556m3 on mine).

Happy to test any new release :-)

Thanks !

BIBOLV, 2022-01-10:
Hi,
I tried to do a quick search and could not find any usable documentation of this meter, so you will need to do a research and find out why your meter is not reporting - from the looks it`s muddy and wet so maybe it dead. I guess it is property of your water company so I would suggest to start with them. When you will be able to receive some telegrams from it either unencrypted or you will get a key and be able to decrypt them then post them here and we will see if we can get further from there.

cedsib, 2022-01-11:
**Thanks for your prompt answer :-)**

👍  Good news I may have found my meter
👎 Seems encrypted and I don't know how to find the key.

**Meter found :**
`Received telegram from: 72901162
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x7b
                device: rtlwmbus[00000001]
                  rssi: 45 dBm
                driver: dme_07`

When I try with one frame from debug I get the same :
`wmbusmeters 3e44a511621190727b077a9a00300533f7a8558a980791fff6d5affff1e9ff4e37078db3ff365fe07767fdd2baae266365845b789dedff1bd87c02d2ff451d
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 72901162
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x7b
                driver: dme_07`

**Telegrams excerpts :**

```
(serial) received ascii "T1;0;0;2022-01-11 13:15:56.000;36;63;72901162;0x3e44a511621190727b077a9b00300553aab85336c7ffffbc68b9ffdf1a1461cfffbcee46eaff10c6f0cc3bd2ce94d25fd4b2b6fa1c965771ff263ca7f93476<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:15:56.000;36;63;72901162;0x3e44a511621190727b077a9b00300553aab85336c7ffffbc68b9ffdf1a1461cfffbcee46eaff10c6f0cc3bd2ce94d25fd4b2b6fa1c965771ff263ca7f93476<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2022-01-11 13:16:08.000;36;38;72901162;0x1effa5ff621190727b077a9b0010056fff0cf9f7ff1213e01cf9baffff0bfb<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:16:08.000;36;38;72901162;0x1effa5ff621190727b077a9b0010056fff0cf9f7ff1213e01cf9baffff0bfb<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2022-01-11 13:16:20.000;39;38;FF9011FF;0x1e44a511ff1190ff7b07ff9bff10056fb80cf9f77a1213e01cf9baffff0bfb<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:16:20.000;39;38;FF9011FF;0x1e44a511ff1190ff7b07ff9bff10056fb80cf9f77a1213e01cf9baffff0bfb<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2022-01-11 13:16:33.000;39;37;FF901162;0x1e44a511621190ff7bff7a9b0010056fb8fff9f7ff1213e01cf9ba4e720bfb<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:16:33.000;39;37;FF901162;0x1e44a511621190ff7bff7a9b0010056fb8fff9f7ff1213e01cf9ba4e720bfb<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2022-01-11 13:16:41.000;15;14;72901166;0x1effff11661190727b07ff9a0010ffffffffffff8dffffffff0d2fff8869fe<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:16:41.000;15;14;72901166;0x1effff11661190727b07ff9a0010ffffffffffff8dffffffff0d2fff8869fe<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2022-01-11 13:16:46.000;41;45;72901162;0x1e44a51162119072ff077a9b0010056fb80cf9f77aff13e01cf9ba4e720bfb<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:16:46.000;41;45;72901162;0x1e44a51162119072ff077a9b0010056fb80cf9f77aff13e01cf9ba4e720bfb<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2022-01-11 13:17:01.000;32;50;72901162;0x1e44a511621190727b07ff9b0010056fb80cf9f77aff13e01cf9ffff720bfb<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2022-01-11 13:17:01.000;32;50;72901162;0x1e44a511621190727b07ff9b0010056fb80cf9f77aff13e01cf9ffff720bfb<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
```

🙏  **Do you know what are the steps to find the key ?** 🙏

BIBOLV, 2022-01-12:
Content indeed looks encrypted, so we are back to - who owns the meter. If it is your water company then you could try to contact them and ask if they can provide you the decryption key, from other users there is information of successful cases when they got the key and also unsuccessful ones, so depends of your luck. If meter is yours then you should have received key together with meter, or you can try to contact seller, maybe they will be able to provide it to you. 
AFAIK there is no other way to retrieve the key.

BIBOLV, 2022-01-12:
https://github.com/weetmuts/wmbusmeters/issues/446 is one of latest examples.

BIBOLV, 2023-01-02:
I did a quick googling and could not find any confirmation that Scout-S supports wmbus. Closing this issue for now. If you manage to capture unencrypted packages, please reopen issue and provide telegrams.

closed 2023-01-02 by BIBOLV