wmbusmeters/wmbusmeters #444: New config parameters for meters

nedyarrd, 2022-01-09:
I have idea for new configuration options for meters. 

1) initial value
2) invert direction

1) For me it can be used for old used water meters, that have initial value of some value i.e. m3 of water. Users that bought those and dont have special software and hardware to reset counters in wmbus meters have sometimes problems with i.e. integration of that meter int HASS. Because meter starts at value 777.77m3 ;-) not from zero as in mechanical part.
     real_vol = wmbusmeter_vol - initial_value;

2) i have reversed installed meter, that counts backwards ;-) Lets invert
    real_vol = -wmbusmeters_vol

BIBOLV, 2022-01-10:
Hi,
I have exactly same situation as you described in point 3. I used Template at Home Assistant side to get correct value of meter that I needed and it is really simple.
IMO wmbusmeters should do one thing great and that is deliver readings from devices and all other manipulations should be done at backend.
@weetmuts maybe voting should be implemented for future requests to get understanding what actually community and users want/need? Maybe such future is what many users are looking for after all.

weetmuts, 2022-01-11:
Yes, voting is a good idea @BIBOLV . Maybe we could also move to a standalone community forum. Using issues is maybe not the best way to deal with questions and general discussions.

I have been thinking of the purpose of wmbusmeters... :-) this is my current thinking... 

1) talk to the dongles (wmbus chips and soon mbus hardware)
2) Understand the wmbus and mbus protocols (and potentially other low level utility protocols like the HAN port on electricity meters in sweden.)
3) normalize measurements and add meter specific knowledge (like error bits and what the different values mean)
4) relay the values somewhere else. 

Normalize is good, since the wmbus protocol has tons of odd units and scales (maybe the thinking was to make it easier for the meter hardware to just report american gallons based on rotations of a propeller tuned for american gallons, instead of having to convert to litres or m3 in the meter itself.) Adding meter specific knowledge does not hurt either, since otherwise the end-user himself must guess what the error bits are etc. There are a limited number of meters so why not have them collected in wmbusmeters.

So, currently there are two ways to modify the json data, e.g. addconversions where we can add a transform to another unit in addition to the normalized unit, ie you have total_m3 then addconversions=l will add total_l for the consumption in litres. You can also static fields field_room=123 to all meters or a specific meter.

So, for the purpose of normalizing I could very well accept some calculations as well. In the spirit
of field_room=123 we could have calcfield_real_vol_m3=total_m3 - 456

The idea is simple, the implementation is not. How complex expressions should we accept? 
Should we have a complete formula parser, how should we handle the units in the expression, etc.

nedyarrd, 2022-01-14:
I think that implementing expresions is not the point, because all values are linear, so to get correct value in cases what we described is only two values needed and one sign.
y = a*x+b
Y - correct value
a - multiplier (can be with sign)
b - offset

That would simplify idea, to two variables maximum. That method is often used in electric meters, like we recieve invoice ;-)