wmbusmeters/wmbusmeters #445: No meters configured - Landis+Gyr ULTRAHEAT®T550

thaserge, 2022-01-10:
Hi! 
I need your help. 
There is an ULTRACHEAT heat meter I receive data from it through an external GSM modem.
Modem sends CSV files to FTP server.
With telegram inside
`68F8F86808007200464470A7320404270000000974040970040C0E082303000C14079519000B2D0500000B3B0808000A5B52000A5F51000A6206004C14061818004C0E490603000C7800464470891071609B102D020100DB102D0201009B103B6009009A105B78009A105F74000C22726701003C22000000007C2200000000426C01018C2006000000008C3006000000008C80100600000000CC200600000000CC300600000000CC801006000000009A115B64009A115F63009B113B5208009B112D020100BC0122000000008C010E490603008C2106000000008C3106000000008C811006000000008C011406181800046D310ACA210F21040010A0C116
`
I am planning to process this data using your utility

But she says she doesn't know anything about my meter
[image]

I would be grateful if you can help with the solution of my question

weetmuts, 2022-01-13:
I just pushed a first version for your meter, build the latest git and run:
./build/wmbusmeters --format=json 68F8F86808007200464470A7320404270000000974040970040C0E082303000C14079519000B2D0500000B3B0808000A5B52000A5F51000A6206004C14061818004C0E490603000C7800464470891071609B102D020100DB102D0201009B103B6009009A105B78009A105F74000C22726701003C22000000007C2200000000426C01018C2006000000008C3006000000008C80100600000000CC200600000000CC300600000000CC801006000000009A115B64009A115F63009B113B5208009B112D020100BC0122000000008C010E490603008C2106000000008C3106000000008C811006000000008C011406181800046D310ACA210F21040010A0C116  MyUltra auto 70444600 NOKEY

It will print:
{"media":"heat","meter":"ultraheat","name":"MyUltra","id":"70444600","heat_kwh":8974.444444,"volume_m3":1995.07,"power_kw":0.5,"flow_m3h":0.808,"flow_c":52,"return_c":51,"timestamp":"2022-01-13T09:12:12Z"}

There are a lot more data in this telegram, you can see it using this command:
./build/wmbusmeters --analyze 68F8F868.....telegram hex.....0010A0C116  MyUltra auto 70444600 NOKEY

See if you can add some more decoding in src/meter_ultraheat.cc for your telegram!

Hartislap, 2022-09-27:
Hi,

i have one of those L&G T550/UH50 Ultraheat and want to get it integrated into HomeAssistant. As wmbusmeters seems to be in active development I prefer using this instead libmbus...
As I use the wired mbus I am not sure how to get information, so maybe you can give me a hint what to do. :-)
I am using a M-Bus HAT on a raspy, the meter can be found by libmbus:
$ mbus-serial-scan-secondary -b 2400 /dev/serial0 
Found a device on secondary address 68386406A7320404 [using address mask 6FFFFFFFFFFFFFFF]

So what would be the step to get this working with wmbusmeters? I played a little bit arround but did not get anything:

$ wmbusmeters MBUS=/dev/serial0:mbus:2400
Started config MBUS=mbus on /dev/serial0
No meters configured. Printing id:s of all telegrams heard!

$ wmbusmeters --ppjson --format=json --logtelegrams MBUS=/dev/serial0:mbus:2400 UH50MBus auto:MBUSS:mbus 68386406 NOKEY
(meter) UH50MBus 68386406 needs polling but has no pollinterval set!
Started config MBUS=mbus on /dev/serial0

I do not get any response, so I am not sure if the parameters are completely wrong or something is simply not working...

Hartislap, 2022-09-27:
Perfect, thank you, got the first MQTT message in HA!

Message 110 received on wmbusmeters/68386406 at 5:20 PM:
{
    "media": "heat",
    "meter": "ultraheat",
    "name": "UH50MBus",
    "id": "68386406",
    "heat_kwh": null,
    "volume_m3": 15.11,
    "power_kw": 0,
    "flow_m3h": 0.001,
    "flow_c": 29,
    "return_c": 30,
    "timestamp": "2022-09-27T15:20:45Z"
}

Now I need better weather in Germany to get "usefull" values. :-D

BIBOLV, 2023-01-02:
Closing issue as it looks like provided solution is working, Please reopen issue if additional help is needed.

closed 2023-01-02 by BIBOLV