wmbusmeters/wmbusmeters #446: No luck with getting the kem file from the company.

nagydavid, 2022-01-11:
Dear All,
I would like to ask is there anyway to get the encryption key for the multical21 since my utility company is being a ass-hat, by not providing it, since they r afraid I will do something and alter the reading. 

Thanks in advance

BIBOLV, 2022-01-12:
Simple answer is - no, it is not possible. 
AES-128 is used to encrypt data and it would make many ages of universe to brute force and find correct key. You can read about it [here](https://crypto.stackexchange.com/questions/48667/how-long-would-it-take-to-brute-force-an-aes-128-key) and [here](https://crypto.stackexchange.com/questions/76738/has-aes-128-been-fully-broken) and also [this](https://www.compass-security.com/fileadmin/Research/White_Papers/2013-01_wireless-mbus-security.pdf) is nice research about wmbus protocol as such.
Although there is [suggestion](https://zewaren.net/wmbus-izar-meter.html) that using KPA (known-plaintext attack) it is fairly simple to get the key, but I have not tested it so I can't confirm and to do that one would need to be savvy in many areas like coding, cryptography, wmbus protocol and specific meter details.

weetmuts, 2022-01-12:
Agreeing with @BIBOLV , no it is not possible to find the key for wmbus standard AES encrypted meters, like the multical21.

For some other meters that use their own homebrew encryption (which of couse is somewhat faulty) then there are situations when they can be decoded, which is described in the link above.

weetmuts, 2022-01-12:
@nagydavid You can try to explain to your water meter company that the security of mechanical water meters is that they monitor the sanity of your consumption and they verify the seals of the mechanical meter, and the meter itself, at least once every 10 years. 

The same is true for electronic wmbus meters, sanity check of consumption and check the seals and the consumption recorded by the meter itself when they replace it after 10 years. Otherwise someone can have bypassed the wmbus meter just as easily as the mechanical meter.

If you, through a lot of effort, have managed to transmit fake signals, with the right kind of transmitter strenght, exactly right protocol bits etc, while at the same time managed to shield the original meter, then you will be detected anyway when they come to replace the meter, since the original meter will show a different consumption. 

You can also let them know that many other users have received their KEM file to listen and monitor their own water consumption.

nagydavid, 2022-01-14:
@BIBOLV @weetmuts 

Thank you for the detailed answer from both of you, now I am trying to use the company where I work to initiate some collab for data exchange :)

closed 2022-01-14 by nagydavid