wmbusmeters/wmbusmeters #447: Sappel 169mhz

landaisbenj, 2022-01-16:
Hello. Did you know if I can help with this : 
[image]

I m not really good in hardware, but I can make a controller If I have plan.

weetmuts, 2022-01-16:
Hello @landaisbenj !  We can try, but first we need the telegram data. wmbusmeters is the decoder that takes the bytes and understands them, but it does not know how to decode the radio waves. This is done either by purchasing a dongle like im871a/amb8465/rc1800 which do decoding in hardware. or you buy a generic rtl-sdr dongle and decode the radio waves using software. Such decoding can be done with rtl_433 (https://github.com/merbanan/rtl_433)
or rtl_wmbus (https://github.com/xaelsouth/rtl-wmbus/)

So far I have not heard of anyone successfully decoding 169MHz wmbus signals (they are usually called N-mode)
right @merbanan? But both projecs are interested in adding decoding of these signales.
But you have to help them by buying an rtl-sdr dongle and then capture suitable radio traffic for analysis.

I have not heard of anyone using 169MHz either using commercial dongles.
But another user @nestorayso recommeded radiocrafts modules:

weetmuts, 2022-01-16:
You can also participate here: https://github.com/weetmuts/wmbusmeters/issues/10

merbanan, 2022-01-16:
The issue I had is that no one was able to receive a signal on the 169MHz. If I get proper signal samples I will be able to decode them with rtl_433. If you have a rtl-sdr dongle then you can try with rtl_433 -f 169M -S unknown.

landaisbenj, 2022-01-16:
Hmm alright. So if i understand, i need to listen with a rtl-sdr dongle or radiocrafts, decode with rtl_433 (or something else ?) and "translate" the result ("telegram"?) with wmbusmester ? 

I can buy a rtl-sdr, but why (because i'm a big nood on this) would i do better than other ?

I saw radiocrafts may be a good solution but i can't saw how to create a sniffer...

weetmuts, 2022-01-16:
You would do better than the others, if you are more persistent! ;-) 

It takes a couple of tries to capture suitable telegrams. The best tool is definitely rtl_433 since
it can auto-capture signals of intereset even though they cannot be decoded.
Run: rtl_433 -f 169M -S unknown.
for a week and upload the signals somewhere so that @merbanan can download them.

weetmuts, 2022-01-16:
> Hmm alright. So if i understand...

The pipeline looks like this:

wmbusmeters + im871a/amb8465/radiocraft hardware -> mqtt/curl somewhere
or
rtl_sdr -> rtl_wmbus -> wmbusmeters -> mqtt/curl/somewhere
or
rtl_sdr -> rtl_433 -> wmbusmeters -> mqtt/curl/somewhere

landaisbenj, 2022-01-16:
Ok, so if i buy [this](https://www.passion-radio.fr/recepteurs-sdr/rtl-sdr-r820t2-248.html), how can listen vhf, i can use rtl_433 on rpi to capture signal on 169 MHZ (even if it's called "433")?

landaisbenj, 2022-01-16:
with any antenna?

weetmuts, 2022-01-16:
Yep, rtl_433 can listen to multiple frequencies the 433 is the historical starting point.

The antenna is not that important when you are close to the meter. Later when you want
to read from longer distances or through walls, then you might want to use a tuned antenna.
See here for example: https://weetmuts.github.io/wmbusmeterswiki/homemadedipole.html

landaisbenj, 2022-01-16:
Ok thanks. It's seems to excit me. I have found some another physical measure who can help me to capture signal: https://www.anfr.fr/fileadmin/mediatheque/documents/expace/2016_Rapport_technique_compteur_Suez.pdf
[image]

Did you know somthing on ondeo system protocol ? 

Whatevere, i buy this dongle.

weetmuts, 2022-01-16:
No, I have not heard of the ondeo protocol before. We will see what it is after we managed to capture the signals.

The dongle is cheap entertainment, you can just run rtl_433 and have it print all sorts of interesting signals, temperature, car tire pressure etc etc. that are transmitting in your surroundings.

landaisbenj, 2022-01-16:
Yes i saw that ! It's why i'm getting this even it don't work for water meter.. Thanks a lot.

nestorayuso, 2022-01-16:
on 868MHz meters transmit every 10 seconds. on 169 MHz meters transmit every 4 hour or more.
You need to perform a data capture for 24 hours, not just minutes.

landaisbenj, 2022-01-16:
Thank for the informtion. I will try to perform with hardware, and make a 24h capture.

merbanan, 2022-01-16:
I am very interested of a good signal sample. This 4 hour interval explains why it is so hard to get a reading.

landaisbenj, 2022-01-16:
https://www.anfr.fr/fileadmin/mediatheque/documents/expace/2016_Rapport_technique_compteur_Suez.pdf explain : 
On the water distribution network, the modules are programmed by default to send the
counting frames every 6 hours

mattiad83, 2022-01-23:
hello to everyone! i'm intrested to this project. i have captured a packet.. i think.. 
[g068_169M_250k.zip](https://github.com/weetmuts/wmbusmeters/files/7921485/g068_169M_250k.zip)

mattiad83, 2022-01-24:
If was helpful i have captured 24H @ 169Mhz. About 430MB of zipped CU8 files.

merbanan, 2022-01-24:
@mattiad83 can you redo with the following parameters?

rtl_433 -s 1000k -f 169500k -S unknown

I looked through the samples and there is a carrier found but I think you miss the actual band where the wmbus-n signal is located.

merbanan, 2022-01-24:
I gave bad frequency settings. With -s 1000 we should cover a larger band. It might be to large. I have never seen a wmbus-n signal so I'm not sure of the settings needed.

mattiad83, 2022-01-24:
@merbanan now i'm capturing with new parameters. 
Tomorrow i upload the caught samples. In attachment the frist. 
[g001_169.5M_1000k.zip](https://github.com/weetmuts/wmbusmeters/files/7929540/g001_169.5M_1000k.zip)

landaisbenj, 2022-01-25:
I have received rtl sdr dongle. I just need to install it on a pi and test it, and i can try after to get a capture... More info this weekend.

mattiad83, 2022-01-25:
@merbanan Some catched signals. i have a lot of recordings but i can't upload here for size limits.
[g2919_169.5M_1000k.zip](https://github.com/weetmuts/wmbusmeters/files/7937429/g2919_169.5M_1000k.zip)

merbanan, 2022-02-06:
All of the files contain signals but there are several frequencies in there. I don't know what is a real signal and what might be aliasing. We need documentation about this device that it is indeed wmbus and not some other protocol.

sgargel, 2022-10-20:
News on this? I'm interested too!

weetmuts, 2022-10-21:
Just so you know, wmbusmeters now support the AMB8665-M hardware dongle for 169MHz. :-)

AkdM, 2022-11-11:
Hey @sgargel @merbanan @mattiad83 @landaisbenj
Looks like you're in France too ([old meters used 868MHz, they now use 169MHz everywhere](http://telereleve-energies.fr/emetteurs-wireless-mbus)).

Have you had any chance capturing and decoding the signals? I am kinda a newbie in that domain, so I'm not really sure what to do with those `.cu8` files and how to analyze them afterwards.

Thanks!

optiluca, 2023-01-20:
Commenting on this to keep it alive. 😊 The issue of lack of support for 169mhz is quite a limiting one, as far as I can tell all smart meters in my area now operate on this band. Since I provably have a few dozen 169mhz meters in my area, I've made a 24hr capture which might hopefully help someone to reverse engineer the protocol. You can find it here: https://github.com/xaelsouth/rtl-wmbus/issues/26#issuecomment-1357343519

Guigui37, 2023-03-06:
I have approximately the same water meter as @landaisbenj here.

I followed recommended capture process on https://triq.org/rtl_433/ANALYZE.html#grab-a-sample
```To get a clean signal remove the receiver antenna and place the device at 10cm to the receiver, that mostly isolates the transmissions.```

So I put a laptop and usb key within 10cm from the water meter, and I think it worked, with following command:
`rtl_433 -s 1000k -f 169500k -S unknown`

[image]

I got a signal at 2 am, then 8 am, each 6 hours, so it follows @landaisbenj document, specifying those water meters communicates every 6 hours.

> https://www.anfr.fr/fileadmin/mediatheque/documents/expace/2016_Rapport_technique_compteur_Suez.pdf explain : On the water distribution network, the modules are programmed by default to send the counting frames every 6 hours.

So here are my first captures:

[firstcaptures.zip](https://github.com/weetmuts/wmbusmeters/files/10895733/firstcaptures.zip)

Index on my water meter was "433.728" 1h30 after g002 and g003 captures

Is there anything understandable in them ? Do I need to try to change the capture command ?

I let captures running to have more data.

Guigui37, 2023-03-07:
Here are captures since yesterday morning.

[captures.tar.gz](https://github.com/wmbusmeters/wmbusmeters/files/10909833/captures.tar.gz)

```
-rw-r--r-- 1 guigui guigui 524288 mars   6 02:03 g001_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   6 08:03 g002_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   6 08:05 g003_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   6 14:02 g004_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   6 20:01 g005_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   6 20:04 g006_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   7 02:01 g007_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   7 08:02 g008_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   7 08:05 g009_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 524288 mars   7 14:03 g010_169.5M_1000k.cu8
-rw-r--r-- 1 guigui guigui 131072 mars   7 14:05 g011_169.5M_1000k.cu8
```

Some values on my water meter:

- g002/g003: around 433.728
- g004: 433.744
- g005/g006: 433.877
- g010/g011: 434.051

It seems every two transmissions (so every 12 hours), there are 2 captures, at 8 am, and 8 pm (20h)

g011 seems strange, less data, and having a look on triq.org does not show same things as the others. Maybe I made some interferences while taking a picture of the water meter.

Again, triq.org seem to show some interesting things, but I am a little bit stuck for now.

Any advices please @merbanan and @weetmuts ? Could you have a look at them please ?
Thanks!

merbanan, 2023-03-07:
All signals are clipped. Can you lower the gain ? I can sort of imagine some bits but I think the distortion hinders reception.

Guigui37, 2023-05-07:
I have found some interesting things, I must upload my last files here quickly, I thought I understood what's needed to have a understandable signal ;-)

Inpriser, 2023-08-01:
Hello,
Any progress on that decoding ?
It seems to be the wize protocol used on these meters : https://en.wikipedia.org/wiki/Wize_technology

AkdM, 2023-08-01:
You seem to be right @Inpriser, however it looks like only the members of the Wize Alliance have access to the technical specifications of Wize technology using the frequency 169 MHz (quoted from Wikipedia). I’m not sure how we would have access to such documents without being a member of such alliance :(

Inpriser, 2023-08-02:
Well I found this presentation where we can find some insights : https://innovation.grdf.fr/statics/filemanager/AAP%20169%20MHz/Informations%20sur%20l%E2%80%99appel%20%C3%A0%20projets%20Hybridation.pdf

And protocol specification seems to be available here : https://www.wize-alliance.com/content/download/620/2430?version=6

I have such wize water meter and just bought an rtl-sdr stick. I would be glad to contribute on how to retrieve consumption data

CappyT, 2023-10-09:
commenting to keep alive, i can provide some signal captures too, from a gas smartmeter.

fsutter, 2023-10-20:
@CappyT ,
is it from a Gazpar?
I'm also interesting in understanding the protocol.
Thanks

CappyT, 2023-10-20:
> @CappyT , is it from a Gazpar? I'm also interesting in understanding the protocol. Thanks

No, here in italy it's basically just a single company making them for the entire state:
https://www.fiorentini.com/it/soluzioni/prodotti/contatori-gas/contatori-gas-smart/rse-rsv/

I believe the protocol is the same between all of them, so decoding a mode N meter would reveal everything from them.

closed 2024-01-19 by github-actions[bot]

mickae1, 2024-01-19:
need it 🥺

ldina, 2024-02-24:
Hi, I have run rtl_433 -s 1000k -f169500k -S unknown
and in 4 minutes got a lot of files that you can see here
[169.5.zip](https://github.com/wmbusmeters/wmbusmeters/files/14394613/169.5.zip)
I'm based in Italy and here there supposed to be a lot of things in this frequency