wmbusmeters/wmbusmeters #448: Are multiple receivers handled by a single instance of `wmbusmeters`?

pswiatki, 2022-01-16:
Is it possible to configure several receivers (say: one for 433, another for 868 MHz) for a single instantiation of `wmbusmeters`?

weetmuts, 2022-01-16:
Yep, set the names of the rtl_sdr dongles like this:
rtl_eeprom -s Dongle433
rtl_eeprom -s Dongle868

Then plug them both in and run:
wmbusmeters rtlwmbus[Dongle433]:433.82M rtlwmbus[Dongle868]

Now you know which dongle is set to 433 and 868 so that
you can attached tuned atennas for the right fq to the right dongle.

In the conf file you simply do multiple lines:
device=rtlwmbus[Dongle433]:433.82M 
device=rtlwmbus[Dongle868]

pswiatki, 2022-01-16:
Great! Thank you so much, @weetmuts. Actually, I was the one who brought the RTL dongle naming concept into the spotlight here on your forum. So, I know the [renaming] technique. All should work (I hope), but I shall try to make sure. Will post the restults.

weetmuts, 2022-03-28:
@pswiatki Did it work?

pswiatki, 2022-03-28:
Let me double check and I will let you know as soon as I can (today / tomorrow).

weetmuts, 2022-05-25:
I think it works, since I have tested it myself. :-) Closing this for now.

closed 2022-05-25 by weetmuts