wmbusmeters/wmbusmeters #450: More config exampels for psql

hngfngzl, 2022-01-22:
Hello,

i got wmbusmeters running with RTL2838. And the log brings a lot of data also in the .json file. The goal is to get the data written into a postgresql database. I got the database running. I also got the syntax to write in form the command line. But with the existing example I don't get it running.
I have many meters that are found. I want to write all this to the database. The filtering and all that stuff is planed to to in the database. But at the moment I don't get the right input in the wmbusmeters.conf.

lsusb brings
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T 

My Text:
wmbusmeters --shell="psql wmbusmeters -c "INSERT INTO stand VALUES '\$METER_ID', '\$METER_TOTAL_M3,'\$METER_TIMESTAMP');" /dev/ttyUSB3:0bda:2838

Can you give me mor examples for a working Text in the wmbusmeters.conf?

Best Regards
Daniel

BIBOLV, 2022-01-23:
Hi,
Did you reviewed and followed example exactly? From your posted command I see that left parenthesis is missing and dollar signs are not escaped. Can you double check and try running with --debug for more details.

weetmuts, 2022-01-24:
Hi @hngfngzl ! 
It is quite difficult to get a working psql command passed on the command line because of multiple levels of escaping.
I suggest you use the conf file. I just tested this:

create table water;
create table consumption (id varchar(8) not null, total_m3 double precision not null, received timestamp not null) ;

Then in the wmbusmeters.conf
shell=PGPASSWORD=12345678 psql -U wmbusmeters --host=localhost --dbname=water -c "insert into consumption values ('$METER_ID', $METER_TOTAL_M3,'$METER_TIMESTAMP');"

Then I started from the command line, later you can use it as a daemon of course:
sudo ./build/wmbusmeters --useconfig=/

Waited for some telegrams and checked /var/log/wmbusmeters.log:
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "PGPASSWORD=12345678 psql -U wmbusmeters --host=localhost --dbname=water -c "insert into consumption values ('$METER_ID', $METER_TOTAL_M3,'$METER_TIMESTAMP');""
(shell) env "METER_JSON={"media":"cold water","meter":"multical21","name":"Vatten","id":"76348799","total_m3":6.409,"target_m3":6.409,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":23,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-01-24T09:01:46Z","device":"im871a[00102759]","rssi_dbm":-45}"
(shell) env "METER_ID=76348799"
(shell) env "METER_NAME=Vatten"
(shell) env "METER_MEDIA=cold water"
(shell) env "METER_TYPE=multical21"
(shell) env "METER_TIMESTAMP=2022-01-24T09:01:46Z"
(shell) env "METER_TIMESTAMP_UTC=2022-01-24T09:01:46Z"
(shell) env "METER_TIMESTAMP_UT=1643014906"
(shell) env "METER_TIMESTAMP_LT=2022-01-24 10:01.46"
(shell) env "METER_DEVICE=im871a[00102759]"
(shell) env "METER_RSSI_DBM=-45"
(shell) env "METER_TOTAL_M3=6.409"
(shell) env "METER_TARGET_M3=6.409"
(shell) env "METER_MAX_FLOW_M3H=0"
(shell) env "METER_FLOW_TEMPERATURE_C=127"
(shell) env "METER_EXTERNAL_TEMPERATURE_C=23"
(shell) env "METER_CURRENT_STATUS=DRY"
(shell) env "METER_TIME_DRY=22-31 days"
(shell) env "METER_TIME_REVERSED="
(shell) env "METER_TIME_LEAKING="
(shell) env "METER_TIME_BURSTING="
(shell) waiting for child 252268 to complete.
(shell) /bin/sh: return code 0

And then from within psql:
select * from consumption:

    id    |       total       |          received          
----------+-------------------+----------------------------
 76348799 |             6.409 | 2022-01-24 09:01:10
 76348799 |             6.409 | 2022-01-24 09:01:29
 76348799 |             6.409 | 2022-01-24 09:01:46

weetmuts, 2022-01-24:
If you really want to use the command line, the example supplied in the README looks good as well:
```
wmbusmeters --shell="psql waterreadings -c \"insert into readings values ('\$METER_ID',\$METER_TOTAL_M3,'\$METER_TIMESTAMP') \" " /dev/ttyUSB0:amb8465 MyColdWater multical21:c1 12345678 NOKEY
```
Note that the `\"` and `\$` are escaped.

hngfngzl, 2022-01-24:
Hello Mr Öhrström,
hello Weemuts,

thanks for the hints. I’ll try. I am a German – I am a better then a “normal” user but not a coder, so sorry for the this beginner question: What means “escaped”?

Best Regards
Daniel Wolf

weetmuts, 2022-01-24:
> What means “escaped”?

A lot of characters have special meanings when you write a bash command, but also inside the psql command.
Typically " and ' means quoting, so if you want to pass a quote as a command and prevent it from being interpreted
as a quote, you have to escape it. Also $ means here is a variable in bash (and in a lot of other languages).

So if you want bash to print a single quote and a dollar signs you cannot type
`echo '$`
you have to type:
`echo \'\$`

This is a complicated topic and you have to learn the basics of bash.
https://tldp.org/LDP/Bash-Beginners-Guide/html/
and in particular:
https://tldp.org/LDP/Bash-Beginners-Guide/html/sect_03_03.html

weetmuts, 2022-02-06:
Can we close this issue now?

hngfngzl, 2022-02-06:
Soory Mr Öhrström,

I couldn't test it out the last 2 weeks. I'll do so an and give a reply.

closed 2022-03-28 by weetmuts