wmbusmeters/wmbusmeters #452: Support for Zenner Celsius C5 ISF

probutus, 2022-01-22:
I got my new Zenner C5 ISF ( [ https://pim.zenner.com/wp-content/uploads/documents/data_sheets/heat_meter/EN/DB_WMZ_zelsius_C5_ISF_EN.pdf](url) )along with the AES key today and created a driver file for it. It is strongly similar to the driver_sharky774.cc file

`root@andromeda:/home/pi/wmbusmeters/build# ./wmbusmeters ../../simulation_c5isf.txt zenner_wmz c5isf 32002044 XXXXXX
zenner_wmz      32002044        26 kWh  0 kWh   2.242 m3        0 m3    0.164 m3/h      2.5 kW  56.5 °C 43.22 °C        0 °C    2022-01-22 23:48.22
root@andromeda:/home/pi/wmbusmeters/build# ./wmbusmeters --format=json ../../simulation_c5isf.txt zenner_wmz c5isf 32002044 XXXXXXXXXX
{"media":"heat","meter":"c5isf","name":"zenner_wmz","id":"32002044","total_energy_consumption_kwh":26,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":2.242,"total_volume_tariff2_m3":0,"volume_flow_m3h":0.164,"power_kw":2.5,"flow_temperature_c":56.5,"return_temperature_c":43.22,"temperature_difference_c":0,"timestamp":"2022-01-22T22:49:16Z"}`

The issue I have now is that I do not see that the temperature difference is updated properly.

I attached the new file for reference.

[driver_c5isf.tar.gz](https://github.com/weetmuts/wmbusmeters/files/7919457/driver_c5isf.tar.gz)

probutus, 2022-01-25:
I got the feedback from Zenner regarding the meaning of the values in the T1 Telegram. It is a T1B Telegram looking like this:

T1B Telegram

telegram=|5E44496A4420003288047A0A0050052F2F#04061A0000000413C20800008404060000000082046CC121043BA4000000042D1900000002591216025DE21002FD17000084800106000000008280016CC121948001AE25000000002F2F2F2F2F2F|+0
(c5isf) 000   : 5e length (94 bytes)
(c5isf) 001   : 44 dll-c (from meter SND_NR)
(c5isf) 002   : 496a dll-mfct (ZRI)
(c5isf) 004   : 44200032 dll-id (32002044)
(c5isf) 008   : 88 dll-version
(c5isf) 009   : 04 dll-type (Heat meter)
(c5isf) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(c5isf) 011   : 0a tpl-acc-field
(c5isf) 012   : 00 tpl-sts-field (OK)
(c5isf) 013   : 5005 tpl-cfg 0550 (AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
(c5isf) 015   : 2f2f decrypt check bytes
(c5isf) 017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(c5isf) 018   : 06 vif (Energy kWh)
(c5isf) 019 C!: 1A000000 ("total_energy_consumption_kwh":26)          -> OK Heat Energy
(c5isf) 023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(c5isf) 024   : 13 vif (Volume l)
(c5isf) 025 C!: C2080000 ("total_volume_m3":2.242)                    -> OK Total Volume
(c5isf) 029   : 84 dif (32 Bit Integer/Binary Instantaneous value)
(c5isf) 030   : 04 dife (subunit=0 tariff=0 storagenr=8)
(c5isf) 031   : 06 vif (Energy kWh)
(c5isf) 032 C?: 00000000                                              -> Missing: Duedate Energy
(c5isf) 036   : 82 dif (16 Bit Integer/Binary Instantaneous value)
(c5isf) 037   : 04 dife (subunit=0 tariff=0 storagenr=8)
(c5isf) 038   : 6C vif (Date type G)
(c5isf) 039 C?: C121                                                  -> Missing: Duedate date Energy
(c5isf) 041   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(c5isf) 042   : 3B vif (Volume flow l/h)
(c5isf) 043 C!: A4000000 ("volume_flow_m3h":0.164)                    -> OK Flow Volume
(c5isf) 047   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(c5isf) 048   : 2D vif (Power 10² W)
(c5isf) 049 C!: 19000000 ("power_kw":2.5)                             -> OK Power
(c5isf) 053   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(c5isf) 054   : 59 vif (Flow temperature 10⁻² °C)
(c5isf) 055 C!: 1216 ("flow_temperature_c":56.5)                      -> OK Flow Temperature
(c5isf) 057   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(c5isf) 058   : 5D vif (Return temperature 10⁻² °C)
(c5isf) 059 C!: E210 ("return_temperature_c":43.22)                   -> OK Return Temperature
(c5isf) 061   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(c5isf) 062   : FD vif (Second extension FD of VIF-codes)
(c5isf) 063   : 17 vife (Error flags (binary))
(c5isf) 064 C?: 0000                                                  -> Missing: Error Flags
(c5isf) 066   : 84 dif (32 Bit Integer/Binary Instantaneous value)
(c5isf) 067   : 80 dife (subunit=0 tariff=0 storagenr=0)
(c5isf) 068   : 01 dife (subunit=0 tariff=0 storagenr=32)
(c5isf) 069   : 06 vif (Energy kWh)
(c5isf) 070 C?: 00000000                                               -> Missing: Energy Log last month
(c5isf) 074   : 82 dif (16 Bit Integer/Binary Instantaneous value)
(c5isf) 075   : 80 dife (subunit=0 tariff=0 storagenr=0)
(c5isf) 076   : 01 dife (subunit=0 tariff=0 storagenr=32)
(c5isf) 077   : 6C vif (Date type G) 
(c5isf) 078 C?: C121                                                   -> Missing: Energy Logdate last month
(c5isf) 080   : 94 dif (32 Bit Integer/Binary Maximum value)
(c5isf) 081   : 80 dife (subunit=0 tariff=0 storagenr=0)
(c5isf) 082   : 01 dife (subunit=0 tariff=0 storagenr=32)
(c5isf) 083   : AE vif (Power kW)
(c5isf) 084   : 25 vife (per month)
(c5isf) 085 C?: 00000000                                              -> Missing: Maximum Power this month
(c5isf) 089   : 2F skip
(c5isf) 090   : 2F skip
(c5isf) 091   : 2F skip
(c5isf) 092   : 2F skip
(c5isf) 093   : 2F skip
(c5isf) 094   : 2F skip
zenner_wmz      32002044        26 kWh  0 kWh   2.242 m3        0 m3    0.164 m3/h      2.5 kW  56.5 °C 43.22 °C        0 °C    2022-01-23 00:32.33

Can someone point me in the direction how I can set up the mapping of the values that the values get assigned properly and will be shown in the JSON TEelegram?

weetmuts, 2022-01-25:
Thank you for the driver! How nice of Zenner to respond to your question. 
I have added the missing fields and pushed to github please test. For example like this:

wmbusmeters --analyze 5E44496A4420003288047A0A0050052F2F#04061A0000000413C20800008404060000000082046CC121043BA4000000042D1900000002591216025DE21002FD17000084800106000000008280016CC121948001AE25000000002F2F2F2F2F2F

Do you think you can ask Zenner about the bits in the error flags? The lookup table now only has a single silly bit translating to "?".

The temperature difference was a value supplied by the sharky meter as calculated by the meter.
As you can see in the analyze above, no difference is in the telegram content.
I removed some tariff 1&2 that does not exist in the c5isf meter either.

I am not entirely sure about the differene between of due date energy and last month energy. Usually meters
only have one (target_energy or last_month_energy etc) which is the meters total energy sum when the month changes. This makes it possible to readout the previous month any time during the next month. 
Perhaps you will know more after using the meter some more?

probutus, 2022-01-25:
I tested the driver and it's working:
{"media":"heat","meter":"c5isf","name":"zenner_wmz","id":"32002044","total_energy_consumption_kwh":136,"total_volume_m3":13.275,"due_energy_consumption_kwh":0,"due_date":"2022-01-01","volume_flow_m3h":0,"power_kw":0,"total_energy_consumption_last_month_kwh":0,"last_month_date":"2022-01-01","max_power_last_month_kw":0,"flow_temperature_c":51.77,"return_temperature_c":43.7,"status":"OK","timestamp":"2022-01-25T20:17:00Z","device":"cul","rssi_dbm":-49}

The due date energy and the due date date energy is the yearly snapshot of the energy (in my case 1.1.2022 where the total value is conserved)
Additionally to that there is the values of the last month stored

Regarding the error codes I have this list:

`Code / Error type resp. possible reason 
1 Temperature below of measuring range
2 Temperature above of measuring range 
3 Short circuit return sensor / Check sensors
4 Interruption return sensor 
5 Short circuit supply sensor Check sensors
6 Interruption supply sensor Check sensors
7 Battery voltage Exchange device
8 Hardware error Exchange device
9 Hardware error Exchange device
10 Error in the measuring system 
20 No water in the measuring tube 
30 Reverse water flow detected 
40 Air inside the medium Vent system (**)
50 Measured value outside overload range 
100 Hardware error 
800 Wireless interface 
1000 Status end of the battery 
2000 Status initial verification expired 

Error codes show faults detected by zelsius® C5-IUF. If more than one error appears, the sum
of the error codes is displayed: for example, error 1005 = error 1000 and error 5

Zenner gave me two description of telegrams:
T1A - ESCO telegram with monthly values -> I have not seen this type of telegram yet but I got an example I show below
T1B - Utility Telegram with diagnostic values -> this is the telegram above

Is there a way to support both types of telegrams (although I did not see the T1A yet)?

From the specification the T1A telegram binary looks like this (example):

E544496A44554455880D7A320200002F2F04060000000004130000000002FD17240084800106000000008280016C2124C480010600000080C280016CFFFF84810106000000808281016CFFFFC481010600000080C281016CFFFF84820106000000808282016CFFFFC482010600000080C282016CFFFF84830106000000808283016CFFFFC483010600000080C283016CFFFF84840106000000808284016CFFFFC484010600000080C284016CFFFF84850106000000808285016CFFFFC485010600000080C285016CFFFF84860106000000808286016CFFFFC486010600000080C286016CFFFF

The mfct/type/version combo was not found in the driver lookup table.
000   : e5 length (229 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 496a dll-mfct (ZRI)
004   : 44554455 dll-id (55445544)
008   : 88 dll-version
009   : 0d dll-type (Heat/Cooling load meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 32 tpl-acc-field
012   : 02 tpl-sts-field (ERROR)
013   : 0000 tpl-cfg 0000 ( )
015   : 2F skip
016   : 2F skip
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 06 vif (Energy kWh)
019 C?: 00000000   **Heat Energy current**
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 13 vif (Volume l)
025 C?: 00000000 ** Volume current ***
029   : 02 dif (16 Bit Integer/Binary Instantaneous value)
030   : FD vif (Second extension FD of VIF-codes)
031   : 17 vife (Error flags (binary))
032 C?: 2400  ** Error code **
034   : 84 dif (32 Bit Integer/Binary Instantaneous value)
035   : 80 dife (subunit=0 tariff=0 storagenr=0)
036   : 01 dife (subunit=0 tariff=0 storagenr=32)
037   : 06 vif (Energy kWh) ** Energy Month -1 **
038 C?: 00000000
042   : 82 dif (16 Bit Integer/Binary Instantaneous value)
[... 118 lines cut ...]
206 C?: 00000080
210   : 82 dif (16 Bit Integer/Binary Instantaneous value)
211   : 86 dife (subunit=0 tariff=0 storagenr=12)
212   : 01 dife (subunit=0 tariff=0 storagenr=44)
213   : 6C vif (Date type G) ** Date Month -13 **
214 C?: FFFF
216   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
217   : 86 dife (subunit=0 tariff=0 storagenr=13)
218   : 01 dife (subunit=0 tariff=0 storagenr=45)
219   : 06 vif (Energy kWh) ** Energy Month -14 **
220 C?: 00000080
224   : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
225   : 86 dife (subunit=0 tariff=0 storagenr=13)
226   : 01 dife (subunit=0 tariff=0 storagenr=45)
227   : 6C vif (Date type G) ** Date Month -14 **
228 C?: FFFF
{
    "media":"heat/cooling load",
    "meter":"munia",
    "name":"",
    "id":"55445544",
    "current_temperature_c":0,
    "current_relative_humidity_rh":0,
    "timestamp":"2022-01-25T20:31:56Z"
}

weetmuts, 2022-01-25:
Yep, you can add the T1A data easily to the driver. Just change the name and the storage nr.

Hmm, so the status value contains combinations of decimal values... ok, need to add support for this. The current table
is based in binary bits being set. Or exact match of values.

The analyze is improved and will pick the right driver for you now.

probutus, 2022-01-26:
I have some questions regarding implementing the T1A telegram. I see in the header:

000 : e5 length (229 bytes)
001 : 44 dll-c (from meter SND_NR)
002 : 496a dll-mfct (ZRI)
004 : 44554455 dll-id (55445544)
008 : 88 dll-version
009 : 0d dll-type (Heat/Cooling load meter)

that the dll version is identical (0x88) to the other T1B telegram. I can only distinguish the content via the dll-type. I understood that wmbusmeters does distinguish only va the manufacturer/dll version, is this correct?

How can I tell my driver which type of telegram to parse?

Second question I have is about the meaning of 

        StorageNr(0),
        TariffNr(0),
        IndexNr(1),

Is all this only about the grouping in the json telegram or is this also used for parsing?

Is the order I group the "addFieldWithExtractor()" functions is the order of the values in the telegram?

probutus, 2022-01-27:
I split the driver into two files 

driver_c5isf_t1b.cc dealing with the T1B telegram (the one already known) and added 
driver_c5isf_t1a.cc dealing with the T1A telegram I got the specification for from Zenner. This telegram is used by metering companies to collect the consumption values of the last 14 months and is apparently sent all 120sec.

My meter does not send this telegram so I could only test the driver with the sample telegram provided by zenner.

[driver_c5isf.zip](https://github.com/weetmuts/wmbusmeters/files/7954186/driver_c5isf.zip)

weetmuts, 2022-02-10:
Thank you very much for your work! I have been thinking about the telegrams that this meter sends.
Can you do a pull request? This would make it easier to discuss and test. 

They telegrams are very distinct, which makes it reasonable to use multiple drivers for different telegrams sent by the same meter. It might work out of the box, but this is a first for wmbusmeters.  :-)

Previously there are several other meters that send two types of telegrams, usually one often with the consumption data, and one more seldom with meter meta data and perhaps historical data. Sometimes OMS standard compliant data and in another proprietary data.

But so far, they have sent the data using the same device and version numbers..... so it sort of was natural to handle all telegrams in the same driver. Also the state is maintained in a single driver C++ object. So multiple updates modify the same in-memory object. Then the whole object is reported in every mqtt message. Which is both good and bad. Good I think. 

But it creates a question when the first mqtt message should be sent. When any partial data has arrived or wait until you have received a telegram of each type and then after every telegram. This is usually only a program when wmbusmeters restarts and the C++ object is restared empty. Perhaps this should be configurable.

In this case it is possible to separate the telegrams into multiple drivers since the meter seems to use different version numbers for the telegrams, which is good. However there is some overlap with the original driver. Should the original driver be removed?

What happens if you run "make test"  ?

probutus, 2022-02-10:
the telegrams are independent of each other. both T1A telegrams are used from the metering companies to collect the historical data over the last months and are to be sent all 120sec

The t1b telegram contains more diagnostic data like flow temperature and return temperature as well as the circulated amount of water.

The original driver file can be replaced by the three new files.

When running the --analyze command on the telegram everything looks ok but when I do "make test" it somehow stops when parsing the t1a2 telegram and has issue to correctly map the vendor, type and version.

Extract:
Testing drivers
OK json: driver_amiplus.cc
OK fields: driver_amiplus.cc
OK json: driver_apator08.cc
OK fields: driver_apator08.cc
OK json: driver_aventieshca.cc
OK fields: driver_aventieshca.cc
OK json: driver_aventieswm.cc
OK fields: driver_aventieswm.cc
wmbusmeters returned error code: 1
(meter) Heat: meter detection could not find driver for id: 58455445 mfct: (IDV) Unknown (0xa496) type: Unknown (0xd7) ver: 0x80
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Heat: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 58455445 mfct: (IDV) Unknown (0xa496) type: Unknown (0xd7) ver: 0x80
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
wmbusmeters: src/meters.cc:905: MeterCommonImplementation::addStringFieldWithExtractorAndLookup(std::string, Quantity, DifVifKey, MeasurementType, ValueInformation, StorageNr, TariffNr, IndexNr, int, std::string, std::function<void(std::__cxx11::basic_string<char>)>, std::function<std::__cxx11::basic_string<char>()>, Translate::Lookup)::<lambda(FieldInfo*, Meter*, Telegram*)>: Assertion `0' failed.
Aborted (core dumped)
wmbusmeters returned error code: 1
(meter) Heat: meter detection could not find driver for id: 58455445 mfct: (IDV) Unknown (0xa496) type: Unknown (0xd7) ver: 0x80
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Heat: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 58455445 mfct: (IDV) Unknown (0xa496) type: Unknown (0xd7) ver: 0x80
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
wmbusmeters: src/meters.cc:905: MeterCommonImplementation::addStringFieldWithExtractorAndLookup(std::string, Quantity, DifVifKey, MeasurementType, ValueInformation, StorageNr, TariffNr, IndexNr, int, std::string, std::function<void(std::__cxx11::basic_string<char>)>, std::function<std::__cxx11::basic_string<char>()>, Translate::Lookup)::<lambda(FieldInfo*, Meter*, Telegram*)>: Assertion `0' failed.
Aborted (core dumped)
0a1
> {"media":"water","meter":"c5isf_t1a2","name":"Heat","id":"55445555","total_volume_m3":0,"total_volume_consumption_last_month_m3":0,"last_month_date":"2017-04-01","total_volume_consumption_month-2_m3":21474836.48,"month-2_date":"2127-15-31","total_volume_consumption_month-3_m3":21474836.48,"month-3_date":"2127-15-31","total_volume_consumption_month-4_m3":21474836.48,"month-4_date":"2127-15-31","total_volume_consumption_month-5_m3":21474836.48,"month-5_date":"2127-15-31","total_volume_consumption_month-6_m3":21474836.48,"month-6_date":"2127-15-31","total_volume_consumption_month-7_m3":21474836.48,"month-7_date":"2127-15-31","total_volume_consumption_month-8_m3":21474836.48,"month-8_date":"2127-15-31","total_volume_consumption_month-9_m3":21474836.48,"month-9_date":"2127-15-31","total_volume_consumption_month-10_m3":21474836.48,"month-10_date":"2127-15-31","total_volume_consumption_month-11_m3":21474836.48,"month-11_date":"2127-15-31","total_volume_consumption_month-12_m3":21474836.48,"month-12_date":"2127-15-31","total_volume_consumption_month-13_m3":21474836.48,"month-13_date":"2127-15-31","total_volume_consumption_month-14_m3":21474836.48,"month-14_date":"2127-15-31","timestamp":"1111-11-11T11:11:11Z"}
0a1
> Heat;55445555;1111-11-11 11:11.11
ERROR: driver_c5isf_t1a2.cc
Slower tests...
OK: reading all stdin before closing down
RUNNING Test alarms ...
STDERR---------------------------------
[ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2022-02-10 19:10)
(wmbus) successfully reset wmbus device
STDOUT---------------------------------
TMP/OUTPUT-----------------------------
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","total_m3":6.408,"target_m3":6.408,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-02-10T18:10:52Z"}==
METER =={"media":"cold water","meter":"multical21","name":"Water","id":"76348799","total_m3":6.408,"target_m3":6.408,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":19,"current_status":"DRY","time_dry":"22-31 days","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-02-10T18:10:58Z"}==
TMP/ALARM------------------------------
ALARM_SHELL DeviceInactivity [ALARM DeviceInactivity] 4 seconds of inactivity resetting simulations/simulation_alarm.txt simulation (timeout 4s expected mon-sun(00-23) now 2022-02-10 19:10)
---------------------------------------
OK: Test alarms
Some tests failed!
make: *** [Makefile:238: test] Error 1

probutus, 2022-02-10:
Pull request created. This is my very first one
#474

weetmuts, 2022-03-28:
@probutus Still thinking! I have not forgotten this. :-)

probutus, 2022-03-30:
sure, no worries, I'll wait :-)

weetmuts, 2022-04-11:
Ok, after a lot of thinking. Since the telegrams represent different aspects of the same meter, there should be a single C++ object containing the state of the meter. I have merged your three drivers in a single one.
https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_c5isf.cc

I have also added a decimals to index transltor for the status bits so those work as well.

I also made some changes to the json key names.

Please test!

probutus, 2022-04-18:
Thanks for the update. I tested the latest version and it looks like the nanoCUL does not get detected anymore (switching back to the original binary the nanocul gets found and works)

root@andromeda:/home/pi# wmbusmeters --debug --verbose --donotprobe=all --logtelegrams /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 auto:t1
(wmbusmeters) version: 1.7.0-25-g11763b2
(config) using device: /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 
(config) number of meters: 0
(lookup) with file/hex "/dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0" unknown
(serial) check if /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(serial) tty /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(verbose) ignoring device /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1165 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1161
<<<
(shell) return code 0
(shell) 1166 exited
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
(lookup) with file/hex "/dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0" unknown
(serial) check if /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(serial) tty /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(verbose) ignoring device /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0
(lookup) with file/hex "/dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0" unknown
(serial) check if /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(serial) tty /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(verbose) ignoring device /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0
(lookup) with file/hex "/dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0" unknown
(serial) check if /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(serial) tty /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(verbose) ignoring device /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0
(lookup) with file/hex "/dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0" unknown
(serial) check if /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(serial) tty /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0 can be accessed
(verbose) ignoring device /dev/serial/by-id/usb-SHK_NANO_CUL_868-if00-port0

probutus, 2022-05-22:
It looks like one value does not get transferred correctly:

Use this Telegram:
5E44496A4420003288047AFC0050052F2F_0406D00E00000413B28A05008404060000000082046CC121043B00000000042D000000000259E719025D051402FD17000084800106C00C00008280016CC125948001AE25090000002F2F2F2F2F2F

In the old driver the value "max_power_last_month_kw":9 got mapped correctly. In the new format it is missing: "max_power_last_month_kw":0,

Can you check?

weetmuts, 2022-05-22:
Thanks for the telegram! Yes, a lot of reworking of the underlying driver structure has been done.
In this case it was the VIFCombinable that got lost.  I have refactored the c5isf driver to the newest style.
Much easier to read and write.

I have made two changes to field handling. A non-optional field, for which no value was found, will print null instead of 0 or "". If a field is marked with PrintProperty::OPTIONAL then if no value has been found, the field will not print at all.

Please verify that it looks ok.

probutus, 2022-05-23:
Thanks! Everything works now. Can you close the topic?

closed 2022-05-24 by weetmuts