wmbusmeters/wmbusmeters #453: Support for Tauron T1 amiplus with other tariff

klew, 2022-01-26:
Hi

I tried to use amiplus driver to read data from my electricity meter, but I got "0 kWh" of energy (both forward and backward).
I paste debug logs from parsing of telegram.
It looks like problem is related to fact that I have 2 tariff on my meter, while amiplus driver is looking for fields which are applicable to meters with single tariff (https://github.com/weetmuts/wmbusmeters/issues/13).
Other paramters like voltage, forward/backward power looks ok.

Would you please add support for meters with other tariff as well? As far as I know in Poland we have up to 3 tariff (depending on contract and type of selected billing)

```
(telegram) TPL CI=7a ACC=00 STS=00 CFG=0570 (AES_CBC_IV nb=7 cntn=0 ra=0 hc=0)
telegram=|804401066448068602027A000070052F2F#066D1E5C11DA21400C78644806868E10036110012500008E20038106531800008E10833C9949000000008E20833C8606000000001B2B5228020B2B3217000BAB3C0000000AFDC9FC0131020AFDC9FC0225020AFDC9FC0331020BABC8FC100000002F2F2F2F2F2F2F2F2F2F2F2F2FDE47|+152
(dvparser) warning: cannot extract double from non-existant key "0E833C"
(amiplus) 000   : 80 length (128 bytes)
(amiplus) 001   : 44 dll-c (from meter SND_NR)
(amiplus) 002   : 0106 dll-mfct (APA)
(amiplus) 004   : 64480686 dll-id (86064864)
(amiplus) 008   : 02 dll-version
(amiplus) 009   : 02 dll-type (Electricity meter)
(amiplus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(amiplus) 011   : 00 tpl-acc-field
(amiplus) 012   : 00 tpl-sts-field (OK)
(amiplus) 013   : 7005 tpl-cfg 0570 (AES_CBC_IV nb=7 cntn=0 ra=0 hc=0 )
(amiplus) 015   : 2f2f decrypt check bytes
(amiplus) 017   : 06 dif (48 Bit Integer/Binary Instantaneous value)
(amiplus) 018   : 6D vif (Date and time type)
(amiplus) 019 C!: 1E5C11DA2140 ("device_date_time":"2022-01-26 17:28")
(amiplus) 025   : 0C dif (8 digit BCD Instantaneous value)
(amiplus) 026   : 78 vif (Fabrication no)
(amiplus) 027 C?: 64480686
(amiplus) 031   : 8E dif (12 digit BCD Instantaneous value)
(amiplus) 032   : 10 dife (subunit=0 tariff=1 storagenr=0)
(amiplus) 033   : 03 vif (Energy Wh)
(amiplus) 034 C?: 611001250000
(amiplus) 040   : 8E dif (12 digit BCD Instantaneous value)
(amiplus) 041   : 20 dife (subunit=0 tariff=2 storagenr=0)
(amiplus) 042   : 03 vif (Energy Wh)
(amiplus) 043 C?: 810653180000
(amiplus) 049   : 8E dif (12 digit BCD Instantaneous value)
(amiplus) 050   : 10 dife (subunit=0 tariff=1 storagenr=0)
(amiplus) 051   : 83 vif (Energy Wh)
(amiplus) 052   : 3C vife (backward flow)
(amiplus) 053 C?: 994900000000
(amiplus) 059   : 8E dif (12 digit BCD Instantaneous value)
(amiplus) 060   : 20 dife (subunit=0 tariff=2 storagenr=0)
(amiplus) 061   : 83 vif (Energy Wh)
(amiplus) 062   : 3C vife (backward flow)
(amiplus) 063 C?: 860600000000
(amiplus) 069   : 1B dif (6 digit BCD Maximum value)
(amiplus) 070   : 2B vif (Power W)
(amiplus) 071 C?: 522802
(amiplus) 074   : 0B dif (6 digit BCD Instantaneous value)
(amiplus) 075   : 2B vif (Power W)
(amiplus) 076 C!: 321700 ("current_power_consumption_kw":1.732)
(amiplus) 079   : 0B dif (6 digit BCD Instantaneous value)
(amiplus) 080   : AB vif (Power W)
(amiplus) 081   : 3C vife (backward flow)
(amiplus) 082 C!: 000000 ("current_power_production_kw":0)
(amiplus) 085   : 0A dif (4 digit BCD Instantaneous value)
(amiplus) 086   : FD vif (Second extension FD of VIF-codes)
(amiplus) 087   : C9 vife (10^0 Volts)
(amiplus) 088   : FC vife (Reserved)
(amiplus) 089   : 01 vife (Credit of 10^-2 of the nominal local legal currency units)
(amiplus) 090 C!: 3102 ("voltage_at_phase_1_v":231)
(amiplus) 092   : 0A dif (4 digit BCD Instantaneous value)
(amiplus) 093   : FD vif (Second extension FD of VIF-codes)
(amiplus) 094   : C9 vife (10^0 Volts)
(amiplus) 095   : FC vife (Reserved)
(amiplus) 096   : 02 vife (Credit of 10^-1 of the nominal local legal currency units)
(amiplus) 097 C!: 2502 ("voltage_at_phase_2_v":225)
(amiplus) 099   : 0A dif (4 digit BCD Instantaneous value)
(amiplus) 100   : FD vif (Second extension FD of VIF-codes)
(amiplus) 101   : C9 vife (10^0 Volts)
(amiplus) 102   : FC vife (Reserved)
(amiplus) 103   : 03 vife (Credit of 10^0 of the nominal local legal currency units)
(amiplus) 104 C!: 3102 ("voltage_at_phase_3_v":231)
(amiplus) 106   : 0B dif (6 digit BCD Instantaneous value)
(amiplus) 107   : AB vif (Power W)
(amiplus) 108   : C8 vife (upper limit)
(amiplus) 109   : FC vife (additive correction constant: unit of VIF * 10^-3)
(amiplus) 110   : 10 vife (?)
(amiplus) 111 C?: 000000
(amiplus) 114   : 2F skip
(amiplus) 115   : 2F skip
(amiplus) 116   : 2F skip
(amiplus) 117   : 2F skip
(amiplus) 118   : 2F skip
(amiplus) 119   : 2F skip
(amiplus) 120   : 2F skip
(amiplus) 121   : 2F skip
(amiplus) 122   : 2F skip
(amiplus) 123   : 2F skip
(amiplus) 124   : 2F skip
(amiplus) 125   : 2F skip
(amiplus) 126   : 2F skip
(amiplus) 127   : DE dif (12 digit BCD Maximum value storagenr=1)
(amiplus) 128   : 47 dife (subunit=1 tariff=0 storagenr=15)
MyElectricity	86064864	0 kWh	1.732 kW	0 kWh	0 kW	231 V	225 V	231 V	2022-01-26 17:28.34
```

closed 2022-02-05 by weetmuts

klew, 2022-06-15:
@klaudiusz223 I had all tariffs since the begining. However I have HAN activated for 2 years and it was used with different device. Then it stopped working when photovoltaic installation was added and I couldn't activate it from web. I contacted Tauron and they said that this service is not available with photoviltaic installation, so I contacted department responsible for AMIplus/HANplus service and they activated everything and only after that point of time I tried to use my own usb dongle to read that data.

klaudiusz223, 2022-06-15:
@klew 
Thanks for the information. How did you contact Tauron?
I've sent email to han@tauron-dystrybucja.pl long time ago but had no answer.
Is the contact detail on this page https://amiplus.tauron-dystrybucja.pl/kontakt  or anywhere else?
Klaudiusz

klew, 2022-06-15:
@klaudiusz223 I wrote to that han@ address. They replied after 1 week.

PR #455 (2022-02-05): amiplus driver: added energy parameters for tariffs 1, 2, and 3
Closes #453