wmbusmeters/wmbusmeters #456: mqtt support without mosquitto_pub

rubdos, 2022-01-29:
Hello!

I've got wmbusmeters working since a long time now, and generally very happy with it. A few weeks ago, I got a SMART warning on one of my SSDs, sitting at 380TB written and becoming critical in the spare areas. Part from this was some extremely verbose logging from several things I run, which I tuned down. I also noticed that `mosquitto` was logging some of these every 15 seconds:

```
mosquitto[1276]: 1643462409: New connection from ::1 on port 1883.
mosquitto[1276]: 1643462409: New client connected from ::1 as mosq-AMwFUaLp6cbvyAbAp5 (p2, c1, k60).
mosquitto[1276]: 1643462409: Client mosq-AMwFUaLp6cbvyAbAp5 disconnected.
```

It took a while to find out this was the `mosquitto_pub` command that I had configured for wmbusmeters; I have since changed the command to this: `shell=/usr/bin/mosquitto_pub -V 5 -I WMbusmeters -h localhost -t wmbusmeters/"$METER_NAME" -m "$METER_JSON"`. I have also set mosquitto not to have persistent logging.

---

Anyway, I thought it might be nice to have mqtt support directly in wmbusmeters. That way, there's no reconnection happening every 15 seconds.

BIBOLV, 2022-01-31:
Hi,
Implementing mqtt library in wmbusmeters will not prevent a need to work by protocol specification and thus creating keepalive connections on specific intervals.
As working with external commands works perfectly well and escapes need to make wmbusmeters code even more complex, I would say is a good thing and implementing some external libraries would make all code base and maintenance unnecessary complicated.
IMO you should try to fix this on your end by placing all logs on tmpfs and making log rotation and by doing that you should be able significantly decrease write operations and second thing, that you already did - limit or decrease logging as such to minimum.

weetmuts, 2022-02-03:
Thanks! Where is mosquitto_pub logging? Is this enabled by default? Maybe we can update the example in the README and the default conf to include a setting to avoid unnecessary logging?

I agree with @BIBOLV that using external commands is much preferable and is the Unix way.

rubdos, 2022-02-04:
Ok! I get the sentiment.

FWIW, it is *not* mosquitto_pub that's logging, it is the mosquitto daemon that logs every connection (attempt), and it seems to me like the right thing to do. Meanwhile, I configured journald not to persist the mosquitto log anymore, because it's chatty. Sorry for the confusion!

weetmuts, 2022-02-05:
No worries! Have a nice day!

closed 2022-02-05 by weetmuts