wmbusmeters/wmbusmeters #457: Documentation change concerning CUL

Kosygor, 2022-01-29:
Hello 
There is high chance that I **do not get something right** but I was switching from RTL-SDR to CUL (arduino nano clone + cc1101). 
I have Multical21 which sends c1 telegrams and [Techem Smoke Detector 2](https://github.com/weetmuts/wmbusmeters/issues/200#issuecomment-1015086261) which sends t1 telegrams.
At the begging I have set device=auto:t1  and was able just to see tsd2. I could not set CUL to c1 mode, so I had to [compiled ](https://sourceforge.net/p/culfw/code/HEAD/tree/trunk/culfw/Devices/nanoCUL/board.h#l92)my own version of CUL fw (basically set MBUS on and other proprietary features off). 
All of the sudden when CUL is set to auto:c1 it can receive both tsd2 and multical21 frame,. so this part of readme:
`The cul dongle can listen to either s1, c1 or t1, **but only one at a time**.` 
seems to be invalid.

weetmuts, 2022-02-09:
Interesting. Can you provide a --debug log when this happens? Ie you receive both meters, one on c1 and one on t1.

weetmuts, 2022-03-28:
Ping?

Kosygor, 2022-04-05:
Sorry for delay Ping is way to high.
Config:
```
data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=auto:c1
  logtelegrams=false
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_ID "$METER_JSON"
  ignoreduplicates=false
meters:
  - |-
    name=wodomierz
    driver=multical21
    id=74401316
    key=keyX
  - |-
    name=Czujnik dymu schody
    driver=tsd2
    id=91633566
    key=
  - |-
    name=Czujnik dymu kotłownia
    driver=tsd2
    id=91631186
    key=
  - |-
    name=Czujnik dymu garaż
    driver=tsd2
    id=91633560
    key=
  - |-
    name=Czujnik dymu pralnia
    driver=tsd2
    id=91633562
    key=
  - |-
    name=Czujnik dymu kuchnia
    driver=tsd2
    id=91633564
    key=
  - |-
    name=Czujnik dymu gościnny
    driver=tsd2
    id=91633565
    key=
mqtt:
  host: 192.168.0.111
  port: 1883
  user: mqtt
  password: '!secret mqttpwd'

```
Log:

```
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_ID "$METER_JSON""
(shell) env "METER_JSON={"media":"cold water","meter":"multical21","name":"wodomierz","id":"74401316","total_m3":57.661,"target_m3":54.766,"max_flow_m3h":0,"flow_temperature_c":3,"external_temperature_c":17,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"1-8 hours","timestamp":"2022-04-05T18:47:36Z","device":"cul","rssi_dbm":-76}"
(shell) env "METER_ID=74401316"
(shell) env "METER_NAME=wodomierz"
(shell) env "METER_MEDIA=cold water"
(shell) env "METER_TYPE=multical21"
(shell) env "METER_TIMESTAMP=2022-04-05T18:47:36Z"
(shell) env "METER_TIMESTAMP_UTC=2022-04-05T18:47:36Z"
(shell) env "METER_TIMESTAMP_UT=1649184456"
(shell) env "METER_TIMESTAMP_LT=2022-04-05 20:47.36"
(shell) env "METER_DEVICE=cul"
(shell) env "METER_RSSI_DBM=-76"
(shell) env "METER_TOTAL_M3=57.661"
(shell) env "METER_TARGET_M3=54.766"
(shell) env "METER_MAX_FLOW_M3H=0"
(shell) env "METER_FLOW_TEMPERATURE_C=3"
(shell) env "METER_EXTERNAL_TEMPERATURE_C=17"
(shell) env "METER_CURRENT_STATUS="
(shell) env "METER_TIME_DRY="
(shell) env "METER_TIME_REVERSED="
(shell) env "METER_TIME_LEAKING="
(shell) env "METER_TIME_BURSTING=1-8 hours"
(shell) waiting for child 327 to complete.
(shell) /bin/sh: return code 0
(wmbus) parseDLL @0 43
(wmbus) parseELL @10 33
(meter) Czujnik dymu pralnia: for me? 74401316 in 91633562
(meter) Czujnik dymu pralnia: not for me: not my id
```
...
```
(meter) Czujnik dymu kuchnia: for me? 91633564 in 91633564
(meter) Czujnik dymu kuchnia: yes for me
(meter) created "Czujnik dymu kuchnia" "tsd2" "91633564" not-encrypted
(meter) used meter template Czujnik dymu kuchnia 91633564 tsd2 to match 91633564
(meter) started meter 3 (Czujnik dymu kuchnia 91633564 tsd2)
(wmbus) parseDLL @0 42
(wmbus) parseELL @10 32
(wmbus) parseNWL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(meter) Czujnik dymu kuchnia: for me? 91633564 in 91633564
(meter) Czujnik dymu kuchnia: yes for me
(meter) Czujnik dymu kuchnia tsd2 handling telegram from 91633564
(meter) Czujnik dymu kuchnia 91633564 "294468506435639176F0A0009F2B822C0020822C008022070D317A7C6D6B7C6D008DC3009ED4000FE500"
(wmbus) parseDLL @0 42
(telegram) DLL L=29 C=44 (from meter SND_NR) M=5068 (TCH) A=91633564 VER=76 TYPE=f0 (Smoke detector) (driver tsd2) DEV=cul RSSI=-71
(wmbus) parseELL @10 32
(wmbus) parseNWL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(telegram) TPL CI=a0
telegram=|294468506435639176F0A0|009F2B822C0020822C008022070D317A7C6D6B7C6D008DC3009ED4000FE500|+82
(tsd2) 000   : 29 length (41 bytes)
(tsd2) 001   : 44 dll-c (from meter SND_NR)
(tsd2) 002   : 6850 dll-mfct (TCH)
(tsd2) 004   : 64356391 dll-id (91633564)
(tsd2) 008   : 76 dll-version
(tsd2) 009   : f0 dll-type (Smoke detector)
(tsd2) 010   : a0 tpl-ci-field (Mfct specific)
(tsd2) 012 C!: 2b9f previous date (2021-12-31T02:00:00Z)
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_ID "$METER_JSON""
(shell) env "METER_JSON={"media":"smoke detector","meter":"tsd2","name":"Czujnik dymu kuchnia","id":"91633564","status":"OK","prev_date":"2021-12-31T02:00:00Z","timestamp":"2022-04-05T18:48:04Z","device":"cul","rssi_dbm":-71}"
(shell) env "METER_ID=91633564"
(shell) env "METER_NAME=Czujnik dymu kuchnia"
(shell) env "METER_MEDIA=smoke detector"
(shell) env "METER_TYPE=tsd2"
(shell) env "METER_TIMESTAMP=2022-04-05T18:48:04Z"
(shell) env "METER_TIMESTAMP_UTC=2022-04-05T18:48:04Z"
(shell) env "METER_TIMESTAMP_UT=1649184484"
(shell) env "METER_TIMESTAMP_LT=2022-04-05 20:48.04"
(shell) env "METER_DEVICE=cul"
(shell) env "METER_RSSI_DBM=-71"
(shell) env "METER_STATUS=OK"
(shell) env "METER_PREV_DATE=2021-12-31T02:00:00Z"
(shell) waiting for child 353 to complete.
(shell) /bin/sh: return code 0
(meter) Czujnik dymu schody: for me? 91633564 in 91633566
(meter) Czujnik dymu schody: not for me: not my id
(meter) wodomierz: for me? 91633564 in 74401316
(meter) wodomierz: not for me: not my id
```

Kosygor, 2022-05-06:
Update: after switching to [cul extended frame fw](https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html) it still works

weetmuts, 2022-05-25:
Ha! You are quite right! When I set cul to t1, it can only receive on t1.

But when I set cul to c1, them it recieves both on t1 and c1.

How very curious. I have updated the README. Thanks for the report!

closed 2022-05-25 by weetmuts