wmbusmeters/wmbusmeters #458: Adding support for Itron Cyble LRFV2

rclsilver, 2022-01-30:
Hello,

I just bought a RTL 2838 device to query my Water Meter. The water meter is a Itron Cyble LRFV2. Its frequency seems to be 169MHz.

I started to run the following command to capture cu8 telegrams: rtl_433 -s 1000k -f 169500k -S unknown

You can find first files in the attached zip file [wmbusmeters.zip](https://github.com/weetmuts/wmbusmeters/files/7966716/wmbusmeters.zip).

Do you need more informations to add the support for my device?

Thank you, your project is great!

weetmuts, 2022-01-30:
@merbanan here are some more N mode signals.
@rclsilver Thanks! First we need to decode the radio signal itself, then we decode the bits with wmbusmeters.

merbanan, 2022-01-30:
It was just noise :/

merbanan, 2022-01-30:
@rclsilver it seems this signal is transmitted very very rarely. Try recording for 24h and we'll see if we can find a proper signal.

rclsilver, 2022-01-30:
Thanks for your reply. I'll post 24h data tomorrow !

rclsilver, 2022-01-31:
Hi,

After some hours, I've some files with a different size, maybe it's not only noise for them.
[capture.tar.gz](https://github.com/weetmuts/wmbusmeters/files/7970786/capture.tar.gz)

rclsilver, 2022-02-04:
Hi @merbanan, do you need more data?

merbanan, 2022-02-04:
Just more time.

merbanan, 2022-02-04:
You can use https://triq.org/pdv/ to view the signals. While there are somethings that trigger the rtl_433 recorder I think it is only wide band noise.

These signals seem very hard to record. It might be that there is something that trips up the signal detector. It is very hard to debug as no one has been able to produce a recording of the signal. I might need to get hold of the sensor to be able to get any signals from it.

What should work is to place the rtl-sdr antenna cable without the antenna attached in close proximity to the transmitter.

merbanan, 2022-02-04:
@rclsilver can you buy this, and reship to Sweden?

https://www.ebay.com/itm/283841012536?hash=item42163e2338:g:0rcAAOSw3f5doNQW

They don't ship to Sweden directly. I asked but didn't get any response.

rclsilver, 2022-02-04:
Hey @merbanan where can I contact tout?

merbanan, 2022-02-04:
@rclsilver you can reach me at banan@ludd.ltu.se.

neutrinus, 2022-02-06:
Guys, it seems like Cyble meters uses RADIAN protocol:
http://www.lamaisonsimon.fr/wiki/doku.php?id=maison2:compteur_d_eau:compteur_d_eau

The meter will only transmit if requested.

merbanan, 2022-02-06:
So yeah this device most likely use the RADIAN protocol. Thus not compatible with the wmbus standard.

And because the sensor needs a request from a collector to send its data there is little chance it will be supported by rtl_433.

I think most of the devices on 169MHz will use this old protocol.

merbanan, 2022-02-06:
This issue can be closed as there is nothing to support.

weetmuts, 2022-03-28:
Yes, it looks pretty impossible for us to make this work. Closing this.

closed 2022-03-28 by weetmuts

logon84, 2022-05-19:
Hi. My water company installed yesterday a Cyble LRFv2. From what I see, cyble LRFv2 is not using RADIAN:
(Page 20) https://www.wize-alliance.com › ...PDF
V2 THE WIZE CATALOGUE
Cyble everblue does use RADIAN, but not LRFv2

logon84, 2022-05-19:
> @logon84 well I'm happy to help with decoding if you are able to record a signal.
> 
> https://github.com/merbanan/rtl_433/blob/master/docs/CONTRIBUTING.md#supporting-additional-devices-and-test-data

Thanks, I am still looking for the needed hardware in order to do so. Once I have all I need I'll upload a signal :)

logon84, 2022-05-19:
@merbanan "u" versión of rtl2832 should be ok for the rtl_433 tool?

logon84, 2022-05-23:
While I wait for my SDR device to ship I have been reading about WIZE. I think the capture frequency in capture.tar.gz file by @rclsilver is wrong. From what I see in wize papers, wize uses different channels in different frequencies around 169mhz. See page 6 of:
https://www.wize-alliance.com/content/download/621/file/WIZE%20-%2002%20Regional%20Parameters%20-%20v1.1.pdf
Also bps should be 2400 or 4800.

logon84, 2022-06-03:
I already have the hardware, but while capturing, a lot of files are created whitin seconds.....After analizing signals, I see a big signal at 169.9Mhz which I am not interested in, because Cyble device frequency should be between 169.406250Mhz and 169.468750Mhz. Is there a way to limit the capture bandwith? I am running:

rtl_433 -s 2400k -f 169443750 -S unknown

Thanks

logon84, 2022-06-05:
Hi. I made a 3h capture directly from my flat, close to the window (the rest of the city also has Cyble LRFV2 watermeters, so I think some other watermeter data could have reached my antenna). From what I see on https://triq.org/pdv/, small files don't have data, and in the bigger ones I can see some waves...hope the capture is helpfull
[3h capture.zip](https://github.com/weetmuts/wmbusmeters/files/8840330/3h.capture.zip)
.

jiquem, 2023-10-27:
Did you managee to decode the data ?

jiquem, 2023-11-26:
hello @merbanan, did you manage to decode the data ?

Somel1978, 2024-04-17:
Did this ever worked. I Have this installed but its not getting any data on a NESDR Nano2.