wmbusmeters/wmbusmeters #460: Package for Debian?

petterreinholdtsen, 2022-02-01:
Are you interested in having wmbusmeters packaged in Debian?  I've registered requests for packages for it and its dependency in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=998720 and https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=998721 and could help make it happen, if you are interested.

weetmuts, 2022-02-01:
Yes! I am preparing for a deb package generation! Expect something in this or next week!

weetmuts, 2022-03-28:
Been very busy with paid work (not related to wmbusmeters) will take a look at this again now!

petterreinholdtsen, 2022-04-27:
A good start could be to use 'dh_make' from the dh-make package to set up the initial build rules in debian/.  I just tested by using the 'single package' profile and it seem to build just fine.  Just need to fine tune the files in debian/ to make a complete Debian package.

petterreinholdtsen, 2022-05-04:
https://www.debian.org/doc/manuals/maint-guide/ can be a good place to start to read up on how to get the package into Debian. http://www.hungry.com/~pere/debian-sponsoring.html is my sponsoring preferences.

weetmuts, 2022-09-11:
Ok, I am trying right now, but I have terrible problems with the debhelper-compat 13.

I cannot install it. My Ubuntu is pretty recent Ubuntu 20.04.5 LTS.

Is this related? What is the actual solution?
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=966640

petterreinholdtsen, 2022-09-11:
[Fredrik Öhrström]
> I cannot install it. My Ubuntu is pretty recent Ubuntu 20.04.5 LTS.
>
> Is this related? What is the actual solution?
> https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=966640

I would do 'make debhelper' and figure out what the latest debhelper
level listed there is, and switch to that level.

<URL: https://tracker.debian.org/pkg/debhelper > show version 13 has
been around for a while.

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-09-11:
Ok, I upgraded to Ubuntu 22 now the debhelper problem is gone.
But I am trying to build a native package, so deb/changelog says:
wmbusmeters (1.9.0) unstable; urgency=medium

(I.e. not using 1.9.0-1 that would indicate a package referencing a orig source package.)

However I still get:
dpkg-source: error: can't build with source format '3.0 (quilt)': no upstream tarball found at ../wmbusmeters_1.9.0.orig.tar.{bz2,gz,lzma,xz}

Why is this the case?

petterreinholdtsen, 2022-09-11:
[Fredrik Öhrström]
> dpkg-source: error: can't build with source format '3.0 (quilt)': no
> upstream tarball found at
> ../wmbusmeters_1.9.0.orig.tar.{bz2,gz,lzma,xz}
>
> Why is this the case?

The source format is listed in debian/source/format, and the '3.0
(quilt)' format require a upstream tarball.  If you want to just test
without a upstream tarball, remove the file debian/source/format
temporarily (or add '1.0' or something like that to it).
-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-09-11:
Thank you, but you say temporarily. Should I not build a native package?

petterreinholdtsen, 2022-09-11:
[Fredrik Öhrström]
> Thank you, but you say temporarily. Should I not build a native
> package?

My initial request was for a package _in_ Debian, not _a_ Debian
package.  I believe wmbusmeters as a package in debian should be a
non-native package.  This of course has pros and cons, but I believe it
allow for easier maintenance within Debian.

I assumed you were figuring out how to build _a_ Debian package as a
prerequisite to getting the package uploaded into Debian, and thus
suggested to drop the 3.0 (quilt) format until it was closer to ready to
go into Debian.

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-09-11:
Excellent! Thank you, then I understand better. Should I for example
fetch the orig package from github? Where it is automatically created
when I tag a release.

petterreinholdtsen, 2022-09-11:
[Fredrik Öhrström]
> Excellent! Thank you, then I understand better. Should I for example
> fetch the orig package from github? Where it is automatically created
> when I tag a release.

It is one way to do it.  I am not sure if the tarball generated from a
release tag will be identical every time it is downloaded, so it might
cause problem if you download at two different times and expect to get
the same tarball.  Debian will checksum and enforce the use of identical
tarballs across uploads.

I normally let 'uscan' download tarballs and set up a debian/watch file
to locate the latest release and tarball.  I also normally use
git-buildpackage to handle packaging of non-native packages, but that
might not be a good match if you are doing both upstream and downstream
packaging. :)

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2022-09-12:
I just pushed something that I think might be a working deb generation code. :-)

Checkout the current github master and tag it locally with: "git tag 1.9.1"
(Local testing since I do not want to generate a new release right now.)

Now do "make deb RELEASE=1.9.1"
This will create the packaging/wmbusmeters_1.9.1.orig.tar.gz which seems to be
reproducible since it extracts the tag 1.9.1 and removes the .git directory and 
touches all the files to have the same timestamp as the tagged commit timestamp. 

Then it runs debuild and I signed it using my gpg key. I tested the generated deb and it seems to work. 

I uploaded the files here: https://wmbusmeters.org/releases
(Sorry for the content type being hardcoded to text/html right now, wget works though.)

Which files is it that you want (and find using uscan) to trigger a new deb package deploy to debian?

Attaching my public key.
[oehrstroem_gpg.txt](https://github.com/weetmuts/wmbusmeters/files/9551208/oehrstroem_gpg.txt)

weetmuts, 2022-09-13:
I am using source/format "3.0 (native)" instead of "3.0 (quilt)" since I get a really
weird error when using quilt:
```
add_directory() only handles directories at /usr/share/perl5/Dpkg/Source/Package/V2.pm line 588.
dpkg-buildpackage: error: dpkg-source -b . subprocess returned exit status 29
debuild: fatal error at line 1182:
```

But since the version number is 1.9.1-1 I suppose it still builds a package based on the orig archive
and not a native package? Or.... :-)

petterreinholdtsen, 2022-09-13:
[Fredrik Öhrström]
> But since the version number is 1.9.1-1 I suppose it still builds a
> package based on the orig archive and not a native package? Or.... :-)

I suspect the build tools are confused, as the .dsc file did not link to
the .orig.tar.gz file, and I get this when I try to build it after
downloading and unpacking the source package:

   dpkg-source -ICVS -I.#* -I.cvsignore -I.bzr -I.svn -I.git -b .
  dpkg-source: error: can't build with source format '3.0 (native)':
  native package version may not have a revision
  
  dpkg-buildpackage: error: dpkg-source -ICVS -I.#* -I.cvsignore -I.bzr
    -I.svn -I.git -b . subprocess returned exit status 25
  debuild: fatal error at line 1182:

I also note the .changes file lack the Closes: line.  But I guess this
will be handled in a final upload.

If you want to maintain it in Debian and me to sponsor the upload,
please ping me on IRC and we can figure out how to do the last cleanup.
See <URL: http://www.hungry.com/~pere/debian-sponsoring.html > for
details.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-01-20:
I was just told you plan to close this issue without fixing it.  Is this
correct:

[github-actions[bot]]
> This issue is stale because it has been open for 2 month with no
> activity. Remove stale label or comment or this will be closed in 1
> month.

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2023-01-20:
Nope, this is an effect of the auto-stale robot marker. This issue is still alive!

petterreinholdtsen, 2023-01-20:
[Fredrik Öhrström]
> Nope, this is an effect of the auto-stale robot marker. This issue is
> still alive!

Perhaps time bring the robot under human control?

Where do we stand with this issue, then?

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-03-05:
[Petter Reinholdtsen]
> Where do we stand with this issue, then?

I guess the silence mean you are either busy or no longer interested in
making this package available from the Debian project.  Would be useful
to know which it is. :)

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2023-03-05:
I am interested. I am just busy writing the new driver loading code and doing the normal paid work outside of this project. :-)

petterreinholdtsen, 2023-05-05:
> This issue is stale because it has been open for 2 month with no
> activity. Remove stale label or comment or this will be closed in 1
> month.

I am still interested, even in this feedback indicate that you are not?

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-06-21:
The relevant packages were uploaded to Debian a while back, as can be seen on https://qa.debian.org/developer.php?email=oehrstroem%40gmail.com .  I am happy to help update the packages if approached on IRC, ref my sponsoring preferences.  Sadly the email provider of the package maintainer throw away my emails, so that is not a useful channel.

weetmuts, 2023-06-25:
Thank you Petter! Did you have a chance to upload the bug fixed version 1.13.1-3 ?

https://mentors.debian.net/package/wmbusmeters/

weetmuts, 2023-06-25:
You write "email provider of the package maintainer throw away my emails, "

I.e. do you mean that you get an error response from gmail when you send emails to oehrstroem@gmail.com?

The last email I sent to pere@debian.org was on the 26 of may this year.

petterreinholdtsen, 2023-06-25:
[Fredrik Öhrström]
> Thank you Petter! Did you have a chance to upload the bug fixed
> version 1.13.1-3 ?

I got your email about it, and even replied via email despite prefering
IRC to avoid gathering any backlog. The email was rejected, so I have
been waiting for you to show up on IRC according to my Debian package
sponsoring preferences ever since, as you lack the ability to receive my
email.

-- 
Happy hacking
Petter Reinholdtsen

weetmuts, 2023-06-27:
As is obvious from my email address, I use gmail, a huge service from Google with 1.8 billion users. Claiming that I lack the ability to receive your emails (and that is somehow my fault) is just silly. If you get a rejection, it is something on your side, in particular since you have successfully sent me emails before.

I only sent the email because I was not sure if the IRC message had reached you, since you do not store any IRC communications directed to you when you are disconnected. I did comply with your preference for IRC but I cannot mind read you, to figure out that I have to connect to IRC because you want to tell me something.

If you want to continue sponsoring me, then we can proceed in the following way:
1) You fix your email provider so that you can email me, at least to tell me to connect to IRC.
2) If you also want to contribute to the package maintainence, then you submit pull requests to this github repo or to the debian salsa repo.

If this is not possible, then the most efficient use of our time is to depart ways, you stop sponsoring rtl-wmbus and wmbusmeters, and I will search for a new sponsor who is willing to use email.

petterreinholdtsen, 2023-06-27:
[Fredrik Öhrström]
> If this is not possible, then the most efficient use of our time is to
> depart ways, you stop sponsoring rtl-wmbus and wmbusmeters, and I will
> search for a new sponsor who is willing to use email.

Feel free to look for other sponsors that fit your work flow better, if
my preferences as documented on
<URL: http://www.hungry.com/~pere/debian-sponsoring.html > do not work
for you.

-- 
Happy hacking
Petter Reinholdtsen

closed 2023-09-26 by github-actions[bot]