wmbusmeters/wmbusmeters #462: Telegrams for Qundis smoke detector

kongo09, 2022-02-01:
https://github.com/weetmuts/wmbusmeters/blob/70b7e512314151ca0c0c6b7637620dee9f0141ca/src/meter_qsmoke.cc#L349

I noticed in the code that the status for the Qundis smoke detectors is not yet understood. While I don't have such a device myself, I do receive Qundis smoke detector telegrams. I'm not sure if that helps you guys in completing the driver, but I just share some here. If this is not helpful, simply close this issue, please.

```
Received telegram from: 47858593
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Smoke detector (0x1a)
                   ver: 0x23
                device: rtlwmbus[00000001]
                  rssi: 16 dBm
                driver: qsmoke
telegram=|#3744934493858547231A7AEB00002081027C034955230082026CFFFF81037C034C41230082036CFFFF02FD170000326CFFFF046D3115C122|+841
```
or
```
Received telegram from: 47858586
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Smoke detector (0x1a)
                   ver: 0x23
                device: rtlwmbus[00000001]
                  rssi: 25 dBm
                driver: qsmoke
telegram=|#3744934486858547231A7AEF00002081027C034955230082026CFFFF81037C034C41230082036CFFFF02FD170000326CFFFF046D3315C122|+1008
```
or
```
Received telegram from: 47858588
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Smoke detector (0x1a)
                   ver: 0x23
                device: rtlwmbus[00000001]
                  rssi: 31 dBm
                driver: qsmoke
telegram=|#3744934488858547231A7A0C00002081027C034955230082026CFFFF81037C034C41230082036CFFFF02FD170000326CFFFF046D3515C122|+1180

BIBOLV, 2023-01-02:
Thank you for messages but status fields in those telegrams are standard ones. If you manage to capture telegrams with not implemented status fields and also know what those represent - please reopen issue and provide that information. Closing issue for now.

closed 2023-01-02 by BIBOLV

weetmuts, 2023-01-02:
Yep, as Bibo wrote,  the driver has been improved since your issue and fully decodes the telegram you provided. Thanks for the telegrams!