wmbusmeters/wmbusmeters #463: Missing data for eBZ wMB-E01 (ebzwmbe)

LordShelmchen, 2022-02-02:
These are my first attempts with the M-Bus system, so it may well be that I have overlooked something somewhere.

My electricity meter is a eBZ DD3 with a eBZ wMB-E01 (ebzwmbe) M-Bus module.

Unfortunately not all data is read out, only the value of total_energy_consumption_kwh

Command:

> wmbusmeters /dev/ttyUSB0:t1 powermeter ebzwmbe 00001234 11223344556677889900112233445566

Output:

> Started config im871a[00103079] on /dev/ttyUSB0 listening on t1
> powermeter    12345678        200.58 kWh      0 kW    0 kW    0 kW    0 kW    2022-02-02 21:17.28
> powermeter    12345678        200.585 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-02 21:17.48
> powermeter    12345678        200.59 kWh      0 kW    0 kW    0 kW    0 kW    2022-02-02 21:18.08

wmbusmeters recognizes this configuration:

Command:

> wmbusmeters /dev/ttyUSB0:t1

Output:

> Started config im871a[00103079] on /dev/ttyUSB0 listening on t1
> No meters configured. Printing id:s of all telegrams heard!
> Received telegram from: 00001234
>           manufacturer: (EBZ) eBZ, Germany (0x145a)
>                   type: Radio converter (meter side) (0x37) encrypted
>                    ver: 0x02
>       Concerning meter: 12345678
>           manufacturer: (EBZ) eBZ, Germany (0x145a)
>                   type: Electricity meter (0x02) encrypted
>                    ver: 0x01
>                 device: im871a[00103079]
>                   rssi: -79 dBm
>                 driver: ebzwmbe

What am I doing wrong? Or is it a bug?

weetmuts, 2022-02-03:
You have to provide a decrypted telegram. Add --logtelegrams to the command line and post a few here. Thanks!

LordShelmchen, 2022-02-03:
No problem, i hope it will help to solve the problem:

> Started config im871a[00103079] on /dev/ttyUSB0 listening on t1
> telegram=|4B445A146450000002378C20F3900F002C25F38B00006CBB07CEB215901172635213015A140102F3002007102F2F#04039D3E03000429E82201000DFD110E333632353331313031305A424531|+45
> Stromzaehler    01234567        212.637 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-03 14:04.07
> telegram=|4B445A146450000002378C20F4900F002C25F48B00008A60B0539EA2ADC572635213015A140102F4002007102F2F#0403A13E03000429A11A01000DFD110E333632353331313031305A424531|+65
> Stromzaehler    01234567        212.641 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-03 14:04.27
> telegram=|4B445A146450000002378C20F5900F002C25F58B0000E20EFFEC89642DDC72635213015A140102F5002007102F2F#0403A63E03000429101F01000DFD110E333632353331313031305A424531|+85
> Stromzaehler    01234567        212.646 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-03 14:04.47
> telegram=|4B445A146450000002378C20F6900F002C25F68B00006AEA1376FA7A6E0B72635213015A140102F6002007102F2F#0403AA3E03000429C51701000DFD110E333632353331313031305A424531|+105
> Stromzaehler    01234567        212.65 kWh      0 kW    0 kW    0 kW    0 kW    2022-02-03 14:05.07
> telegram=|4B445A146450000002378C20F7900F002C25F78B00009068150A7C2089A672635213015A140102F7002007102F2F#0403AE3E03000429871C01000DFD110E333632353331313031305A424531|+125
> Stromzaehler    01234567        212.654 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-03 14:05.27
> telegram=|4B445A146450000002378C20F8900F002C25F88B0000AEE6DE05C01D1F3E72635213015A140102F8002007102F2F#0403B23E03000429C02101000DFD110E333632353331313031305A424531|+145
> Stromzaehler    01234567        212.658 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-03 14:05.47
> telegram=|4B445A146450000002378C20F9900F002C25F98B0000A09D04547AE92F3272635213015A140102F9002007102F2F#0403B63E03000429600F01000DFD110E333632353331313031305A424531|+165
> Stromzaehler    01234567        212.662 kWh     0 kW    0 kW    0 kW    0 kW    2022-02-03 14:06.07

weetmuts, 2022-02-03:
Thanks! If you run the command: 
wmbusmeters --analyze 4B445A146450000002378C20F3900F002C25F38B00006CBB07CEB215901172635213015A140102F3002007102F2F#04039D3E03000429E82201000DFD110E333632353331313031305A424531 Elen ehzp 00005064 NOKEY

You will see that your telegram only contains:
048 C!: 9D3E0300 total energy (212.637000 kwh)
054 C!: E8220100 current power (0.744720 kw)
(and a customer string)

So use ehzp for the moment. I need to think about why the meter uses the same mfct/version/media as ebzwmbe. 
Clearly these meters are always a combo of wmbus radio converter plugin + a meter and they might have multiple meter configurations that all report the same meter version but with widely different content. 

We need a more flexibel way of handling these kind of meters.... I think I have an idea to make fields optional.

LordShelmchen, 2022-02-03:
Thanks a lot for help! Thats works for me.

If i should ask my electricity provider something about the module/meter, let me know.

BIBOLV, 2023-01-02:
Closing issue as provided solution works.

closed 2023-01-02 by BIBOLV