wmbusmeters/wmbusmeters #464: Unable to decrypt amiplus meter

a-nych, 2022-02-03:
Hello, while trying to read my amiplus wmbus data I get the following error:
```
(meter) dom: for me? 86003979 in
(meter) dom: yes for me
(meter) constructed "dom" "amiplus" "86003979" encrypted
(meter) used meter template dom  auto to match 86003979
(meter) started meter 1 (dom 86003979 auto)
(wmbus) parseDLL @0 81
(wmbus) parseELL @10 71
(wmbus) parseNWL @10 71
(wmbus) parseAFL @10 71
(wmbus) parseTPL @10 71
(meter) dom: for me? 86003979 in 86003979
(meter) dom: yes for me
(meter) dom amiplus handling telegram from 86003979
(meter) dom 86003979 "504401067939008602027A000340053B3C5161D46844D2ADD501E69D389913CF6979ED7F15F63763EC3A65357F981F512400E6B3375E49EE193EEC4483534D5DBACC4E0B800E38FEEB06E9548F9D49E479"
(wmbus) parseDLL @0 81
(telegram) DLL L=50 C=44 (from meter SND_NR) M=0601 (APA) A=86003979 VER=02 TYPE=02 (Electricity meter) (driver amiplus) DEV=rtl433[cmd_0] RSSI=999
(wmbus) parseELL @10 71
(wmbus) parseNWL @10 71
(wmbus) parseAFL @10 71
(wmbus) parseTPL @10 71
(TPL) AES CBC IV decrypting "3B3C5161D46844D2ADD501E69D389913CF6979ED7F15F63763EC3A65357F981F512400E6B3375E49EE193EEC4483534D5DBACC4E0B800E38FEEB06E9548F9D49E479"
(TPL) num encrypted blocks 4 (64 bytes and remaining unencrypted 2 bytes)
(TPL) IV 01067939008602020000000000000000
(TPL) decrypted  "A4FC04A409BDB47F43465A937A15944A0CD0D2AA38D2BB7F0E5DAED0FB3BF4F5CBDCF4D44E432B3A513CD49EFEFC3FFC247D33CB944B063B16A14A311443D17B"
(TPL) appended   "A4FC04A409BDB47F43465A937A15944A0CD0D2AA38D2BB7F0E5DAED0FB3BF4F5CBDCF4D44E432B3A513CD49EFEFC3FFC247D33CB944B063B16A14A311443D17BE479"
(wmbus) WARNING! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 86003979 mfct: (APA) Apator, Poland (0x601) type: Electricity meter (0x02) ver: 0x02
(telegram) TPL CI=7a ACC=00 STS=03 CFG=0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0)
(meter) newly created meter (dom 86003979 amiplus) did not handle telegram!
(wmbus) telegram from 86003979 ignored by all configured meters!
```

The decryption key is copy-pasted from my meters website so it should be correct.

pswiatki, 2022-02-03:
Not sure, but if your provider is TAURON then I had an issue with their first key. Only when I requested a key refresh the second key started working with `wmbusmeters`. Hope it helps.

a-nych, 2022-02-03:
@pswiatki yes it is - how did you request a key refresh? Did you deactivate and reactive the HAN functionality on `elicznik`?

pswiatki, 2022-02-03:
It was over two years ago and I don't remember any more (the meter does not belong to me, I was able, however, to get some help from the landlord. I remember it was doable directly from the `elicznik` website. Perhaps deactivation / reactivation.

a-nych, 2022-02-04:
Yes, deactivation/reactivation gave me the correct encryption key. Thanks for the help @pswiatki!

closed 2022-02-04 by a-nych