wmbusmeters/wmbusmeters #466: telegram contains a register with unknown size - apator-m-08

llastowski, 2022-02-04:
I have been testing Apator-08 water meter and the telegram is not supported by the driver - 61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216

weetmuts, 2022-02-07:
You have to give some more info. What is the expected consumption for that telegram?
Which driver do you use?

Santa77, 2022-04-05:
### TestCase:

**CMD1:**

`
../wmbusmeters-dev/build/wmbusmeters --debug 61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216
`

from result u can get 

`
Received telegram from: 000118b7
          manufacturer: (APT) Unknown (0x8614)
                  type: Gas meter (0x03)
                   ver: 0x03
                driver: apator08
`

so u create simulation_97.txt

`
telegram=||61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216|`

**CMD2:**

`santa@DESKTOP-HRQ5FG0:~/wsl/mqtt-sender$ ../wmbusmeters-dev/build/wmbusmeters --debug simulation_97.txt MyMeter apator08 000118b7 00000000000000000000000000000000
`

result:

`
(wmbusmeters) version: 1.7.0
(config) using device: simulation_97.txt:any
(config) number of meters: 1
(lookup) with file/hex "simulation_97.txt" unknown
(lookup) driver: simulation file
(main) opening simulation_97.txt:any
Started config simulation on simulation_97.txt listening on any
(simulation) in simulation_97.txt
(main) regular reset of simulation  on simulation_97.txt will happen every 82800 seconds
(main) added simulation_97.txt to files
(wmbus) no alarm (expected activity) for simulation
(main) simulation_97.txt already configured as simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216"
(mbus) checkMBUSFrame
 "61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216"
(wmbus) checkWMBUSFrame "61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216"
(simulator) is_mbus=false is_wmbus=true
(simulator) is wmbus telegram.
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 98
(wmbus) parseELL @10 88
(wmbus) parseNWL @10 88
(wmbus) parseAFL @10 88
(wmbus) parseTPL @10 88
(meter) MyMeter: for me? 000118b7 in
(meter) MyMeter: yes for me
(meter) constructed "MyMeter" "apator08" "000118b7" encrypted
(meter) used meter template MyMeter  auto to match 000118b7
(meter) started meter 1 (MyMeter 000118b7 auto)
(wmbus) parseDLL @0 98
(wmbus) parseELL @10 88
(wmbus) parseNWL @10 88
(wmbus) parseAFL @10 88
(wmbus) parseTPL @10 88
(meter) MyMeter: for me? 000118b7 in 000118b7
(meter) MyMeter: yes for me
(meter) MyMeter apator08 handling telegram from 000118b7
(meter) MyMeter 000118b7 "61441486B71801000303A0A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216"
(wmbus) parseDLL @0 98
(telegram) DLL L=61 C=44 (from meter SND_NR) M=8614 (APT) A=000118b7 VER=03 TYPE=03 (Gas meter) (driver apator08) DEV= RSSI=0
(wmbus) parseELL @10 88
(wmbus) parseNWL @10 88
(wmbus) parseAFL @10 88
(wmbus) parseTPL @10 88
(telegram) TPL CI=a0
(apator08) 000   : 61 length (97 bytes)
(apator08) 001   : 44 dll-c (from meter SND_NR)
(apator08) 002   : 1486 dll-mfct (APT)
(apator08) 004   : b7180100 dll-id (000118b7)
(apator08) 008   : 03 dll-version
(apator08) 009   : 03 dll-type (Gas meter)
(apator08) 010   : a0 tpl-ci-field (Mfct specific)
(apator08) 011 C?: A4000000B8412F0000000000002C000000000000280000000000002700000024120F0A170910233D31562820160F0A1709101FD0717700304F464E371C08030000000005FF0182000000348B880F0200292A0A05040216 mfct specific
(apator08) 025 C!: *** 10|a4000000 total consumption (0.054667 m3)
MyMeter 000118b7        0.054667 m3     2022-04-05 19:39.41
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation
`

looks ok for me....

`
MyMeter 000118b7        0.054667 m3     2022-04-05 19:39.41
`

weetmuts, 2022-05-25:
Thanks @Santa77 ! You can also run the same analyze command on the web, like this:
https://wmbusmeters.org/analyze/61441486B71801000303a0a4000000B8412f0000000000002c000000000000280000000000002700000024120f0a170910233d31562820160f0a1709101fd0717700304f464e371c08030000000005ff0182000000348B880f0200292a0a05040216

@llastowski your telegram is detected as an apator-08 and a value is extracted.
Is it the right value?

From the title you mention "register with unknown size" but that error message comes from 
the apator162 driver. Do you have both apator08 and apator162 meters?

weetmuts, 2022-05-25:
@Santa77 So the local analyze command looks like this:
wmbusmeters --analyze <hex>
or if you need a key:
wmbusmeters --analyze=<key> <hex> 
or if you want to force a driver
wmbusmeters --analyze=<driver> <hex> 
or both:
wmbusmeters --analyze=<driver>:<key> <hex>

llastowski, 2022-05-26:
> Thanks @Santa77 ! You can also run the same analyze command on the web, like this: [...]

Thanks for the analysis - it finally turned out that this was a apator-08 sensor advertised as apator-16-2 and arrived without the model serial number. This is where all the confusion came from :)
Apator-08 is EOL and no new units are sold, I will continue with 16-2 version which is fully supported.

llastowski, 2022-05-27:
yes - thanks!

closed 2022-05-27 by llastowski