wmbusmeters/wmbusmeters #467: Support for Qundis Warm Water (30°C-90°C) meter

heeplr, 2022-02-04:
wmbusmeters told me to open an issue, so of course i'll do.

I use rtl_433 to decode all my meters and wmbusmeters to decode datagrams. This one isn't supported.

```
./build/wmbusmeters 374493444836351218067ac70000200c13911900004c1391170000426cbf2ccc081391170000c2086cbf2c02bb560000326cffff046d1e02de2
1fed0
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 12353648
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x18
                driver: unknown!

```

Here are datagrams collected by rtl_433 including the telegrams as "data" attribute: [qundis.txt](https://github.com/weetmuts/wmbusmeters/files/8002890/qundis.txt)

 I hope that's useful.

geektoor, 2022-02-05:
I have the same issue, also for some "cold" water meters. They use manufacture code 0x4493, type 0x07 and ver 0x18 (so only the type is different by those meters, warm uses 0x06 and cold 0x07). But if i read the code correctly, the matching is already there: https://github.com/weetmuts/wmbusmeters/blob/master/src/meter_detection.h#L69
I'm currently using the docker container as i'm testing around, but the docker version states its 1.6.0-51-gf04434f, so it should be recent enough to contain the code.

BTW. you can set the driver to lse_07_17 for those meters, works fine (at least the values i could compare from display to telegrams match). So it just seems to be an issue with the detection.

heeplr, 2022-02-06:
Indeed it works with the lse_07_17 driver. Autodetection would be nice, tho.
I'll leave this open for now. Maybe close this issue if autodetection isn't possible.

weetmuts, 2022-02-09:
Hi! This new telegram is different enough to warrant a new driver and I have pushed one.

The new driver format is self contained and a single .cc file. This makes it much easier to add new drivers. Here is the new one: https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_qwater.cc

Autodetection works with this new driver. Please test!

weetmuts, 2022-02-09:
It seems like the qwater sends two types of telegrams. You can check the contents like this:
wmbusmeters  --analyze 374493444836351218067ac70000200c13911900004c1391170000426cbf2ccc081391170000c2086cbf2c02bb560000326cffff046d1e02de21fed0

and 

wmbusmeters --analyze 47449344483635121806780dff5f350082db00005e0107c113ffff48200000bf2c91170000df2120200000008001000000060019001000160018000d001300350017002f046d030dc422eea7

The driver handles them both, because the second contains mostly proprietary data so nothing can be extracted right now from it. Probably the same content as the other telegram.

heeplr, 2022-02-09:
Seems to works for me. I'll start using it in my pipeline and collect data. 
Thank's a ton for your work!

closed 2022-02-09 by heeplr