wmbusmeters/wmbusmeters #468: Support for ITRON Water meter

heeplr, 2022-02-04:
```
$ ./build/wmbusmeters 384497263003112103077a760000a00413b7130000066d2e17a9c422004413300f0000426cbf2c047f0000060c027f852a0e79678372082100
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 21110330
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Water meter (0x07)
                   ver: 0x03
                driver: unknown!
```

datagrams collected by rtl_433:  [itron.txt](https://github.com/weetmuts/wmbusmeters/files/8002909/itron.txt)

weetmuts, 2022-02-06:
@heeplr Did you see that Bibo create a driver for you?

heeplr, 2022-02-06:
Yes, and it works like a charm. Thank's a ton @BIBOLV !

closed 2022-02-06 by heeplr