wmbusmeters/wmbusmeters #469: Diehl IZAR 868 I not recognized

Chasbrot, 2022-02-04:
Hi! 
I have a IZAR RC 868 I and it is not recognized. I don't know if it's either not supported, a bug or a simple configuration problem.

### Auto-Detection:
sudo rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus | wmbusmeters --logtelegrams stdin:rtlwmbus
Received telegram from: 19011957
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x78
                device: rtlwmbus[]
                  rssi: 99 dBm
                driver: unknown!
telegram=|#1944A511780657190119A2410E00131700225E78428FF2111518|+21

### If i force the izar driver:
(meter) MyTapWater: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) Not printing this warning again for id: 19011957 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x78
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|1944A511780657190119A2#210E00131771C1EB82866726CB4317|+3
MyTapWater      19011957                000000  91.006 m3       91.122 m3       2022-01-31      7 y     no_alarm    no_alarm 8 s     0       2022-02-04 12:42.41

### Versions:
wmbusmeters: 1.6.0-50-g70b7e51
70b7e512314151ca0c0c6b7637620dee9f0141ca
rtl_wmbus: d2be82c
d2be82cfa2180efb19937ad806dce62b5fe846a0

### The meter:
[image]

Is it possible to add support? Thank you in advance!

weetmuts, 2022-03-28:
Thank you! I have now pushed support for this meters mfct/media/version combo to wmbusmeters. Please test!

Chasbrot, 2022-03-28:
Hi, thank you, now the meter is detected.

sudo rtl_sdr -f 868.93M -s 1600000 - 2>/dev/null | rtl_wmbus | wmbusmeters stdin:rtlwmbus
Received telegram from: 19011957
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x78
                device: rtlwmbus[]
                  rssi: 110 dBm
                driver: izar

sudo rtl_sdr -f 868.93M -s 1600000 - 2>/dev/null | rtl_wmbus | wmbusmeters stdin:rtlwmbus MyTapWater izar 19011957 NOKEY
Started config rtlwmbus on stdin listening on any
MyTapWater      19011957                000000  89.218 m3       90.315 m3       2022-02-28      6.5 y   no_alarm        no_alarm        8 s     0       2022-03-28 14:38.33

closed 2022-04-15 by weetmuts