wmbusmeters/wmbusmeters #471: Problem Getting Telegrams from WaterStar M

SWAT9119, 2022-02-08:
I have a cul which should recieve telegrams from a WaterStar M via c1

This is the config:
`loglevel=debug
device=/dev/ttyUSB0:c1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
`

This is the Meterfile-Config (at etc/wmbuseters.d/
`name=EG_Warmwasser_Kueche
type=waterstarm
id=22029791
key=XXXXXXXXXXXXXXXXXXXX` (in the config is the real AES-Key)

This is the log:
`(wmbusmeters) version: 1.6.0-2-g1ba5660
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: /dev/ttyUSB0:c1 
(config) number of meters: 1
(lookup) with file/hex "/dev/ttyUSB0"
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary "000A"
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
[... 251 lines cut ...]
(cul) checkCULFrame "bA946FA12919702220206F61A7AA40090052D89CE38B7B303E3321DD340827B1F46DA4D900D88B2139784C8C9B811A46C0E01D4E3E8EB577932A2A790276309FCCA0F8B"
(cul) no eol found yet, partial frame
(serial) received binary "43343145334339413439414134343632453843384532414536444546423535303245413539373339303333393036453836464333394546333430423744"
(cul) checkCULFrame "bA946FA12919702220206F61A7AA40090052D89CE38B7B303E3321DD340827B1F46DA4D900D88B2139784C8C9B811A46C0E01D4E3E8EB577932A2A790276309FCCA0F8BC41E3C9A49AA4462E8C8E2AE6DEFB5502EA59739033906E86FC39EF340B7D"
(cul) no eol found yet, partial frame
(serial) received binary "3244464231303137304544393735423334443442443843353834333942443735434334464144303043384431434443374435443437383343413933314635373736413739343036443046373832383533324238433644323644464243354343443734304335450D0A"
(cul) checkCULFrame "bA946FA12919702220206F61A7AA40090052D89CE38B7B303E3321DD340827B1F46DA4D900D88B2139784C8C9B811A46C0E01D4E3E8EB577932A2A790276309FCCA0F8BC41E3C9A49AA4462E8C8E2AE6DEFB5502EA59739033906E86FC39EF340B7D2DFB10170ED975B34D4BD8C58439BD75CC4FAD00C8D1CDC7D5D4783CA931F5776A79406D0F7828532B8C6D26DFBC5CCD740C5E<0D><0A>"
(cul) checkCULFrame RSSI_RAW=94
(cul) checkCULFrame LQI=6
(cul) bad hex "A946FA12919702220206F61A7AA40090052D89CE38B7B303E3321DD340827B1F46DA4D900D88B2139784C8C9B811A46C0E01D4E3E8EB577932A2A790276309FCCA0F8BC41E3C9A49AA4462E8C8E2AE6DEFB5502EA59739033906E86FC39EF340B7D2DFB10170ED975B34D4BD8C58439BD75CC4FAD00C8D1CDC7D5D4783CA931F5776A79406D0F7828532B8C6D26DFBC5CCD74"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 cul
(serialtty) closed /dev/ttyUSB0 (cul)
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted cul
(serial) closing 1 devices
`

What Do I wrong?

SWAT9119, 2022-02-08:
Update:

When I deactivate the AES-Encryption in the meter, I got follow log:

`root@odroid:/etc# wmbusmeters --analyze --debug --logtelegrams /dev/ttyUSB0:cul:c1
(wmbusmeters) version: 1.6.0-2-g1ba5660
(config) using device: /dev/ttyUSB0:cul:c1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0"
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary "000A"
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "3F2028F8807820697320756E6B6E6F776E2920557365206F6E65206F66204220622043206520462047204B206C204D20522054207420562057205820780D0A"
(cul) probe response "? (<F8><80>x is unknown) Use one of B b C e F G K l M R T t V W X x<0D><0A>"
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "5620312E3637206E616E6F43554C3836385F723536380D0A"
(cul) probe response "V 1.67 nanoCUL868_r568<0D><0A>"
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? yes
(main) opening /dev/ttyUSB0:cul:c1
Started config cul on /dev/ttyUSB0 listening on c1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
[... 8 lines cut ...]
(cul) received "CMODE"(serial /dev/ttyUSB0) sent "5832310A0D"
(serial) EVENT thread interrupted
(wmbus) no alarm (expected activity) for cul
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received binary "6241393436464131323334323130"
(cul) checkCULFrame "bA946FA1234210"
(cul) no eol found yet, partial frame
(serial) received binary "32323230323036454134393741323630303930303543393744353130374134454631443132353337454642424541453034304331443041394344444541"
(cul) checkCULFrame "bA946FA12342102220206EA497A26009005C97D5107A4EF1D12537EFBBEAE040C1D0A9CDDEA"
(cul) no eol found yet, partial frame
(serial) received binary "413034463537463646464446353742353742313534323332343231313144433432363631364137424332343633313544433246413146463735314643304331374131453044334345333144354644353031383630393136374235414644444643453139353536444434303134424235453545443245433837"
(cul) checkCULFrame "bA946FA12342102220206EA497A26009005C97D5107A4EF1D12537EFBBEAE040C1D0A9CDDEAA04F57F6FFDF57B57B15423242111DC426616A7BC246315DC2FA1FF751FC0C17A1E0D3CE31D5FD5018609167B5AFDDFCE19556DD4014BB5E5ED2EC87"
(cul) no eol found yet, partial frame
(serial) received binary "42333739343945393539423341333530344634444437334241423734344632463143433042383043373237314145424344363137323443383035423530"
(cul) checkCULFrame "bA946FA12342102220206EA497A26009005C97D5107A4EF1D12537EFBBEAE040C1D0A9CDDEAA04F57F6FFDF57B57B15423242111DC426616A7BC246315DC2FA1FF751FC0C17A1E0D3CE31D5FD5018609167B5AFDDFCE19556DD4014BB5E5ED2EC87B37949E959B3A3504F4DD73BAB744F2F1CC0B80C7271AEBCD61724C805B50"
(cul) no eol found yet, partial frame
[... cut ...] 594B8FD61577C9C71E69E<0D><0A>"
(cul) checkCULFrame RSSI_RAW=-98
(cul) checkCULFrame LQI=115
(cul) bad hex "A946FA12342102220206EA497A26009005C97D5107A4EF1D12537EFBBEAE040C1D0A9CDDEAA04F57F6FFDF57B57B15423242111DC426616A7BC246315DC2FA1FF751FC0C17A1E0D3CE31D5FD5018609167B5AFDDFCE19556DD4014BB5E5ED2EC87B37949E959B3A3504F4DD73BAB744F2F1CC0B80C7271AEBCD61724C805B50F930DFB24E32218A721AF594B8FD61577C9C71"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
`

weetmuts, 2022-02-08:
Nowadays most telegrams are encrypted. The reason the mac fails is probably because you had not supplied the right key? Did it work when you used mode 5?

The previous waterstarm I have seen use mode 5 with a couple of bytes at then end which are not encrypted. Now your telegrams use the more advanced mode 7 with mac etc, but from the the headers it looks like they are encrypted.

Please double check the key. Perhaps set it to something obvious like all zeroes first. Then try both mode 5 and 7.

SWAT9119, 2022-02-08:
Thank you for your fast answer!

I have set the AES_5 and AES_7 Mode - the result you can find in the attachment. With AES 5 - Mode I get less information than with AES 7.

The Key is sure the right - I set it with the software from the manufacter

Any idea?

[wmbusmeters_AES7.log](https://github.com/weetmuts/wmbusmeters/files/8026120/wmbusmeters_AES7.log)
[wmbusmeters_AES5.log](https://github.com/weetmuts/wmbusmeters/files/8026121/wmbusmeters_AES5.log)
.

weetmuts, 2022-02-08:
Thank you! It just looks like it is the wrong key.

I already have a test case with the waterstarm aes 5 encrypted which works.  So my guess is that the key is wrong.

Have you tested to set it to something easy like: 00112233445566778899AABBCCDDEEFF

weetmuts, 2022-02-08:
Or if the ordering is different in the admin tool, you might try the reverse 
FFEEDDCCBBAA99887766554433221100

SWAT9119, 2022-02-08:
I´ve tried your key - now I got following message:

`root@odroid:~# wmbusmeters --verbose --debug --logtelegrams --useconfig=/
(config) "device" "/dev/ttyUSB0:cul:c1"
(config) "logtelegrams" "true"
(config) "format" "json"
(config) "meterfiles" "/var/log/wmbusmeters/meter_readings"
(config) "meterfilesaction" "overwrite"
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log"
(config) "" ""
(config) loading meter file /etc/wmbusmeters.d/EG_Warmwasser_Kueche
(config) name=EG_Warmwasser_Kueche
(config) type=waterstarm
(config) id=22029791
(config) key=<notprinted>
(cmdline) setting link modes to c1,t1 for meter EG_Warmwasser_Kueche
**(wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 18446744073709545356) > this->size() (which is 180)**
`

weetmuts, 2022-02-08:
Can you post a telegram encrypted with the new key? Can you run just from the command line and skip the config:

wmbusmeters --debug --logtelegrams /dev/ttyUSB0:c1 MyMeter waterstarm 22029791 00112233445566778899AABBCCDDEEFF

weetmuts, 2022-02-08:
Have you built the latest version from github?

SWAT9119, 2022-02-08:
OK - I´ve set a new key for a several times - now it seems to work with AES-5-encryption, but not with AES 7.

Yes I have the latest version from github.

Thank you for your fast help!!!

weetmuts, 2022-02-08:
Great!

I really would like to make sure it works with aes 7 as well. This is the first time I see the waterstarm sending mode 7.
Can you try again with mode 7?

weetmuts, 2022-02-08:
It would be very valuable if you could send me a mode 5 telegram encrypted with 001122334...
and a mode 7 telegram encrypted with 001123344....

SWAT9119, 2022-02-08:
No problem - you can find it in the attachment...

Do you need any further data?
[AES5.txt](https://github.com/weetmuts/wmbusmeters/files/8026827/AES5.txt)
[AES7.txt](https://github.com/weetmuts/wmbusmeters/files/8026828/AES7.txt)

weetmuts, 2022-02-08:
Awesome! There was indeed a bug in mode 7 when it was used on a telegram that has non-encrypted data at the end.
(in this case it is the model/verison and parameter set.
Fix pushed. Now it decrypts your mode 7 telegrams you can test it like this if you like:

./build/wmbusmeters --analyze=00112233445566778899AABBCCDDEEFF 4E46FA129197022202068C0003900F002C256C0000006B9DBFDB10E2AA677A3400200710BA9337856E2FDA09B4FB2AD7E25CC8C0875B84C1DFF0240FBFEF3223E5E0B70403FD0C08000002FD0B0011

SWAT9119, 2022-02-09:
I have executed the command in /etc/wmbusmeters - I got follow message - but I don´t recieve readable telegrams - with AES5 it works...

`root@odroid:/etc/wmbusmeters# ./build/wmbusmeters --analyze=00112233445566778899AABBCCDDEEFF 4E46FA129197022202068C0003900F002C256C0000006B9DBFDB10E2AA677A3400200710BA9337856E2FDA09B4FB2AD7E25CC8C0875B84C1DFF0240FBFEF3223E5E0B70403FD0C08000002FD0B0011

Using driver "waterstarm" based on mfct/type/version driver lookup table.
But a better match could perhaps be driver "mkradio3" with 6/26 content bytes understood.
`

Do I have to change something else?

SWAT9119, 2022-02-09:
When I manual decrypt the recieved data, I get follow payload:

`(wmbus) Unknown tpl-ci-field 36
`

weetmuts, 2022-02-09:
do: git pull; make clean; make

i rhink your bknary is not up to date

weetmuts, 2022-02-09:
When I run the analyze command I get:
```
Auto driver  : waterstarm
Best driver  : waterstarm 19/19
Using driver : waterstarm 19/19
000   : 4e length (78 bytes)
001   : 46 dll-c (from meter SND_IR)
002   : fa12 dll-mfct (DWZ)
004   : 91970222 dll-id (22029791)
008   : 02 dll-version
009   : 06 dll-type (Warm Water (30°C-90°C) meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 00 ell-cc (slow_resp)
012   : 03 ell-acc
013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
014   : 0f afl-len (15)
015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
018   : 6c000000 afl-counter (108)
022   : 6B9DBFDB10E2AA67 afl-mac 8 bytes
030   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
031   : 34 tpl-acc-field
032   : 00 tpl-sts-field (OK)
033   : 2007 tpl-cfg 0720 (AES_CBC_NO_IV )
035   : 10 tpl-cfg-ext (KDFS=1)
036   : 2f2f decrypt check bytes
038   : 04 dif (32 Bit Integer/Binary Instantaneous value)
039   : 6D vif (Date and time type)
040 C!: 1634C822 at date (2022-02-08 20:22)
044   : 04 dif (32 Bit Integer/Binary Instantaneous value)
045   : 13 vif (Volume l)
046 C!: 6D000000 total consumption (0.109000 m3)
050   : 02 dif (16 Bit Integer/Binary Instantaneous value)
051   : FD vif (Second extension FD of VIF-codes)
052   : 17 vife (Error flags (binary))
053 C!: 0000 info codes (OK)
055   : 04 dif (32 Bit Integer/Binary Instantaneous value)
056   : 93 vif (Volume l)
057   : 3C vife (backward flow)
058 C!: 00000000 total water backwards (0.000000 m3)
062   : 2F skip
063   : 2F skip
064   : 2F skip
065   : 2F skip
066   : 2F skip
067   : 2F skip
068   : 03 dif (24 Bit Integer/Binary Instantaneous value)
069   : FD vif (Second extension FD of VIF-codes)
070   : 0C vife (Model/Version)
071 C!: 080000 meter version (000008)
074   : 02 dif (16 Bit Integer/Binary Instantaneous value)
075   : FD vif (Second extension FD of VIF-codes)
076   : 0B vife (Parameter set identification)
077 C!: 0011 parameter set (1100)

{
    "media":"warm water",
    "meter":"waterstarm",
    "name":"",
    "id":"22029791",
    "meter_timestamp":"2022-02-08 20:22",
    "total_m3":0.109,
    "total_backwards_m3":0,
    "current_status":"OK",
    "meter_version":"000008",
    "parameter_set":"1100",
    "timestamp":"2022-02-09T07:36:27Z"
}
```

weetmuts, 2022-02-09:
```
./build/wmbusmeters --version
wmbusmeters: 1.6.0-72-gc1f3031
c1f3031a946b52a4492f44af171e19cf22bb5982
```

SWAT9119, 2022-02-09:
I tried your command - I have the version 1.6.0-72-gc1f3031 - but the decryption doesn´t work - see the log...
[Output.txt](https://github.com/weetmuts/wmbusmeters/files/8030501/Output.txt)

weetmuts, 2022-02-09:
But that is something different. First try the analyze command to verify that you can decode your telegram.
Then we can take a look why your cul dongle sends bad data to wmbusmeters.

SWAT9119, 2022-02-09:
Ok I understand.

> Awesome! There was indeed a bug in mode 7 when it was used on a telegram that has non-encrypted data at the end. (in this case it is the model/verison and parameter set. Fix pushed. Now it decrypts [...]

When I run this command I got following answer, so I think the decoding is working:

`root@odroid:/etc/wmbusmeters# ./build/wmbusmeters --analyze=00112233445566778899AABBCCDDEEFF 4E46FA129197022202068C0003900F002C256C0000006B9DBFDB10E2AA677A3400200710BA9337856E2FDA09B4FB2AD7E25CC8C0875B84C1DFF0240FBFEF3223E5E0B70403FD0C08000002FD0B0011
Auto driver  : waterstarm
Best driver  : waterstarm 19/19
Using driver : waterstarm 19/19
000   : 4e length (78 bytes)
001   : 46 dll-c (from meter SND_IR)
002   : fa12 dll-mfct (DWZ)
004   : 91970222 dll-id (22029791)
008   : 02 dll-version
009   : 06 dll-type (Warm Water (30°C-90°C) meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 00 ell-cc (slow_resp)
012   : 03 ell-acc
013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
014   : 0f afl-len (15)
015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
018   : 6c000000 afl-counter (108)
022   : 6B9DBFDB10E2AA67 afl-mac 8 bytes
030   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
031   : 34 tpl-acc-field
032   : 00 tpl-sts-field (OK)
033   : 2007 tpl-cfg 0720 (AES_CBC_NO_IV )
035   : 10 tpl-cfg-ext (KDFS=1)
036   : 2f2f decrypt check bytes
038   : 04 dif (32 Bit Integer/Binary Instantaneous value)
039   : 6D vif (Date and time type)
040 C!: 1634C822 at date (2022-02-08 20:22)
044   : 04 dif (32 Bit Integer/Binary Instantaneous value)
045   : 13 vif (Volume l)
046 C!: 6D000000 total consumption (0.109000 m3)
050   : 02 dif (16 Bit Integer/Binary Instantaneous value)
051   : FD vif (Second extension FD of VIF-codes)
052   : 17 vife (Error flags (binary))
053 C!: 0000 info codes (OK)
055   : 04 dif (32 Bit Integer/Binary Instantaneous value)
056   : 93 vif (Volume l)
057   : 3C vife (backward flow)
058 C!: 00000000 total water backwards (0.000000 m3)
062   : 2F skip
063   : 2F skip
064   : 2F skip
065   : 2F skip
066   : 2F skip
067   : 2F skip
068   : 03 dif (24 Bit Integer/Binary Instantaneous value)
069   : FD vif (Second extension FD of VIF-codes)
070   : 0C vife (Model/Version)
071 C!: 080000 meter version (000008)
074   : 02 dif (16 Bit Integer/Binary Instantaneous value)
075   : FD vif (Second extension FD of VIF-codes)
076   : 0B vife (Parameter set identification)
077 C!: 0011 parameter set (1100)

{
    "media":"warm water",
    "meter":"waterstarm",
    "name":"",
    "id":"22029791",
    "meter_timestamp":"2022-02-08 20:22",
    "total_m3":0.109,
    "total_backwards_m3":0,
    "current_status":"OK",
    "meter_version":"000008",
    "parameter_set":"1100",
    "timestamp":"2022-02-09T07:52:40Z"
`

SWAT9119, 2022-02-09:
But the actuell telegrams I recieve are not decoded

weetmuts, 2022-02-09:
Nope, because they are not even considered by wmbusmeters. They are dropped since the data line from cul looks like this: 
"bA944FA12919702220206EADF7ADA009025B0CBA47A56799CBD0FBF813FADBEED8FE243CAEE248CD34CBC4D6FC17CBD24BCAA077CF801478147F7B0D0BE30C6C17EB0553A7F451E739A994634AF7408FDDAA\
18D5110903C8C99A31E5E89E62BD8562FC70A50493219823D92D3CFE115DE3375D92ED9FB39A6A61BE88348CB377141F17636467A5E1FA7119D3FFA839F07B58D362C2<0D><0A>"

and when wmbusmeters has extracted the content hex, there is an odd number of hex characters.... which 
is impossible since they have to be even. 

Please check the code in wmbus_cul.cc line 352. Note that the cul is started with X21, see above in the code.
The comment about using X01 is outdated.

You could also run for some longer time to see if the error is persisten or intermittent. Since you did recive values before.

SWAT9119, 2022-02-09:
Sorry I don´t know what you mean and what I´ve to do.

You mean line 322, because its a C1-Telegram?

It looks so:

`    if (data[1] == 'Y')
    {
        // C1 telegram in frame format B
        // bY..44............<CR><LF>
        *hex_frame_length = eolp;
        vector<uchar> hex;
        // If reception is started with X01, then there are no RSSI bytes.
        // If started with X21, then there are two RSSI bytes (4 hex digits at the end).
        // Now we always start with X01.
        hex.insert(hex.end(), data.begin()+2, data.begin()+eolp-eof_len-4); // Remove CRLF, RSSI and LQI
        payload.clear();
        bool ok = hex2bin(hex, &payload);
        if (!ok)
        {
            string s = safeString(hex);
            debug("(cul) bad hex \"%s\"\n", s.c_str());
            warning("(cul) warning: the hex string is not proper! Ignoring telegram!\n");
            return ErrorInFrame;
        }
        ok = trimCRCsFrameFormatB(payload);
        if (!ok)
        {
            warning("(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!\n");
            return ErrorInFrame;
        }
        debug("(cul) received full C1 frame\n");
        return FullFrame;
    }`

What Do I have to change?

weetmuts, 2022-02-09:
According to the cul, it reports b not bY. So it is a T1 telegram. But if you start wmbusmeters with c1, 
then I suppose it should not be that way. Perhaps you need to unplug the cul and replug it.

SWAT9119, 2022-02-09:
I unpluged and repluged the cul - no change.

So I have to use the AES5 telegram - right?

SWAT9119, 2022-02-09:
OK - I´ll try the T1 an C1 difference - maybe this is the solution.

Whats the difference between AES 5 and 7 - is 5 less safe?

weetmuts, 2022-02-09:
No AES 5 is just as safe. 128 bit AES.

The AES 7 is more advanced and introduces the authentication and fragmentation layer. Larger telegrams
in multiple radio transmissions etc. Theoretically it could be possible to have a separate authentication key from the confidentiality key, though it does not seem to be supported in the standard right now. 

Go with AES 5 if it works. The telegrams are shorter.

SWAT9119, 2022-02-09:
Thanks!

One last question: Is it better to use c1 oder t1? Has it any depencecies to the range?

weetmuts, 2022-02-09:
I do not know, you have to test. Depends on all sorts of things. My nanocul is really bad at receiving c1. C1 was created for more energy efficient transmissions.

SWAT9119, 2022-02-09:
**I´ve found the problem now:**

I can programm the meter with the software "Engelmann Device Monitor" an there I´ve set "long Telegram Mode" - then I don´t get the right telegrams.

With "Short Telegram Mode" I get the telegrams, encryptet (and decryptet) with AES 5 and AES 7.

Sorry for the expense!

weetmuts, 2022-02-09:
No worries. Glad to hear that it is working.

closed 2022-02-17 by weetmuts

apfelflo89, 2022-09-28:
Hey sorry to revive that issue, the manual states "master key" as the encryption key in factory default. Is there a reference key? Where can I find that? Is the software "Engelmann Device Monitor"  mandatory for using the meters?

magcode, 2022-09-28:
Same here, I'd also be interested in that. I have a waterstar M but no information about the configured encryption. Would love to try with some standard key...

SWAT9119, 2022-09-30:
I used the Engelmann Device Monitor with an IR Optokopf from Weidmann-Elektronik to programm a key on the meters. I´d contact with Engelmann, but they don´t publish the master key. You have to set an indiviual key for each meter

magcode, 2022-09-30:
Yes, I understand this approach.
I do have the Weidmann device but struggle to purchase the software.

On top I think that my facility service may not be pleased if I change the key.
I may try to contact them and ask for the key. But don't have much hope.

kki03, 2023-01-27:
@SWAT9119 I have Techem 4.5.1 Radio E and Techem radio 4 devices (basically rebranded Engelmann) which I can not read with wmbusmeter because I do not get the AES key.

Since you use the Engelmann Device Monitor with an IR Optokopf I´m curious if you can read the device data via IR interface without having the AES key. And I suppose you´re using this Optokopf https://shop.weidmann-elektronik.de/index.php?page=product&info=24

SWAT9119, 2023-01-28:
@kki03:

With the Engelmann Device Monitor and the Optokopf you mentioned I was able to change the AES-Key of every device. You can´t read out the current AES-Key - but you can change the key to a new one

mpl1337, 2025-11-15:
@SWAT9119 @kki03 

Do you have a Download Link for this Software?