wmbusmeters/wmbusmeters #472: No telegrams on nanoCUL 868

suawekstw, 2022-02-08:
Hi

I've installed wmbusmeters as addon for HA, which is running on Proxmox.
I'm using nanoCUL 868 extra, which I've bought from smart-home-komponente on Amazon: [https://www.amazon.de/-/pl/dp/B09QJZ9KR4/ref=sr_1_1?__mk_pl_PL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=BCWE61V9YT2&keywords=nanocul+868+extra&qid=1644348093&sprefix=nanocul+868+extr%2Caps%2C154&sr=8-1](url)
I've flashed this device with firmware in accordance with: [https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html](url)
Computer with HA is located around 15meters (in a straight line) from water meters.
I have two water meters equipped with Diehl IZAR RC 868 I R4 PL.
Althought different actions I cannot receive any telegrams.

My current config looks like this:

```
data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=/dev/ttyUSB1:cul:c1
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters: []
mqtt: {}
```

Logfile for this config: 
[log.txt](https://github.com/weetmuts/wmbusmeters/files/8026785/log.txt)

What I've already tested:
- Device is currently connected to computer on USB cable 1,5m long, previously it was connected directly - no difference.
- I've tried different device parameter settings including t1 and some frequency settings with same result.
- Reinstalling of addon - no result.
- Reflashing of device - no result.

Thanks in advance for help.

probutus, 2022-02-09:
I had the same issue with my nanoCUL I bought a year ago I used for some time to control my homegear stuff (which worked ok). As I abandoned homegear I had it floating around and picked it up, flashed the new firmware onto it and did end up in receiving nothing. In the end I sent it back got a new one which worked out of the box.

BUT I noticed along the way that  there are some "bad" variants of the nanoCUL around which are slightly off the requested frequency: See the firmware page here (note the "Frequenzabweichung" firmwares) https://www.nanocul.de/selfsite.php?aktion=Firmware

I created four different wmbus firmwares containing the four different frequency settings. They did not work for me but I can send them to you for testing if you like.

suawekstw, 2022-02-09:
Hi
Thanks for answer. Please send me those firmwares - I'll flash nanoCUL with them. If it will not help I'll proceed like you - send back to supplier for replacement.
Thanks

probutus, 2022-02-09:
Attached the zipfile with the frequency patches following the nanocul page (FREQ2/FREQ1/FREQ0) 

Standard Frequency settings are: 0x21/0x6B/0xD0. The modified setting is part of the name following the nanocul page.

[nanocul_r568_mbus_clt1_bufsize300.zip](https://github.com/weetmuts/wmbusmeters/files/8035061/nanocul_r568_mbus_clt1_bufsize300.zip)

suawekstw, 2022-02-16:
Today I've received recommended dongle (with rtl2832u onboard). It has started working "plug-and-play" - after short re-configuration I've started to receive tons of telegrams including ones sent by my both water meters. That means problem is with signal (for rtl  ~ 125dB) or with nanoCUL itself. I'll send that info to supplier and we'll see what happens.
For now - thanks for help!

probutus, 2022-02-17:
Thanks for feedback. My experience is that it is depending on the dongle you have. My old dongle I exchanged did not receive anything but another dongle with the same antenna receives still a very good signal through three concrete walls. (I have both RTL SDR and a new nanoCUL)

suawekstw, 2022-04-01:
I haven't receive any additional info from supplier so I've sent him nanoCUL back and received refund.
I'm currently operating on rtl2832u without any problems.
Thanks for help. Issue closed.