wmbusmeters/wmbusmeters #477: Diehl Sharky 774 - Partially Encrypted ?

patou50, 2022-02-16:
I recently tried to read the data from my Diehl Sharky 774 heatmeter.

So, I first identified my Heatmeter (there are other nearby in the building). 
`wmbusmeters --t1 rtlwmbus`

```
Received telegram from: 58496405
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Heat meter (0x04) encrypted
                   ver: 0x41
                device: rtlwmbus[00000001]
                  rssi: 36 dBm
                driver: sharky774
```

I was dissappointed to see that it seems to be encrypted.

However, I tried my luck and tried the NOKEY option...
`wmbusmeters --logtelegrams --format=json auto:t1 heatmeter sharky774 58496405 NOKEY`

which outputs

```
telegram=|3E44A5110564495841047A700030052F2F#0C06846800000C13195364000B3B0400000C2B110100000A5A17050A5E76020AA61800004C0647630000426CBF25|+55
{"media":"heat","meter":"sharky774","name":"heatmeter","id":"58496405","total_energy_consumption_kwh":0,"total_volume_m3":645.319,"volume_flow_m3h":0.004,"power_kw":0.111,"flow_temperature_c":51.7,"return_temperature_c":27.6,"temperature_difference_c":24.1,"timestamp":"2022-02-16T18:52:29Z","device":"rtlwmbus[00000001]","rssi_dbm":30}
```

Apart from total_energy_consumption_kwh which is incorrect here, all variables correspond to what I can read with my eyes on the device.
Strangely, can I interpret this as there is no key ? However, very strangely, the total_energy_consumption_kwh is set to 0. It seems to be the same for my neighbours.
Is this somehow related to the key ? So the key would only allow to get this variable extra ?

Germar, 2022-02-16:
Same here. I can see all correct variables except `total_energy_consumption_kwh` with Diehl Sharky 774 and IM871A-USB.

patou50, 2022-02-17:
I have also seen the exact same value of 220000 for variable power_kw.

Have you ever bothered contacting your supplier to get the key ?

weetmuts, 2022-02-17:
Can you post a plain rtl_sdr | rtl_wmbus line of telegram here? It is weird that the telegram does not need to be decrypted but still reports in the header that it is encrypted......

Germar, 2022-02-17:
Works like a charm! Thanks @weetmuts
```
$ wmbusmeters --logtelegrams --oneshot /dev/ttyUSB1:im871a:t1 heatmeter sharky774 61243590 NOKEY
Started config im871a[00100169] on /dev/ttyUSB1 listening on t1
telegram=|3E44A5119035246141047AE20030052F2F#0C06906201000C13736209040B3B2002000C2B420900000A5A96020A5E59020AA61800004C0636370100426CBF25|+34
heatmeter       61243590        16290 kWh       4096.273 m3     0.22 m3/h       0.942 kW        29.6 °C 25.9 °C 3.7 °C  13736 kWh       2022-02-17 15:41.58
```

Can you please show how to get the plain telegram? I've no idea. It shows `type: Heat meter (0x04) encrypted` for me as well.

patou50, 2022-02-17:
Some here, it works now, I get the correct total_energy_consumption_kwh.

Also, no idea why it still shows type: Heat meter (0x04) encrypted. Not sure how I can get the information you request

weetmuts, 2022-02-17:
Thanks! Just run with --debug to see the raw communication coming in. But since you run with NOKEY then the logtelegram output is the raw telegram.... so odd indeed.

My guess is that the programmer of the meter hardcoded the header settings (encryption flags and all) and simply did not run the encryption function when the meter was not using encryption....fortunately the 2f2f indicates that it is already decrypted and wmbusmeters could already handle this situation since this is how logtelegrams print the telegrams,
ie decrypted but with the original headers. This is useful because users can post the telegrams here on github without having to provide the key. But I am not entirely sure that it is standards compliant. Oh well.....

Germar, 2022-02-17:
```
(serial) received binary "A5C2033E44A5119035246141047AFC0030055C2224982DBAD8CBE78A9C8C5664A4AB8AD0E2D92E351B31B730A15545AADCAD8BB74FE37CC845D446501C7D3F5597947E4C3D"
(im871a) checkIM871AFrame "A5C2033E44A5119035246141047AFC0030055C2224982DBAD8CBE78A9C8C5664A4AB8AD0E2D92E351B31B730A15545AADCAD8BB74FE37CC845D446501C7D3F5597947E4C3D"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 126 (-59 dBm)
(im871a) got crc16 3d4c expected 3d4c
(im871a) received full frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) heatmeter: for me? 61243590 in 
(meter) heatmeter: yes for me
(meter) constructed "heatmeter" "sharky774" "61243590" encrypted
(meter) used meter template heatmeter  auto to match 61243590
(meter) started meter 1 (heatmeter 61243590 auto)
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) heatmeter: for me? 61243590 in 61243590
(meter) heatmeter: yes for me
(meter) heatmeter sharky774 handling telegram from 61243590
(meter) heatmeter 61243590 "3E44A5119035246141047AFC0030055C2224982DBAD8CBE78A9C8C5664A4AB8AD0E2D92E351B31B730A15545AADCAD8BB74FE37CC845D446501C7D3F559794"
(wmbus) parseDLL @0 63
(telegram) DLL L=3e C=44 (from meter SND_NR) M=11a5 (DME) A=61243590 VER=41 TYPE=04 (Heat meter) (driver sharky774) DEV=im871a[00100169] RSSI=-59
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(TPL) num encrypted blocks 3 (48 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "5C2224982DBAD8CBE78A9C8C5664A4AB8AD0E2D92E351B31B730A15545AADCAD8BB74FE37CC845D446501C7D3F559794"
(TPL) IV A511903524614104FCFCFCFCFCFCFCFC
(TPL) decrypted  "2F2F0C06966201000C13227509040B3B3201000C2B830600000A5A14030A5E69020AA61800004C0636370100426CBF25"
(dvparser) found new format "0C060C130B3B0C2B0A5A0A5E0AA6184C06426C" with hash 9290, remembering!
(telegram) TPL CI=7a ACC=fc STS=00 CFG=0530 (AES_CBC_IV nb=3 cntn=0 ra=0 hc=0)
telegram=|3E44A5119035246141047AFC0030052F2F#0C06966201000C13227509040B3B3201000C2B830600000A5A14030A5E69020AA61800004C0636370100426CBF25|+62
```

weetmuts, 2022-02-18:
Thanks! It is using a default key that is commonly re-used for a lot of diehl meters.

Germar, 2022-02-18:
Sometimes I still get `"power_kw":220000.XXX`

```
3E44A5119035246141047A020030052F2F#0C06976201000C13077709040B3B0601000C2B660000F00A5A68020A5E74020AA61800004C0636370100426CBF25
{"media":"heat","meter":"sharky774","name":"WaermeMengenZaehler","id":"61243590","total_energy_consumption_kwh":16297,"total_volume_m3":4097.707,"volume_flow_m3h":0.106,"power_kw":220000.066,"flow_temperature_c":26.8,"return_temperature_c":27.4,"temperature_difference_c":-0.6,"operating_time_h":0,"energy_at_set_date_kwh":13736,"set_date":"2021-05-31","timestamp":"2022-02-18T09:04:16Z"}

3E44A5119035246141047A020030052F2F#0C06976201000C13097709040B3B0801000C2B350000F00A5A73020A5E74020AA61800004C0636370100426CBF25
{"media":"heat","meter":"sharky774","name":"WaermeMengenZaehler","id":"61243590","total_energy_consumption_kwh":16297,"total_volume_m3":4097.709,"volume_flow_m3h":0.108,"power_kw":220000.035,"flow_temperature_c":27.3,"return_temperature_c":27.4,"temperature_difference_c":-0.1,"operating_time_h":0,"energy_at_set_date_kwh":13736,"set_date":"2021-05-31","timestamp":"2022-02-18T09:04:17Z"}

3E44A5119035246141047A0A0030052F2F#0C06986201000C13048009040B3B0601000C2B090000F00A5A77020A5E79020AA61800004C0636370100426CBF25
{"media":"heat","meter":"sharky774","name":"WaermeMengenZaehler","id":"61243590","total_energy_consumption_kwh":16298,"total_volume_m3":4098.004,"volume_flow_m3h":0.106,"power_kw":220000.009,"flow_temperature_c":27.7,"return_temperature_c":27.9,"temperature_difference_c":-0.2,"operating_time_h":0,"energy_at_set_date_kwh":13736,"set_date":"2021-05-31","timestamp":"2022-02-18T09:04:18Z"}

3E44A5119035246141047A0C0030052F2F#0C06996201000C13858009040B3B0701000C2B290200F00A5A62020A5E81020AA61800004C0636370100426CBF25
{"media":"heat","meter":"sharky774","name":"WaermeMengenZaehler","id":"61243590","total_energy_consumption_kwh":16299,"total_volume_m3":4098.085,"volume_flow_m3h":0.107,"power_kw":220000.229,"flow_temperature_c":26.2,"return_temperature_c":28.1,"temperature_difference_c":-1.9,"operating_time_h":0,"energy_at_set_date_kwh":13736,"set_date":"2021-05-31","timestamp":"2022-02-18T09:04:18Z"}

3E44A5119035246141047A0D0030252F2F#0C06996201000C13878009040B3B3301000C2B080000F00A5A84020A5E80020AA61800004C0636370100426CBF25
{"media":"heat","meter":"sharky774","name":"WaermeMengenZaehler","id":"61243590","total_energy_consumption_kwh":16299,"total_volume_m3":4098.087,"volume_flow_m3h":0.133,"power_kw":220000.008,"flow_temperature_c":28.4,"return_temperature_c":28,"temperature_difference_c":0.4,"operating_time_h":0,"energy_at_set_date_kwh":13736,"set_date":"2021-05-31","timestamp":"2022-02-18T09:04:18Z"}

3E44A5119035246141047A1A0030052F2F#0C06026301000C13688609040B3B0802000C2B220000F00A5A71020A5E72020AA61800004C0636370100426CBF25
{"media":"heat","meter":"sharky774","name":"WaermeMengenZaehler","id":"61243590","total_energy_consumption_kwh":16302,"total_volume_m3":4098.668,"volume_flow_m3h":0.208,"power_kw":220000.022,"flow_temperature_c":27.1,"return_temperature_c":27.2,"temperature_difference_c":-0.1,"operating_time_h":0,"energy_at_set_date_kwh":13736,"set_date":"2021-05-31","timestamp":"2022-02-18T09:04:21Z"}
```

weetmuts, 2022-02-18:
Yeah, I think it is a negative value encoded in bcd: 
034   : 0C dif (8 digit BCD Instantaneous value)
035   : 2B vif (Power W)
036 C!: 220000F0 ("power_kw":220000.022)

The return temperature is higher than the to temperature. So it makes sense that the power is  negative.
Can you take a look at dvparser.c to see if you can improve this? Maybe it is as simple to detect
that it is broken bcd, with an f at the most significant nybble position. Then replace the f with 0 and
remember to negate the calculated value at the end.

closed 2022-05-25 by weetmuts

MariuszWoszczynski, 2024-05-13:
> hanks! It is using a default key that is commonly re-used for a lot of diehl meters.

Can you tell us what the default key is?

PR #1047 (2023-12-04): Update driver_sharky
Hi,
Changed VifScaling to AutoSigned to fix problem with negative values.

Same change was done in #480 to fix #477 for Sharky 774
I've got the same problem with my Sharky 775 so i copied the solution in driver for Sharky 775.

Tested for some weeks now an problem seems to be solved without any side effects.
But: Please check fix - I am not a developer :)