wmbusmeters/wmbusmeters #479: DiehL hrl-c-g3 lorawan is it supported or not ?

g4seb, 2022-02-18:
Hello,

I am receiving the material this week. I have a Diehl HRLC G3 LORAWAN meter:

https://www.compteur-energie.com/hrl-c-g3-lorawan-diehl.htm

Is this one supported?

Thanks in advance

weetmuts, 2022-02-18:
So, lorawan radio is currently not supported. Do you intend to buy this meter?
You will need a lorawan dongle, like iU880B, to listen to it.

weetmuts, 2022-03-28:
Ping @g4seb ! If a water company provides this meter and has already paid for it to be connected to a specific
lorawan gatewy, then probably, they are not interested in you having access to it as well.. But who knows, you can always ask them. 

If you buy it yourself, then we can add support for reading it using an iU880B dongle. I have implemented
part of the iU880B communication already.

But if you are buying it yourself, I think a plain wmbus-meter is cheaper unless you have to read it over a long long distance.

olicot, 2022-04-21:
Hello
I've got the same water meter, imposed by my french water compagny, I can check water consumption with an website...
I've asked them how I can acces directly to the meter ...still waiting an answser ....
I'll still follow this project, but I think about a "plan B", such as another water meter  :-)

NoskaOff, 2022-08-26:
Same thing and I'm also in France. I have a Homeriders Systems HRL-c-G2 running on 868MHz (v1.6)

cmaile, 2022-11-10:
Any news regarding the HRL-c-G2?

marcinkowalczyk, 2023-01-22:
Hi @weetmuts I've recently got an iu880b - I've tried to put it into t1/c1 but no luck.  

```
(iu880b) rsp got firmware
(iu880b) firmware: 3.0.197 27.01.2022ProLink_LoRaWAN;LMIC_V1.5;EU868;Modem
(iu880b) got device id 00053c6c
(main) regular reset of iu880b  on /dev/ttyUSB0 will happen every 82800 seconds
Warning! Desired link modes t1 cannot be set for device /dev/ttyUSB0:iu880b[00053c6c]
(wmbus) no alarm (expected activity) for iu880b
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
```

Checked with latest master #fcd5f3e61 
My Diehl watter-meter is eqqiped with this LoRa based cover https://lora-alliance.org/lora_products/wireless-radio-water-meter-data-logger-apulse-w-x1f5/

weetmuts, 2023-01-22:
Hi @marcinkowalczyk !
The lora is its own radio mode, so you should use :lora instead of :t1
However! The code in wmbusmeters is not completed yet! But I can finish it for you.

Just double checking. The lora can only be understood if you control/know/establish all the lora keys. 

If this meter was installed by the water company to be integrated in their own data collection
network, then I am pretty sure that you will not get the keys.

If you bought your meter yourself and want to talk lora to it using iu880b, then it is possible
and will help you.

silkyclouds, 2023-03-13:
same problem here. Impossible to read the states as loraWAN uses keys.... the (belgian -> CILE) water provider won't share them. 

I guess the only way we have to snatch the information is to request an API or something.... But here in Belgium, I don't even think the website allowing to read the meters value do already exist.... what a shame.

closed 2023-06-12 by github-actions[bot]