wmbusmeters/wmbusmeters #481: Coexistance of WMBUSMeters with acurite2mqtt weather station reader for the same SDR

convicte, 2022-02-21:
This is not really an issue, but more an inquiry regarding the ability to integrate this project, with [SDR2mqtt/acurtite2mqtt](https://github.com/thejeffreystone/hassio_addons) weather station data collecting add-on, using a single SDR in Home Assistant.

So far I was able to run one or the other using the same SDR and each works just fine. I am a long term user ofSDR2mqtt/acurtite2mqtt add-on and very new to WMBUSMeters, so I am not sure if there are some hard limitations which cannot be overcome.

It seems they are both capturing the data from nearly the same band for me, but if it could be presented to two different add-ons for parsing into sensible MQTT messages, that would be great.

Best case scenario would be to run, both the WMBUS and SDR2mqtt/acurtite2mqtt, in parallel, where one would be able to decode both signals at the same time, or they would share the signals received from the devices 868Mhz band.

For reference, my WMBUSMeters and SDR2mqtt configs, just in case this helps:

````
#WMBUSMeters

data_path: /config/wmbusmeters
conf: |-
  loglevel=normal
  device=rtlwmbus:c1
  logtelegrams=false
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=water_meter
    driver=izar
    id=
    key=
mqtt:
  host:
  port: 1883
  user: 
  password: 
````

````

# SDR2mqtt

mqtt_host: 
mqtt_port: 1883
mqtt_user: 
mqtt_password: 
mqtt_topic: rtl_433
mqtt_retain: 'true'
frequency: '-f 868.3M'
protocol: '-R 52 -R 173'
units: si
discovery_prefix: homeassistant
discovery_interval: 100
debug: false

````

Looking forward to everyone's thoughts.

BIBOLV, 2022-02-25:
Somehow I think similar question was already discussed, but could not find quickly that thread to link it here.
It is not possible to share one dongle between multiple apps simultaneously, because each app when initiates connection to dongle locks it for specific process with specific parameters. There are few options you could do:

- Run each app for some time, for example 10 minutes in sequence not in parallel. There are multiple ways to do it for standalone instances, but not sure if it is possible to achieve out of the box withing HA for addons.
- Get second dongle.
- More unrealistic, but create app that get`s signal from dongle and creates two virtual devices and forwards signals to them. Then two apps could use those virtual devices. But as frequencies are different it is unlikely that it would work also.

convicte, 2022-02-26:
Thank you very much for the answer.

Sounds like a new 50$ dongle is in my future....
No idea how to do the other two.

weetmuts, 2022-05-25:
From your comment it looked like you had decided to buy another dongle.

You can configure cron to switch between the two tools

You can start wmbusmeters with --oneshot once per hour and wait for it to complete
(assuming your wmbus meter sends something within the hour.)
Then run weather for an hour.

https://github.com/weetmuts/wmbusmeters/issues/433