wmbusmeters/wmbusmeters #485: No possible to change shell in ha-addon configuration site

dzikus, 2022-02-25:
If you want to change shell command to something more complicated than orig line you will get such strange formatting with useless wmbusmeters inside container:

For example I want to set my own mosquitto_pub with retain flag, so I'm changing shell line to:
`shell=/usr/bin/mosquitto_pub -h 10.0.0.1 -p 1883 -t wmbusmeters/12345678 -m "$METER_JSON"
`

[image]

but after hitting "Save":

[image]

to-masz, 2022-04-28:
@dzikus I believe this is related to the yaml formatting applied by home assistant ... but it does not change a fact it should work - what issue do you have?

rincewindy, 2022-05-16:
The formatting seems to go wonky any time a non-standard shell command is used, e.g. 
[image]
vs
[image]
It does work, but sub-optimal.

BIBOLV, 2023-03-13:
HA add-on has moved to separate repo - https://github.com/wmbusmeters/wmbusmeters-ha-addon
It has now pre-built container images and config interface changes - please test and close this issue if no other questions.

closed 2023-06-12 by github-actions[bot]