wmbusmeters/wmbusmeters #487: sub 0 temperatures

rattinca, 2022-03-02:
I'm dealing with a temperature sensor that works fine for temp > 0 but sends weird data for temp<0 

Seems to have to do with extractDVdouble working with unsigned int. What can i use to get correct negative numbers?

I cant really understand what hex2bin does. Its not converting Hex to binary, or is it?

rattinca, 2022-03-02:
Never mind, i think if got it... its not hex at all, and the sensor has an F for - wich messes everything up. i will figure something out myself

Germar, 2022-03-05:
I had the same for a Sharky774 and fixed it with #480. Please checkout current master and change `VifScaling::Auto,` in temp value for the driver of your device to `VifScaling::AutoSigned,`

weetmuts, 2022-03-28:
Can you test with the latest version of wmbusmeters on github and provide us with an example telegram?

weetmuts, 2022-04-15:
@rattinca Have you tested with the latest version? There is a fix, but your meter might not have it.
Please provide an example telegram if it is not working.

weetmuts, 2022-05-25:
Closing this for now. F means it is a negative bcd value. To get such signed values out you mark it as VifScaling::AutoSigned.

closed 2022-05-25 by weetmuts