wmbusmeters/wmbusmeters #488: Decoding telegrams  from WebDyn Gateway

joergwulf, 2022-03-03:
We use a commercial gateway (WebDyn) to receive WMBus meter data and transfer the results to an FTP server.
The data is stored as BSON files. These can be converted into JSON. The result is:
[image]
Now we pass the coded part of the meter as parameter:
`/snap/wmbusmeters/current/usr/bin/wmbusmeters 077983561523934433060dff5f3500829700005f0107c113ffff08990000bf2c55860000df21508900000500260049004c00640027006a00c1006900000062001e002f046d1c10dc22`
The result is
[image]
We assume that wmbusmeters needs the whole coded telegram and when passing the payload of the meter the meta data is missing.
Is there any way to decode this telegram?

joergwulf, 2022-03-03:
The problem has been solved. The tool from Webdyn already does more than necessary for this issue. We have to use the data from the JSON output
[image]
converting Base64 to Hex
`4e44a5116160316170077a1f3040050d7ede5f0e1ec9dc6226ee3c13469e4f59832d4681833960f039e707971b1570555eb6fe11f955e99eea5ea03f37954f05f33cb5edc08348fe7c838ded5a605e`
and put it into wmbusmeters
[image]
..then get the id from the output and get the consumption data
[image]
Great tool !
Ist ist possible to get the consumption data with one call (without determining the ID from the first call) ?

rattinca, 2022-03-03:
usually you have the id of your meters beforehand. Its also in the telegram, that part is never encoded.

Germar, 2022-03-05:
You can use wildcard `"*"` for the meter_id
```
wmbusmeters --format=json 4e44a5116160316170077a1f3040050d7ede5f0e1ec9dc6226ee3c13469e4f59832d4681833960f039e707971b1570555eb6fe11f955e99eea5ea03f37954f05f33cb5edc08348fe7c838ded5a605e MyMeter auto "*" NOKEY
{"media":"water","meter":"hydrus","name":"MyMeter","id":"61316061","total_m3":50.813,"total_tariff1_m3":0,"total_tariff2_m3":0,"max_flow_m3h":0,"flow_temperature_c":24.3,"external_temperature_c":0,"current_date":"","total_at_date_m3":50.813,"total_tariff1_at_date_m3":0,"total_tariff2_at_date_m3":50.813,"at_date":"2021-12-31 00:00","actuality_duration_s":0,"operating_time_h":0,"remaining_battery_life_y":10.080767,"status":"TEMPORARY_ERROR AIR_IN_PIPE","timestamp":"2022-03-05T20:47:29Z"}
```

weetmuts, 2022-03-06:
Or ANYID which was introduced since * can be difficult to quote in shells inside shells etc...

joergwulf, 2022-03-07:
"*" works, but ANYID or anyid not:
[image]
Version is 1.6.0

weetmuts, 2022-03-28:
@joergwulf Great that you like wmbusmeters!

Now you can run "wmbusmeters --analyze <telegram> on the web.
https://wmbusmeters.org

It is the same output as from the command line, but it is more convient when
you just want to check the content of the telegram.

You can jump directly to your telegram like this:
https://wmbusmeters.org/analyze/4e44a5116160316170077a1f3040050d7ede5f0e1ec9dc6226ee3c13469e4f59832d4681833960f039e707971b1570555eb6fe11f955e99eea5ea03f37954f05f33cb5edc08348fe7c838ded5a605e

As you can see the hydrus driver misses four fields (marked as red) if you can explain the contents of these fields (well the first is obvious), then we can add them to the json as well.

closed 2022-05-25 by weetmuts