wmbusmeters/wmbusmeters #489: wmbusmasters - autoscan has no found an rtl_sdr dfongle and mosquitto not fired

joergbeicht, 2022-03-03:
I have two cases
Hardware:
rasp pi 4 and usb realtek rtl2838 (with lsusb)
https://www.amazon.de/gp/product/B01B4L48QU/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

first case:
wmbusmasters-docker version worked successfully and I got the telegrams for my multical21 about every 15 seconds. That was great. But mosquitto didn't fire. a 'hello mosqitto world' directly on the console with mosquitto_pub and mosquitto_sub had worked.

second case:
Since mosquitto didn't work with wmbusmaters-docker, I reinstalled the rasp and installed the native version.
Here I get this error message:
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!

I don't know what else I can do. Please help me.

joergbeicht, 2022-03-03:
This is my wmbusmeters.log with loglevel=debug 

(wmbusmeters) shutting down
(wmbusmeters) logging started 2022-03-03 14:56:50 using _
(wmbusmeters) version: _
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_sdr: not found
<<<
(shell) return code 127
(shell) 601 exited
(serial) EVENT thread interrupted
(main) rtl_sdr NOT found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>>/bin/sh: 1: rtl_wmbus: not found
<<<
(shell) return code 127
(shell) 602 exited
(main) rtl_wmbus NOT found in path
[2022-03-03_14:56:51] Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_sdr in the path!
[2022-03-03_14:56:51] Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>>586
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 604 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 609 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
[2022-03-03_14:56:53] (memory) rss 3526656 peak 3.36 MiB
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_sdr: not found
<<<
(serial) EVENT thread interrupted
(shell) return code 127
(shell) 633 exited
(main) rtl_sdr NOT found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_wmbus: not found
<<<
(serial) EVENT thread interrupted
(shell) return code 127
(shell) 634 exited
(main) rtl_wmbus NOT found in path
[2022-03-03_14:56:53] Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_sdr in the path!
[2022-03-03_14:56:53] Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_sdr: not found

BIBOLV, 2022-03-04:
Hi,
As described in docs - you need to install those binaries if you install standalone version. https://github.com/merbanan/rtl_433 https://github.com/weetmuts/rtl-wmbus

weetmuts, 2022-05-25:
Yes, if you run wmbusmeters from the command line, rtl_sdr and rtl_wmbus must be in the PATH.
wmbusmeters when started from a daemon or inside docker, will also look for rtl_sdr/rtl_wmbus in the same
directory that wmsbumeters/wmbusmetersd was found.

closed 2022-05-25 by weetmuts

joergbeicht, 2022-10-04:
Sorry, I had a bereavement privately and I didn't get in touch because of that.