wmbusmeters/wmbusmeters #490: Kamstrup Multical 21 - bad temperature

rclsilver, 2022-03-03:
Hi,

I'm using a Kamstrup Multical 21 and the reported temperature is quite hot. We have 127°C reported.

How can I help you to fix the issue?

Here you can find telegrams received:

```
telegram=|21442D2C353943741B168D20233090560302957934DD0EB8#7000AB08000062030000|+1222490
telegram=|21442D2C353943741B168D20243190560302957934DD0EB8#7000AB08000062030000|+1222506
telegram=|21442D2C353943741B168D20253290560302957934DD0EB8#7000AB08000062030000|+1222522
telegram=|21442D2C353943741B168D20263390560302957934DD0EB8#7000AB08000062030000|+1222538
telegram=|21442D2C353943741B168D20274090560302957934DD0EB8#7000AB08000062030000|+1222555
telegram=|21442D2C353943741B168D20284190560302957934DD0EB8#7000AB08000062030000|+1222571
telegram=|21442D2C353943741B168D20294290560302957934DD0EB8#7000AB08000062030000|+1222587
telegram=|24442D2C353943741B168D202A43905603FEB878#02FF2070000413AB080000441362030000|+1222603
telegram=|21442D2C353943741B168D202B5090560302957934DD0EB8#7000AB08000062030000|+1222619
telegram=|21442D2C353943741B168D202C5190560302957934DD0EB8#7000AB08000062030000|+1222636
telegram=|21442D2C353943741B168D202D5290560302957934DD0EB8#7000AB08000062030000|+1222652
```

weetmuts, 2022-03-04:
That is easy. It does not send temperature. :-) Run this command: wmbusmeters --analyze 24442D2C353943741B168D202A43905603FEB878#02FF2070000413AB080000441362030000

I am working on adding support for fields that go away when the data does not exist in the telegram....

weetmuts, 2022-03-28:
The model/version/config numbers printed on the meter shows which values are sent in the telegram. Closing this for now.

closed 2022-03-28 by weetmuts