wmbusmeters/wmbusmeters #491: Driver Unknown

One1Tick, 2022-03-05:
Hi I have an issue because the driver is unknown. How could I handle this  ?
Thanks for help.

```
FT=e61e VER=3c TYPE=07 (Water meter)
Received telegram from: 00620249
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Bus/System component (0x0e) encrypted
                   ver: 0x02
      Concerning meter: 15788688
          manufacturer: (GWF) Gas u Wassermesserfabrik, Switzerland (0x1ee6)
                  type: Water meter (0x07) encrypted
                   ver: 0x3c
                device: rtlwmbus[00000001]
                  rssi: 91 dBm
                driver: unknown!

```

Cyklops66, 2022-03-05:
Not an Issue. It is described in the Wiki how to build your own driver.

One1Tick, 2022-03-06:
Thanks for your feedback. Sorry, It is explained that to create a new driver we should start based on an existing one but I don't know the type of my device. How could I know this ? Moreover as showd above mine is encrypted... where can I find the key ? According to the wiki :
 `You can see that this telegram is encrypted and therefore you need a key. Now listen to this specific meter, since the driver is auto-detected, we can use auto for the meter driver.`
What does it mean ?
Sorry I am a newbie in development. Thx.

BIBOLV, 2022-03-06:
About key - it has been discussed in many issues, like [this](https://github.com/weetmuts/wmbusmeters/issues/443), so you can read them trough. 
If/when you will get key you are encouraged to create driver and make PR or post decrypted messages here.

One1Tick, 2022-03-10:
Thanks @BIBOLV, I will try to ask my water provider for the key but I think it is hopeless...

weetmuts, 2022-03-28:
Good luck! Please open a new issue when you have the key.