wmbusmeters/wmbusmeters #493: Lost 00000001 closing rtlwmbus[00000001]

F-HALD, 2022-03-07:
What am I doing wrong here?
I already installed rtl-wmbus

pi@meter-reader:~ $ wmbusmeters auto:c1
Started auto rtlwmbus[00000001] listening on c1
No meters configured. Printing id:s of all telegrams heard!
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on c1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on c1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on c1

F-HALD, 2022-03-07:
Debug-log:

(wmbusmeters) version: 1.6.0-95-g5ccbe83
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) check if /dev/ttyAMA0 can be accessed
(serial) tty /dev/ttyAMA0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyAMA0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(amb8465) sent 6 bytes OK
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyAMA0 (detect amb8465)
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyAMA0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyAMA0 (detect im871a)
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect rc1180)
(serial /dev/ttyAMA0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyAMA0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyAMA0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect cul)
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyAMA0 (detect cul)
(cul) are you there? no
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
[... 88 lines cut ...]
(serial) EVENT thread interrupted
(main) rtlsdr device 00000001 not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on c1
(rtlwmbus) using command: rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 2326 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(bgshell) 2326 exited with return code 0
(serial) closing non working fd=5 "00000001"
(wmbus) disconnected 00000001 rtlwmbus
(serialcmd) closed /bin/sh pid=2326 fd=5 (rtlwmbus)
^C(serial) stopping manager
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted rtlwmbus
(serialcmd) closed /bin/sh pid=2326 fd=-1 (rtlwmbus)

BIBOLV, 2022-03-18:
Try running command executed by wmbusmeters yourself and check if it works. From logs looks like that command is exiting all the time.

weetmuts, 2022-05-25:
Try the "rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
from the command line. This will show if you do not have the proper user rights to access the rtl_sdr for example.

Closing this for now since this issue is stale. Hope you solved it!

closed 2022-05-25 by weetmuts

jarsiv, 2024-12-10:
Hmmm, same issue in here:
```
 $ wmbusmeters auto:c1
Started auto rtlwmbus[] listening on c1
No meters configured. Printing id:s of all telegrams heard!
Lost  closing rtlwmbus[]
Started auto rtlwmbus[] listening on c1
Lost  closing rtlwmbus[]
Started auto rtlwmbus[] listening on c1
Lost  closing rtlwmbus[]
Started auto rtlwmbus[] listening on c1
Lost  closing rtlwmbus[]
Started auto rtlwmbus[] listening on c1
Lost  closing rtlwmbus[]
Started auto rtlwmbus[] listening on c1
Lost  closing rtlwmbus[]
Started auto rtlwmbus[] listening on c1
```
Debug:
```
~ $ wmbusmeters --debug auto:c1
(drivers) added builtin driver aquastream
(drivers) added builtin driver elf2
(drivers) added builtin driver elster
(drivers) added builtin driver eltako
(drivers) added builtin driver hydrocalm4
(drivers) added builtin driver ime
(drivers) added builtin driver iperl
(drivers) added builtin driver istaheat
(drivers) added builtin driver itronheat
(drivers) added builtin driver kampress
(drivers) added builtin driver maddalena
(drivers) added builtin driver relhca
(drivers) added builtin driver sensostar
(drivers) added builtin driver supercal
(drivers) added builtin driver werhlemodwm
(wmbusmeters) version: _
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index or serial (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        [-D enable direct sampling (default: off)]
        filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1746 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(shell) return code 0
(shell) 1748 exited
(main) rtl_wmbus found in path
(main) rtlsdr device  not currently used.
(serial) EVENT thread interrupted
(main) ignoring rtlsdr  since it is unavailable.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 1752 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>1741 1473
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 1753 exited
Notice! Other wmbusmeters (pid 1473) is running and it might hog any wmbus devices.
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

```
Is it running still...? I wonder that I will not found any devices around? Couple years ago there were planty of devices.

BIBOLV, 2024-12-29:
Logs report that other instance of wmbusmeters is running, that could cause this instance to not able to see any dongles.