wmbusmeters/wmbusmeters #494: Decrypt a Kamstrup meters KEM file. No module named 'Crypto'

tomas6464, 2022-03-13:
I'm using latest RaspBian and following ["Additional tools"](https://github.com/weetmuts/wmbusmeters#user-content-additional-tools) 
Trying to make a meters config file with command:
`python utils/kem-import.py utils/20220207_0837DownloadMeters.zip.kem 456123`
gives error:
**Traceback (most recent call last):
  File "/home/pi/wmbusmeters/utils/kem-import.py", line 30, in <module>
    import Crypto.Cipher.AES as AES
ModuleNotFoundError: No module named 'Crypto'**

Any suggestions? 
All help is apreciated! But be aware I´m a Windows user so be gently and detailed with the steps :)
I guess I need to install something that is missing on my RaspBerry pi... 
[image]

BIBOLV, 2022-03-14:
`sudo apt-get install python-crypto`

BIBOLV, 2022-03-14:
Try this then
`sudo apt-get install python3-crypto`
I don`t have raspi at the moment to test to provide tested answer.

tomas6464, 2022-03-14:
### Many thank´s @BIBOLV , problem solved by you! 
**First:**
`sudo apt-get install python3-crypto`
**Then** extracting and decrypt of the Kamstrup meters KEM file succeded now. :)
`python utils/kem-import.py utils/20220207_0837DownloadMeters.zip.kem 456123`
Found meter KWM2210 (02K59CA2D8SE)
    number : 23541719
    serial : 23541719
    type   : VolumeCold
    driver : None
    config : 8035200213614336030100314302
    key    : B353964FF03130CB79379B9ABCA6E7AF
    !! unknow meter type, meter file has not been created

It´s a supported water meter **Kamstrup flowiq 2200** so I will set the driver : **flowiq2200**
and manually save a meter file.

**Again @BIBOLV thank you** for the fast and accurate reply! I appreciate it a lot :)
(See attached screen dump for more info
[image]
)

closed 2022-03-14 by tomas6464