wmbusmeters/wmbusmeters #495: Example for curl

diddi68, 2022-03-14:
Hello, I want to send the $METER_JSON via curl, but I'm still not successful. The command is set in wmbusmeters.conf.
The following syntax works an sends the "1" to the destination:
`shell=curl -k -s http://192.168.178.15/config/xmlapi/statechange.cgi?ise_id=22835&new_value=1`
but
`shell=curl -k -s http://192.168.178.15/config/xmlapi/statechange.cgi?ise_id=22835&new_value="$METER_JSON"`
didn't work.
Does anybody know the right syntax for it?

BIBOLV, 2022-03-14:
Hi,
It is question more about your backend. Did you check logs there and tried to debug/understand what is the issue? If it is homematic xml-api backend then based on [docs](https://github.com/homematic-community/XML-API) you need to pass one value for each id/channel. $METER_JSON passes all available values as json, so on your case you need to use $METER_TOTAL_M3 or other specific value that you are interested in.

diddi68, 2022-03-14:
Thank you for your help. Now it works fine (with single values e.g. $METER_TOTAL_M3).

closed 2022-03-14 by diddi68