wmbusmeters/wmbusmeters #496: !! unknow meter type. But it is a known Kamstrup flowiq 2200...

tomas6464, 2022-03-14:
### What can I do to solve this and have a working meter file?

I have **got a key** from the water company and manage (with good help of @BIBOLV ) to decrypt the Kam file.
**Here´s the decrypted result:**
```
Detected a zip file on input ... extracting
Creating target folder: home/pi/wmbusmeters/etc/wmbusmeters.d/
Found meter KWM2210 (02K59CA2D8SE)
    number : 23541719
    serial : 23541719
    type   : VolumeCold
    driver : None
    config : 8035200213614336030100314302
    key    : B353964FF03130CB79379B9ABCA6E7AF
    !! unknow meter type, meter file has not been created
```
**And here´s a snip of the readings:**
```
(amb8465) not enough bytes yet, partial command response 46 67.
(serial) received binary "0A4BF522F85CF06D01B588A0639B9A6C"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20F419DD17214340523E6AADC83F3F1F7D3CF72D91799431CA3069C9BB71D843950A4BF522F85CF06D01B588A0639B9A6C"
(amb8465) not enough bytes yet, partial command response 62 67.
(serial) received binary "CF90BE93AE"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20F419DD17214340523E6AADC83F3F1F7D3CF72D91799431CA3069C9BB71D843950A4BF522F85CF06D01B588A0639B9A6CCF90BE93AE"
(amb8465) received full command frame
(amb8465) unhandled device message 3
(amb8465) unknown response "3F442D2C5963807134048D20F419DD17214340523E6AADC83F3F1F7D3CF72D91799431CA3069C9BB71D843950A4BF522F85CF06D01B588A0639B9A6CCF90BE93"
(serial) received binary "FF033F442D2C"
(amb8465) checkAMB8465Frame "FF033F442D2C"
(amb8465) not enough bytes yet, partial command response 6 67.
(serial) received binary "5963807134048D20F51CDD1721486F"
(amb8465) chunk time 14 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20F51CDD1721486F"
(amb8465) not enough bytes yet, partial command response 21 67.
(serial) received binary "E8DE3D875D63872592612B34576DD1"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20F51CDD1721486FE8DE3D875D63872592612B34576DD1"
(amb8465) not enough bytes yet, partial command response 36 67.
(serial) received binary "C6636E7A801240375A8E966C1EBC6B98"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20F51CDD1721486FE8DE3D875D63872592612B34576DD1C6636E7A801240375A8E966C1EBC6B98"
(amb8465) not enough bytes yet, partial command response 52 67.
(serial) received binary "A60B0FDC0E38984CE116F50FAC3ACE"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20F51CDD1721486FE8DE3D875D63872592612B34576DD1C6636E7A801240375A8E966C1EBC6B98A60B0FDC0E38984CE116F50FAC3ACE"
(amb8465) received full command frame
(amb8465) unhandled device message 3
(amb8465) unknown response "3F442D2C5963807134048D20F51CDD1721486FE8DE3D875D63872592612B34576DD1C6636E7A801240375A8E966C1EBC6B98A60B0FDC0E38984CE116F50FAC3A"
(serial) received binary "FF033444372C19175423"
(amb8465) checkAMB8465Frame "FF033444372C19175423"
(amb8465) not enough bytes yet, partial command response 10 56.
(serial) received binary "3A168D20579CE972207CED27E9C9D1"
(amb8465) chunk time 14 msec
(amb8465) checkAMB8465Frame "FF033444372C191754233A168D20579CE972207CED27E9C9D1"
(amb8465) not enough bytes yet, partial command response 25 56.
(serial) received binary "D37DD916E4841D71D653E4700EDA8943"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033444372C191754233A168D20579CE972207CED27E9C9D1D37DD916E4841D71D653E4700EDA8943"
(amb8465) not enough bytes yet, partial command response 41 56.
(serial) received binary "FAF4E4283D1B4F1C86D6A6A9F30F2E"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033444372C191754233A168D20579CE972207CED27E9C9D1D37DD916E4841D71D653E4700EDA8943FAF4E4283D1B4F1C86D6A6A9F30F2E"
(amb8465) received full command frame
(amb8465) unhandled device message 3
(amb8465) unknown response "3444372C191754233A168D20579CE972207CED27E9C9D1D37DD916E4841D71D653E4700EDA8943FAF4E4283D1B4F1C86D6A6A9F30F"
(serial) received binary "FF0309472D2C59638071340448"
(amb8465) checkAMB8465Frame "FF0309472D2C59638071340448"
(amb8465) received full command frame
(amb8465) unhandled device message 3
(amb8465) unknown response "09472D2C596380713404"
(serial) received binary "FF033F442D2C59"
(amb8465) checkAMB8465Frame "FF033F442D2C59"
(amb8465) not enough bytes yet, partial command response 7 67.
(serial) received binary "63807134048D20F828DD1721D8571466"
(amb8465) chunk time 16 msec
```
Any suggestion is appreciated on what to do next...   
Attached is an image of the meter.
[image]

BIBOLV, 2022-03-14:
From log snippet issue is with receiving proper message, so try to restart wmbusmeters and post full log from start till first message is received.
Maybe check also if baud rate is correct - check near end of manual page.

tomas6464, 2022-03-15:
Thank´s again @BIBOLV 
1. Attached below is now a **longer snippet and from start.**
2. **How** can I **exactly** send the Amber 6485 **reset command** explained in the manual :)
I tried to cut and paste without success...
[try wmbusmeters-admin and select](https://github.com/weetmuts/wmbusmeters#non-standard-baud-rate-set-for-amb8465-usb-stick) `Reset wmbus receives`

pi@mbus:~ $ wmbusmeters --debug auto:c1
(wmbusmeters) version: 1.6.0-95-g5ccbe83
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) check if /dev/ttyAMA0 can be accessed
(serial) tty /dev/ttyAMA0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyAMA0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyAMA0 (detect amb8465)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyAMA0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyAMA0 (detect im871a)
(serial) before config: /dev/ttyAMA0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect rc1180)
(serial /dev/ttyAMA0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyAMA0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyAMA0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyAMA0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect cul)
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyAMA0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyAMA0 (detect cul)
(cul) are you there? no
(serial) before config: /dev/ttyAMA0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyAMA0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
[... 198 lines cut ...]
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20499D2F18212E6D4706E68F60430DFDAA0DDE2BC26A18D3D784EF2ED4"
(amb8465) not enough bytes yet, partial command response 42 67.
(serial) received binary "0C018C302F883FF07403ECA60154"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20499D2F18212E6D4706E68F60430DFDAA0DDE2BC26A18D3D784EF2ED40C018C302F883FF07403ECA60154"
(amb8465) not enough bytes yet, partial command response 56 67.
(serial) received binary "DD51D16C8738D29F46749D"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "FF033F442D2C5963807134048D20499D2F18212E6D4706E68F60430DFDAA0DDE2BC26A18D3D784EF2ED40C018C302F883FF07403ECA60154DD51D16C8738D29F46749D"
(amb8465) received full command frame
(amb8465) unhandled device message 3
(amb8465) unknown response "3F442D2C5963807134048D20499D2F18212E6D4706E68F60430DFDAA0DDE2BC26A18D3D784EF2ED40C018C302F883FF07403ECA60154DD51D16C8738D29F4674"
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 amb8465
(serialtty) closed /dev/ttyUSB0 (amb8465)
(wmbus) deleted amb8465
(serial) closing 1 devices
pi@mbus:~ 
[image]
$

BIBOLV, 2022-03-15:
Well from logs to me looks like configuration of dongle is not completed, so trying to reset dongle to default parameters is only idea I have currently. 
Did you tried to run wmbusmeters-admin? If it does not work, you could try to contact dongle manufacturer to get their application for dongle and reset it trought there.
Also noticed that dongle is detected on second run - can you check if load is not high. Do you run some graphical interface or other heavy processes?

tomas6464, 2022-03-17:
Thanks again @BIBOLV 
**Resetting** the Amber 6485 dongle is now **goal number one.**
And **there is no (intentional) processes** started by me that is **loading the CPU**. But it is the default Rasbian with support for VNC and graphical interface that is loaded though.
**Should I run the Rasbian Lite version instead maybe...?**

**1. On the Raspberry pi**
I tried `wmbusmeters-admin` as cut & paste into the terminal without any respons. I probably did it wrongly... :) 
It would be great to be ably to reset the amb6485 dongle with the Raspberry pi. **What should I exactly type in the SSH terminal window?** Sorry for linux noob questions :)  And thank´s in advance!

**2. On the Win 10 PC**
I´m also trying to reset the Amber 6485 dongle with my Win10 and the ACC3 (AMBER Config Center 3) software from [Wurth](https://www.we-online.com/web/en/electronic_components/produkte_pb/service_pbs/wco/software/acc.php).
This did not work with the v3.3 I found on the net. **I´m now waiting for a response from the Wurth support to hopefully get a newer version** that does not conflict with the latest FTDI-driver.
On the [Wurth support page](https://www.we-online.com/web/en/electronic_components/produkte_pb/service_pbs/wco/wco_technicalsupport.php) I found this interesting comment:
_"Important Notice: Important notice for customers using Würth Elektronik´s eiSos USB devices (sticks, adapters, evaluation boards).
To avoid compatibility problems with the FTDI driver (version 2.12.0 or higher) we recommend to download the newest version of the Würth Elektronik eiSos software products. If you have further questions, please contact our technical support team by using the form below."_
And [here´s ](https://www.we-online.com/web/en/electronic_components/produkte_pb/service_pbs/wco/software/acc.php)more:
_"Due to current export legal restrictions, we are required to control the provision of software.
To download the software, please contact our [technical support](https://www.we-online.com/web/en/electronic_components/produkte_pb/service_pbs/wco/wco_technicalsupport.php) or your local sales representative directly.
Thank you for your understanding."_

/Tomas

tomas6464, 2022-03-18:
Okey! I got a reading. New firmware in the Amber 6485 helped. Thanks again @BIBOLV !
### **But what now?** to get a working driver :)
(attached a longer file aswell)

(amb8465) received full frame
(amb8465) rssi 121 (-14 dBm)
(wmbus) parseDLL @0 64
(telegram) DLL L=3f C=44 (from meter SND_NR) M=2c2d (KAM) A=71806359 VER=34 TYPE=04 (Heat meter) (driver unknown!) DEV=amb8465[00000000] RSSI=-14
(wmbus) parseELL @10 54
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=71 SN=3c301921 (AES_CTR session=12 time=1151747) CRC=b163
Received telegram from: 71806359
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat meter (0x04) encrypted
                   ver: 0x34
                device: amb8465[00000000]
                  rssi: -14 dBm
                driver: unknown!
(wmbus) 000   : 3f length (63 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 2d2c dll-mfct (KAM)
(wmbus) 004   : 59638071 dll-id (71806359)
(wmbus) 008   : 34 dll-version
(wmbus) 009   : 04 dll-type (Heat meter)
(wmbus) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 011   : 20 ell-cc (slow_resp sync)
(wmbus) 012   : 71 ell-acc
(wmbus) 013   : 3c301921 sn (AES_CTR)
(wmbus) 017 CE: B163CD4CD78E07B06BFE6549DED07BADD68152284E22988702C26B98FA637A8EAE1E15630929C73A445BF0D7FA4179 encrypte
[wmbusmeters_first_log.odt](https://github.com/weetmuts/wmbusmeters/files/8306480/wmbusmeters_first_log.odt)
d

BIBOLV, 2022-03-18:
Great! Snippet you added to post is not your meter based on info from your KEM file, but in attached file there is one telegram that is not decoding on-the-fly. So now you should create config file and run wmbusmeters as service or run it from command line, but you should capture more telegrams from your meter and if they are not recognizes then post them here.

tomas6464, 2022-03-19:
Okey!  :)
1. Made my own meter file in /etc/wmbusmeters.d/MyTapWater
```
sudo nano MyTapWater
   name=MyTapWater
   driver=flowiq2200:c1
   id=23541719
   key=B353964FF03130CB79379B9ABCA6E7AF
```
2. Started the wmbusmeter service
```
   sudo systemctl enable wmbusmeters
   sudo systemctl start wmbusmeters
```
3. Check and monitor the wmbusmeters.log
`tail -f /var/log/wmbusmeters/wmbusmeters.log`
A snip (and there is attached a longer wmbusmeters.log file also):
```
(telegram) TPL CI=79
telegram=|3444372C191754233A168D2066889A7420AF1F79F3A92091#000000002E35000012200000C1230000C0010000066000067000051B15|+1587
(flowiq2200) 000   : 34 length (52 bytes)
(flowiq2200) 001   : 44 dll-c (from meter SND_NR)
(flowiq2200) 002   : 372c dll-mfct (KAW)
(flowiq2200) 004   : 19175423 dll-id (23541719)
(flowiq2200) 008   : 3a dll-version
(flowiq2200) 009   : 16 dll-type (Cold water meter)
(flowiq2200) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(flowiq2200) 011   : 20 ell-cc (slow_resp sync)
(flowiq2200) 012   : 66 ell-acc
(flowiq2200) 013   : 889a7420 sn (AES_CTR)
(flowiq2200) 017   : af1f payload crc (calculated af1f OK)
(flowiq2200) 019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(flowiq2200) 020   : f3a9 format signature
(flowiq2200) 022   : 2091 data crc
(flowiq2200) 024 C!: 00000000 info codes (OK)
(flowiq2200) 028 C!: 2E350000 total consumption (13.614000 m3)
(flowiq2200) 032 C!: 12200000 target consumption (8.210000 m3)
(flowiq2200) 036 C!: C123 target_datetime (2022-03-01 00:00)
(flowiq2200) 038 C!: 0000 current flow (0.000000 m3/h)
(flowiq2200) 040 C!: C001 max flow (0.448000 m3/h)
(flowiq2200) 042 C!: 0000 min flow (0.000000 m3/h)
(flowiq2200) 044 C?: 066000067000
(flowiq2200) 050 C!: 05 min flow temperature (5.000000 °C)
(flowiq2200) 051 C!: 1B max flow temperature (27.000000 °C)
(flowiq2200) 052 C!: 15 external temperature (21.000000 °C)
```

4. Check and monitor the meter_readings for "MyTapWater"
`tail -f //var/log/wmbusmeters/meter_readings/MyTapWater`
```
A snip (and there is attached a longer MyTapWater file also):
tail: //var/log/wmbusmeters/meter_readings/MyTapWater: file truncated
{"media":"cold water","meter":"flowiq2200","name":"MyTapWater","id":"23541719","total_m3":13.614,"target_m3":8.21,"target_datetime":"2022-03-01 00:00","current_flow_m3h":0,"max_flow_m3h":0.448,"min_flow_m3h":0,"min_flow_temperature_c":5,"max_flow_temperature_c":27,"external_temperature_c":21,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2022-03-19T19:12:42Z","device":"amb8465[00000000]","rssi_dbm":-11}
```
[MyTapWater.txt](https://github.com/weetmuts/wmbusmeters/files/8309954/MyTapWater.txt)
[wmbusmeters.log.txt](https://github.com/weetmuts/wmbusmeters/files/8309955/wmbusmeters.log.txt)

BIBOLV, 2022-03-19:
Congrats, you successfully received reading from your meter. Now you can proceed and do with it what you intended.

closed 2022-03-20 by tomas6464

pulse-mind, 2024-03-02:
@tomas6464 I have the same water meter than you, would you like to explain how did you proceed ? 
I've read that you request at your water provider the KEY ?

tomas6464, 2024-03-02:
Hi

This was a while since i did... :)

Here comes a few links I used:
https://github.com/wmbusmeters/wmbusmeters/issues/494

https://the78mole.de/wmbus-meters-and-how-to-get-it-into-home-assistant/

You probably know these already...

1. I bought a usb stick that could listen to the meter and connected it to
a raspberrypi.
2. The raspberrypi then decrypts the messages and updates a value.
3. A second pi with homeassistant then reads the values.

I remember that I had to work quite a lot to get it up and running.
A. Tricky to get the key to actually decrypt the messages.
B. I had to update the firmware on the old usb stick I bought on ebay. Took
time... All the knowledge seeking and learning.

But hope this helps a bit.

Regards Tomas

pulse-mind, 2024-03-02:
@tomas6464 Thanks a lot for your answer, 
So the water company provided to you a file that contains the key, is it what happened ?

tomas6464, 2024-03-02:
Yes, got a key quite easily :)

/TL

MichaelBitard, 2024-03-05:
Did you manage to receive something **before** having the kem file? I have a kwm2230 and I don't know where to read the id, there is a bunch of stuff written on the flowiq but nothing seems to look like an id

tomas6464, 2024-03-05:
Hi Michaël

I'm pretty sure I could read some data without the decrypt function
working. I believe I even got some data from neighbours.

There where quite some time ago but a lot of shell window work...  :)

Regards Tomas

MichaelBitard, 2024-03-06:
Thanks, I indeed receive a lot of stuff, but as I cannot determine which is mine, I don't really know what to do next. No ID I receive looks like something written on the device. I tried to convert hexadecimal to decimal still nothing :disappointed: