wmbusmeters/wmbusmeters #497: No wmbus device detected, waiting for a device to be plugged in.

antvgr, 2022-03-16:
Hi !
I can't detect my apator. I don't understand the problem.

My setup: 
Wmbusmeters (W-MBus to MQTT) on HomeAssistant

My configuration
`data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=/dev/swradio0
  logtelegrams=false
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=EauChaude
    driver=auto
    id=00902947
    key=NOKEY
mqtt: {}
`

My logs:
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
Running wmbusmeters ...
(config) "device" "/dev/swradio0"
(wmbusmeters) version: 1.6.0-95-g5ccbe83
(config) using device: /dev/swradio0 
(config) number of meters: 1
(lookup) with file/hex "/dev/swradio0" unknown
(serial) check if /dev/swradio0 can be accessed
(serial) you do not have the correct permissions to open the tty /dev/swradio0 and you do not share the same access group.
(verbose) ignoring device /dev/swradio0
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 257 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>254
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 258 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
(lookup) with file/hex "/dev/swradio0" unknown
(serial) check if /dev/swradio0 can be accessed
(serial) you do not have the correct permissions to open the tty /dev/swradio0 and you do not share the same access group.
(verbose) ignoring device /dev/swradio0`

Thanks ;)

BIBOLV, 2022-03-16:
What Device is behind /dev/swradio0? Rtl-sdr or something else, because Service can't derect any dongles as stated in logs.

antvgr, 2022-03-16:
This is a [RTL2832U](https://amzn.to/3qbkBoa).

Germar, 2022-03-17:
> (serial) check if /dev/swradio0 can be accessed
> (serial) you do not have the correct permissions to open the tty /dev/swradio0 and you do not share the same access group.

Make sure you have permissions to access `/dev/swradio0`.

antvgr, 2022-03-17:
Yes in auto he finds nothing

antvgr, 2022-03-17:
Update: After a reinstall it seems to work !

Logs:
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
Running wmbusmeters ...
(config) "device" "rtlwmbus:368.95M"
(wmbusmeters) version: 1.6.0-95-g5ccbe83
(config) using device: rtlwmbus:368.95M 
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)
<<<
(shell) return code 1
(shell) 258 exited
(serial) EVENT thread interrupted
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 259 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus:368.95M) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus:368.95M
Started config rtlwmbus[00000001] listening on none using fq 368.95M
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 368.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(serial) EVENT thread interrupted
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 368.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serialcmd) opened /bin/sh pid 264 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop`

Just waiting for a message from my meter...

antvgr, 2022-03-18:
After 12 hours no message from meter 🥲

BIBOLV, 2022-03-18:
I would suggest to run rtl_sdr outside of HA docker and confirm that you receive telegrams, you might need to adjust frequency, etc. Once telegrams are received you can transfer config to HA docker and check if then you receive telegrams.

closed 2022-03-27 by antvgr