wmbusmeters/wmbusmeters #498: log : skipping already handled telegram

Docmorfine, 2022-03-18:
Hello,

I have a Sensuscout water meter, I installed wmbusmeters but I can't get the telegram.
I think it's encrypted but I don't get the key error message.

At end, i only get the message that telegram is skipped (before that we see ' parser expected more data!')

Do you think I need a decryption key?

Thank you

here is my log:
```
root@raspberrypi:~# sudo wmbusmeters --debug --t1 rtlwmbus
(wmbusmeters) version: 1.6.0-95-g5ccbe83
(config) using device: rtlwmbus 
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 32612 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 32614 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 32620 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "T1;1;1;2022-03-18 08:38:11.000;13;13;00280503;0x1e44ae4c0305280068077a000010056912e87d47520f9594d8d4daa9a46b1d<0A>T1;1;1;2022-03-18 08:38:11.000;15;16;00280503;0x1e44ae4c0305280068077a000010056912e87d47520f9594d8d4daa9a46b1d<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-03-18 08:38:11.000;13;13;00280503;0x1e44ae4c0305280068077a000010056912e87d47520f9594d8d4daa9a46b1d<0A>T1;1;1;2022-03-18 08:38:11.000;15;16;00280503;0x1e44ae4c0305280068077a000010056912e87d47520f9594d8d4daa9a46b1d<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 31
(telegram) DLL L=1e C=44 (from meter SND_NR) M=4cae (SEN) A=00280503 VER=68 TYPE=07 (Water meter) (driver iperl) DEV=rtlwmbus[00000001] RSSI=13
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(TPL) num encrypted blocks 1 (16 bytes and remaining unencrypted 0 bytes)
(wmbus) parser expected more data! (1524)
(telegram) TPL CI=7a ACC=00 STS=00 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
Received telegram from: 00280503
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x68
                device: rtlwmbus[00000001]
                  rssi: 13 dBm
                driver: iperl
(wmbus) 000   : 1e length (30 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : ae4c dll-mfct (SEN)
(wmbus) 004   : 03052800 dll-id (00280503)
(wmbus) 008   : 68 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 00 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 )
(wmbus) 015 CE: 6912E87D47520F9594D8D4DAA9A46B1D encrypted
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-03-18 08:38:11.000;15;16;00280503;0x1e44ae4c0305280068077a000010056912e87d47520f9594d8d4daa9a46b1d<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
```

Docmorfine, 2022-03-18:
when i run : 

`wmbusmeters --debug --logtelegrams --format=json auto:t1 watermeter iperl 00190352 NOKEY`

i have

```
(serial) received ascii "T1;1;1;2022-03-18 10:14:05.000;66;92;00190352;0x1e44ae4c5203190068077ab400100547b94172046abd28740d2105f1522980<0A>T1;1;1;2022-03-18 10:14:05.000;78;82;00190352;0x1e44ae4c5203190068077ab400100547b94172046abd28740d2105f1522980<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-03-18 10:14:05.000;66;92;00190352;0x1e44ae4c5203190068077ab400100547b94172046abd28740d2105f1522980<0A>T1;1;1;2022-03-18 10:14:05.000;78;82;00190352;0x1e44ae4c5203190068077ab400100547b94172046abd28740d2105f1522980<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) watermeter: for me? 00190352 in 
(meter) watermeter: yes for me
(meter) constructed "watermeter" "iperl" "00190352" encrypted
(meter) used meter template watermeter  auto to match 00190352
(meter) started meter 1 (watermeter 00190352 auto)
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(meter) watermeter: for me? 00190352 in 00190352
(meter) watermeter: yes for me
(meter) watermeter iperl handling telegram from 00190352
(meter) watermeter 00190352 "1E44AE4C5203190068077AB400100547B94172046ABD28740D2105F1522980"
(wmbus) parseDLL @0 31
(telegram) DLL L=1e C=44 (from meter SND_NR) M=4cae (SEN) A=00190352 VER=68 TYPE=07 (Water meter) (driver iperl) DEV=rtlwmbus[00000001] RSSI=66
(wmbus) parseELL @10 21
(wmbus) parseNWL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(TPL) num encrypted blocks 1 (16 bytes and remaining unencrypted 0 bytes)
(wmbus) parser expected more data! (1524)
(telegram) TPL CI=7a ACC=b4 STS=00 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
(meter) newly created meter (watermeter 00190352 iperl) did not handle telegram!
(wmbus) telegram from 00190352 ignored by all configured meters!
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2022-03-18 10:14:05.000;78;82;00190352;0x1e44ae4c5203190068077ab400100547b94172046abd28740d2105f1522980<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
```

BIBOLV, 2022-03-18:
Running --analyze on telegrams you attached it also says that they are encrypted.

closed 2022-05-25 by weetmuts