wmbusmeters/wmbusmeters #499: Missing device mfct: (TCH) - 0x5068 type Heat meter (0x43) ver: 0x39

JensHoRi, 2022-03-19:
Cannot provide more information yet, because still missing AES keys from Supplier for this one, but the type is not known, I think the key is second whats needed, right?

JensHoRi, 2022-03-19:
(serial) received binary "3744685046"
(amb8465) checkAMB8465Frame "3744685046"
(amb8465) not enough bytes yet, partial frame 5 56.
(serial) received binary "4783673943A210672C866100181C01"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "37446850464783673943A210672C866100181C01"
(amb8465) not enough bytes yet, partial frame 20 56.
(serial) received binary "000480794435D50000000000000000"
(amb8465) chunk time 16 msec
(amb8465) checkAMB8465Frame "37446850464783673943A210672C866100181C01000480794435D50000000000000000"
(amb8465) not enough bytes yet, partial frame 35 56.
(serial) received binary "00000000000000000000000000000000"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "37446850464783673943A210672C866100181C01000480794435D5000000000000000000000000000000000000000000000000"
(amb8465) not enough bytes yet, partial frame 51 56.
(serial) received binary "0000000000"
(amb8465) chunk time 15 msec
(amb8465) checkAMB8465Frame "37446850464783673943A210672C866100181C01000480794435D50000000000000000000000000000000000000000000000000000000000"
(amb8465) received full frame
(amb8465) rssi 0 (-74 dBm)
(meter) no meter handled  checking 4 templates.
(wmbus) parseDLL @0 56
(wmbus) parseELL @10 46
(wmbus) parseNWL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(meter) HeizungWW: for me? 67834746 in 67834746
(meter) HeizungWW: meter detection did not match the selected driver compact5! correct driver is: unknown!
(meter) Not printing this warning again for id: 67834746 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x43) ver: 0x39
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) HeizungWW: yes for me
(meter) created "HeizungWW" "compact5" "67834746" not-encrypted
(meter) used meter template HeizungWW 67834746 compact5 to match 67834746
(meter) started meter 1 (HeizungWW 67834746 compact5)
(wmbus) parseDLL @0 56
(wmbus) parseELL @10 46
(wmbus) parseNWL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(meter) HeizungWW: for me? 67834746 in 67834746
(meter) HeizungWW: meter detection did not match the selected driver compact5! correct driver is: unknown!
(meter) Not printing this warning again for id: 67834746 mfct: (TCH) Techem Service (0x5068) type: Heat meter (0x43) ver: 0x39
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) HeizungWW: yes for me
(meter) HeizungWW compact5 handling telegram from 67834746
(meter) HeizungWW 67834746 "37446850464783673943A210672C866100181C01000480794435D50000000000000000000000000000000000000000000000000000000000"
(wmbus) parseDLL @0 56
(telegram) DLL L=37 C=44 (from meter SND_NR) M=5068 (TCH) A=67834746 VER=39 TYPE=43 (Heat meter) (driver unknown!) DEV=amb8465[00000000] RSSI=-74
(wmbus) parseELL @10 46
(wmbus) parseNWL @10 46
(wmbus) parseAFL @10 46
(wmbus) parseTPL @10 46
(telegram) TPL CI=a2
telegram=|37446850464783673943A2|10672C866100181C01000480794435D50000000000000000000000000000000000000000000000000000000000|+9
(compact5) 000   : 37 length (55 bytes)
(compact5) 001   : 44 dll-c (from meter SND_NR)
(compact5) 002   : 6850 dll-mfct (TCH)
(compact5) 004   : 46478367 dll-id (67834746)
(compact5) 008   : 39 dll-version
(compact5) 009   : 43 dll-type (Heat meter)
(compact5) 010   : a2 tpl-ci-field (Mfct specific)
(compact5) 014 C!: 8661 energy used in previous billing period (24966.000000 KWH)
(compact5) 018 C!: 1c01 energy used in current billing period (284.000000 KWH)

Not able to create a src / new device, but here is the debug

JensHoRi, 2022-03-21:
Just a remark, this device is not encrypted. the shown total energy (sum og previous and current) is correct last day 24h.

So equal if it got discovered, it shows those two values correct.

weetmuts, 2022-03-28:
It seems like your meter behaves like a compact5, if so we only need to add the mfct,media,version for the lookup table for compact5.

Please confirm that the extracted values are correct!

You can view your telegram here as well:

https://wmbusmeters.org/analyze/37446850464783673943a210672c866100181c01000480794435d50000000000000000000000000000000000000000000000000000000000

JensHoRi, 2022-03-28:
The total amount is exactly last night 24:00:00. that’s checked. Previous I need to wait for next end of month.

weetmuts, 2022-03-28:
Thanks, fix pushed. Please test.

JensHoRi, 2022-03-28:
Because it’s my only snap version: will it update on Ubuntu 20.04 automatically? Then I will test

BIBOLV, 2022-03-29:
If you installed from edge channel then you will get all latest updates, if you are using default - stable, then you will get only new version releases.

JensHoRi, 2022-04-02:
Tested with actual Telegram:

Auto driver  : compact5
Best driver  : compact5(driver should be upgraded) 04/49
Using driver : compact5(driver should be upgraded) 00/00
000   : 37 length (55 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 6850 dll-mfct (TCH)
004   : 46478367 dll-id (67834746)
008   : 39 dll-version
009   : 43 dll-type (Heat meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 10672C866100206E02000400C1415F7D5503000000000000000000000000000000000000000000000000000000 mfct specific
059 C!: 8661 energy used in previous billing period (24966.000000 KWH) ===> Looks like value of day radio was activated (7th of march)
063 C!: 6e02 energy used in current billing period (622.000000 KWH)

{
    "media":"heat",
    "meter":"compact5",
    "name":"",
    "id":"67834746",
    "total_kwh":25588, =====> Correct value like physically shown on device at 24:00:00 last night
    "current_kwh":622,
    "previous_kwh":24966, ===> looks like value from radio-activation-date. Maybe can see in one year if it changes at 31st of december :-(
    "timestamp":"2022-04-02T10:31:43Z"
}

weetmuts, 2022-05-25:
Seems like it is working, as good as it gets with a proprietary telegram which is not wmbus compliant.

closed 2022-05-25 by weetmuts