wmbusmeters/wmbusmeters #501: Allmess MBUS module supported?

larsxschneider, 2022-03-25:
I have an [im871a dongle](https://wireless-solutions.de/products/m-bus/im871a-usb/) and my water company offered me to install an Allmess meter with an [MBUS module](https://www.hts24.de/installation/wasserzahler/allmess/aufputzwasserzahler/allmess-wasserzahler-systeme-m-bus-modul-m-3040000006.html) (*).

The manufacture offering says "BM +m M-BUS-Modul +m (unparametriert)" (see [doc, page 4](https://www.allmess.de/fileadmin/multimedia/alle_Dateien/PL_P1101_Preisliste2021.pdf)). "M-BUS" sounds good. I don't know what "unparameterized" means in that context.

Does wmbusmeters support this module? If you don't know it - how can I figure if it is supported _ahead_ of the purchase? 😄 What should I look out for?

#346 discusses this manufacturer already but this might be a different module called "EquaScan". Plus, a different protocol is discussed there "C2" which seems not to be supported yet?

Thank you!

---

(*) Here are the different options my water company offered me and I think the first one is propably the MBUS module that might be compatible with wmbusmeters?!

[image]

BIBOLV, 2022-03-25:
From [docs](https://www.itron.com/-/media/feature/products/documents/user-guides-and-manuals/12715acmabmm-pmm2109.pdf) it is not clear if meter is OMS compliant or any usable details actually, but as some Itron meter is supported, then you might get lucky.
unparametriert might mean that it is not configured.

weetmuts, 2022-03-28:
Hi @larsxschneider ! It seems like they offer you a device that attached to your meter, gives you an mbus interface.
This a two-wire based interface and to read it you need an mbus-master device, like https://www.packom.net/m-bus-master-hat/
Unfortunately the im871a cannot speak mbus, it can only speak wmbus.

Wmbusmeters does have  basic support to read such meters. Right now it can only read a single kind of mbus meter, the piigth
temp/hygro meter. To get the readings from such meters, wmbusmeters sends a poll-request over mbus and waits for a response. The mbus master hat usually looks like a serial port, so assuming it is on ttyUSB1 the command line looks like this:

wmbusmeters MAIN=/dev/ttyUSB1:mbus MyTempHygro piigth:MAIN:2400 10000284 NOKEY

It will then poll the MAIN bus (MAIN is just a name given to this serial port that is going to be used for mbus.)
every few seconds and send a activation request for the meter with serial nr 10000284 on  bus MAIN using the speed 2400 bps, and wait for a response that the meter sends back. Then it decodes the response which has a slightly different framing than wmbus telegrams, but the content is formatted exactly the same.

Can we then decode the protocol for your Allmess MBUS water meter module? Probably. :-) 
But please ask them for documentation on the content of the mbus response it sends.

larsxschneider, 2022-03-31:
Thank you very much for the insights - things are much clearer for me now🙇 

I think I will tell the water company to install EquaScan wMIU RF
https://www.itron.com/na/solutions/product-catalog/equascan-wmiu-rf

The specifications look like it could work:
[image]

https://www.itron.com/-/media/feature/products/documents/user-guides-and-manuals/15659ab_manual_equascan-wmiu_0721.pdf

The catch is that the manual says an initial setup needs to be made.

weetmuts, 2022-03-31:
Yes, since you have an im871a dongle it can work. C2 means that wmbusmeters must send a wakeup call to the meter.
This is not yet tested in wmbusmeters, but I have created all the necessary infrastructure to do this.

It seems like the EquaScan can also be configured in unidirectional mode, ie C1.
https://www.itron.com/-/media/feature/products/documents/brochure/unidirectionalwmiuen0518180529.pdf

This is probably a safer bet.

vogler, 2022-03-31:
Just got two EquaScan wMIU RF installed an hour ago. GitHub Search for EquaScan only yields this project and [rtlamr](https://github.com/bemasher/rtlamr).

I don't have any RF dongle. Are there easy alternatives?
My old meters had some rotating half metal disc that I could read with IR diodes. The new ones don't have this visible, the EquaScan manual sounds like the RF module just sits on top and still reads this disc ('opto-electric scan').
Does someone know more about it? The manual mentions some inductive interface - is there a rotating magnet I could read with a hall effect sensor? My old ones did not have one.
There's https://github.com/jomjol/AI-on-the-edge-device which tries to read the digits with an ESP32-Cam, but I'd prefer something reliable/less bulky.

It's a rented apartment, the company installing the meter did not know much about it. I assume I can't just change its mode or get an encryption key.
Polling also sounds like it would decrease its (non-replaceable) battery life. Manual says it publishes only every 5min or 1min if queried? 5min would be too low resolution for automation (e.g. don't turn off lights).

Is there some recommendation for a RTL SDR to use with a RPi?
https://www.aliexpress.com/item/32994719072.html $31.32
https://www.aliexpress.com/item/32901374096.htm $17.28

I'd be happy if someone could enlighten me a bit :)

larsxschneider, 2022-04-22:
I managed to get a HYDRUS 2.0 instead: https://github.com/weetmuts/wmbusmeters/pull/520

Closing this.

closed 2022-04-22 by larsxschneider

weetmuts, 2022-04-27:
@vogler I think that if rtlamr supports your meter, thats the way to go.
wmbusmeters focuses on meters using the european (w)mbus standard.

You can buy the cheapest possible rtlsdr dongle to experiment with, as long
as you are close to the meter and there are no thick concrete walls in between.

The higher quality dongles and a tuned antenna can be used if you need to receive over longer distances.
https://weetmuts.github.io/wmbusmeterswiki/

weetmuts, 2022-04-27:
@vogler Please join the discussion in this issue: https://github.com/weetmuts/wmbusmeters/issues/346