wmbusmeters/wmbusmeters #504: Missing Voltage Battery in waterstarm

JensHoRi, 2022-04-02:
Hi, 
i have some devices (engelmann/lorenz) and missing battery voltage in wmbusmeters.

Here the debug with the value I believe it should be. Because devices are new, actually it has 3 or 2.99 V battery power.

(meter) EGkwEG: for me? 21059897 in 21059897
(meter) EGkwEG: meter detection did not match the selected driver waterstarm! correct driver is: unknown!
(meter) Not printing this warning again for id: 21059897 mfct: (DWZ) Lorenz, Germany (0x12fa) type: Water meter (0x07) ver: 0x00
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) EGkwEG: yes for me
(meter) EGkwEG waterstarm handling telegram from 21059897
(meter) EGkwEG 21059897 "2E44FA129798052100077A070020257060A07BAE144B421A95D8A7436B0AD80189A2AF0EB5407719C28EF9F44A632C"
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=12fa (DWZ) A=21059897 VER=00 TYPE=07 (Water meter) (driver unknown!) DEV=amb8465[00000000] RSSI=-52
(wmbus) parseELL @10 37
(wmbus) parseNWL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(TPL) num encrypted blocks 2 (32 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "7060A07BAE144B421A95D8A7436B0AD80189A2AF0EB5407719C28EF9F44A632C"
(TPL) IV FA129798052100070707070707070707
(TPL) decrypted  "2F2F046D0F28C22404139540000002FD17000001FD481D2F2F2F2F2F2F2F2F2F"
(telegram) TPL CI=7a ACC=07 STS=00 CFG=2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
telegram=|2E44FA129798052100077A070020252F2F_046D0F28C22404139540000002FD17000001FD481D2F2F2F2F2F2F2F2F2F|+789
(dvparser) warning: cannot extract double from non-existant key "04933C"
(dvparser) warning: cannot extract uint24 from non-existant key "03FD0C"
(wmbus) warning: cannot find offset -1 to add more explanation " meter version (000000)"
(dvparser) warning: cannot extract uint16 from non-existant key "02FD0B"
(wmbus) warning: cannot find offset -1 to add more explanation " parameter set (0000)"
(waterstarm) 000   : 2e length (46 bytes) total water backwards (0.000000 m3)
(waterstarm) 001   : 44 dll-c (from meter SND_NR)
(waterstarm) 002   : fa12 dll-mfct (DWZ)
(waterstarm) 004   : 97980521 dll-id (21059897)
(waterstarm) 008   : 00 dll-version
(waterstarm) 009   : 07 dll-type (Water meter)
(waterstarm) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(waterstarm) 011   : 07 tpl-acc-field
(waterstarm) 012   : 00 tpl-sts-field (OK)
(waterstarm) 013   : 2025 tpl-cfg 2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
(waterstarm) 015   : 2f2f decrypt check bytes (OK)
(waterstarm) 017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(waterstarm) 018   : 6D vif (Date and time type)
(waterstarm) 019 C!: 0F28C224 at date (2022-04-02 08:15)
(waterstarm) 023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(waterstarm) 024   : 13 vif (Volume l)
(waterstarm) 025 C!: 95400000 total consumption (16.533000 m3)
(waterstarm) 029   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(waterstarm) 030   : FD vif (Second extension FD of VIF-codes)
(waterstarm) 031   : 17 vife (Error flags (binary))
(waterstarm) 032 C!: 0000 info codes (OK)
(waterstarm) 034   : 01 dif (8 Bit Integer/Binary Instantaneous value)
(waterstarm) 035   : FD vif (Second extension FD of VIF-codes)
(waterstarm) 036   : 48 vife (10^-1 Volts)
(waterstarm) 037 C?: 1D ==========================> Battery Voltage?, 2.99 or 3.00V
(waterstarm) 038   : 2F skip
(waterstarm) 039   : 2F skip
(waterstarm) 040   : 2F skip
(waterstarm) 041   : 2F skip
(waterstarm) 042   : 2F skip
(waterstarm) 043   : 2F skip
(waterstarm) 044   : 2F skip
(waterstarm) 045   : 2F skip
(waterstarm) 046   : 2F skip

weetmuts, 2022-05-24:
Fix pushed! Please test!

JensHoRi, 2022-05-30:
Until yet I didn't see the voltage of the battery :-(

JensHoRi, 2022-06-24:
Just changed to edge-Release, there I got the value! Thanks! I was on stable before, so wasnt able to see it

weetmuts, 2022-06-26:
Thank! Made a new stable release so you do not need to be on edge. Closing this now!

closed 2022-06-26 by weetmuts