wmbusmeters/wmbusmeters #508: Itron compatible cle Realtek RTL2838 DVB-T ?

did2299, 2022-04-06:
Hello,

Is the RTL2838 key compatible with my Itron EverBlu Cyble Enhanced V2.1 water meter?
I can't get information from my meter.

Thank you very much in advance.

BIBOLV, 2022-04-07:
Hi,
Well, if you are asking about receiving some data from meter in general then this is not question to wmbusmeters, but you can check [this](https://github.com/merbanan/rtl_433/issues/1954) or do some googling. 
But if you are asking if wmbusmeters will be able to parse and decode data from that [meter](https://www.itron.com/-/media/feature/products/documents/brochure/everblu_cyble_enhanced_pb_en_0911.pdf) then I guess that no, because meter does not use wmbus protocol, but everblu.

GSzabados, 2022-04-30:
Look at this project:

https://github.com/neutrinus/everblu-meters

did2299, 2022-05-01:
Merci pour ton retour! je vais me commander une clé cc1101

GSzabados, 2022-05-01:
@did2299, if you want to understand how is it working, then have a look here:

http://www.lamaisonsimon.fr/wiki/doku.php?id=maison2:compteur_d_eau:compteur_d_eau

weetmuts, 2022-05-26:
Thanks @GSzabados !  Very interesting information.
I think we can close this issue here.

closed 2022-05-26 by weetmuts