wmbusmeters/wmbusmeters #509: wmbusmeters 1.7.0 SIGSEGVs @src/meters.cc:87 - registerDriver()

dwrobel, 2022-04-06:
I'm in a process of upgrading wmbusmeters to the latest `1.7.0` release for `Fedora`.

However, the build failed as follows:
```
Building 1.7.0
cp build/wmbusmeters.g build/wmbusmeters
cp build/wmbusmeters build/wmbusmeters.g; true build/wmbusmeters
make: *** [Makefile:167: check_docs] Error 1

```
Logs from a builder can be accessed at: https://copr.fedorainfracloud.org/coprs/dwrobel/wmbusmeters/build/4090871/

I reproduced it locally on `gdb`:
```
[dw@dell wmbusmeters-1.7.0]$ ./build/wmbusmeters --help 
Segmentation fault (core dumped)
[dw@dell wmbusmeters-1.7.0]$ gdb --args ./build/wmbusmeters --help 
GNU gdb (GDB) Fedora 11.1-5.fc34
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/wmbusmeters...
(gdb) r
Starting program: /home/dw/projects/rpmbuild/BUILD/wmbusmeters-1.7.0/build/wmbusmeters --help
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
registerDriver(std::function<void (DriverInfo&)>) (setup=...) at src/meters.cc:87
87	            bool foo = p.second.detect(d.mfct, d.type, d.version);
(gdb) p p
$1 = {first = Cannot access memory at address 0x20
(gdb) bt
#0  registerDriver(std::function<void (DriverInfo&)>) (setup=...) at src/meters.cc:87
#1  0x0000555e3e381013 in _sub_I_65535_0.0 ()
#2  0x0000555e3e47862d in __libc_csu_init ()
#3  0x00007fdba2481b01 in __libc_start_main (main=0x555e3e372000 <main(int, char**)>, argc=2, argv=0x7fffd9878948, init=0x555e3e4785e0 <__libc_csu_init>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffd9878938) at ../csu/libc-start.c:279
#4  0x0000555e3e382fce in _start ()
(gdb) l
82	    // Check that no other driver also triggers on the same detection values.
83	    for (auto &d : di.detect())
84	    {
85	        for (auto &p : all_registered_drivers_)
86	        {
87	            bool foo = p.second.detect(d.mfct, d.type, d.version);
88	            if (foo)
89	            {
90	                error("Internal error: driver %s tried to register the same auto detect combo as driver %s alread has taken!\n",
91	                      di.name().str().c_str(), p.second.name().str().c_str());
(gdb) p all_registered_drivers_
$2 = std::map with 0 elements
(gdb) 
```
It looks that the `all_registered_drivers_` map is empty.

weetmuts, 2022-04-06:
Check the changes in the makefile.
The new drivers driver_xxx.cc files are picked up automatically.

I am in the process of converting the old style drivers meter_xxx.cc but it
takes a bit of time. So there is a mix of old and new.

weetmuts, 2022-04-14:
Did you solve it?

dwrobel, 2022-04-15:
Unfortunately, I didn't have time to look at it.

weetmuts, 2022-04-16:
No worries. Is it possible to integrate the rpm building into the
wmbusmeters github source tree and makefile?
That way I can build an rpm as part of testing the release?

//Fredrik

Den fre 15 apr. 2022 kl 20:32 skrev Damian Wrobel ***@***.***

weetmuts, 2022-05-23:
ping @dwrobel :-)

BIBOLV, 2023-01-02:
Closing issue as it should be fixed. Please reopen issue if problem persists.