wmbusmeters/wmbusmeters #511: Add support for rtl_tcp?

andriej, 2022-04-07:
In most known package of librtlsdr there's rtl_tcp (which enables RTL on tcp and can be used on remote hosts).
I have some cheap openwrt router which could be host for USB RTL stick while readings/parsing could be done remotely.

Is it doable to use it with wmbusmeters tho?

weetmuts, 2022-05-01:
It seems like it would send quite a lot of traffic over the network.... if your openwrt router can run rtl_wmbus, then you can relay the decoded telegrams to wmbusmeters on a different computer.

On the openwrt router:
rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a | nc -u 192.168.1.777 4444

On 192.168.1.777:
wmbusmeters 'rtlwmbus:CMD(nc -lku 4444)'

I suppose you could do something similar with rtl_tcp if your openwrt router cannot run rtl_wmbus.

Check the README for how rtl_wmbus is started and tips on using it in a pipe.

weetmuts, 2022-05-25:
@andriej I got it working like this:
On your router run:
rtl_tcp -f 868.95M -s 1600000 -a 192.168.xx.xx
on your computer run:
nc 192.168.xx.xx 1234 | /usr/bin/rtl_wmbus -p s -a | wmbusmeters stdin:rtlwmbus

xx.xx is whatever your router network ip address is.

It should work with the nc 192 ... | rtlwmbus | wmbusmeters 
command inside rtlwmbus:CMD(nc .... ) 
but for some reason it does not.  Something to debug...

Please test!

andriej, 2022-05-25:
I'd love to test it, just wondering how to implement your way of work into running add-on in HA installation.
I have to sleep it over, maybe clues will come - seems like solution is close :-)

andriej, 2022-05-31:
Whatever I try to run addon with, I end up with:
```
Not a valid device "rtlwmbus:CMD("nc 192.168.x.x 1234|/usr/bin/rtl_wmbus -p s -a|wmbusmeters stdin:rtlwmbus ")"
[cmd] /run.sh exited 1
```

weetmuts, 2022-09-12:
It seems like nc exists as soon as stdin is EOF. And stdin becomes EOF immediately
when the background process is started. Add "-q -1" to prevent this from happing so try:

device=rtlwmbus:CMD(nc -q -1 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters stdin:rtlwmbus)

travnick, 2022-09-19:
`nc` (`netcat (The GNU Netcat) 0.7.1`) does not have `-q` and `-1` options.

Anyway I see some strange issues in logs:

```
$ wmbusmeters --use
(config) "# Remember to change auto here to the device you are going to use in production." ""
(config) "#device=rtlwmbus:868.9M" ""
(config) "#device=rtlwmbus:CMD(nc 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters --device=stdin:rtlwmbus)" ""
(config) "#device=rtl433:CMD(rtl_433 -d rtl_tcp:192.168.1.2:1234 -f 868.95M -s 1.2M -Y autolevel -Y squelch -R 104 -M level -M protocol -F csv)" ""
(config) "device" "rtlwmbus:CMD(nc 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters stdin:rtlwmbus)"
(config) "logtelegrams" "false"
(config) "format" "json"
(config) "meterfiles" "/var/lib/wmbusmeters/meter_readings"
(config) "meterfilesaction" "append"
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log"
(config) "" ""
(config) loading meter file /etc/wmbusmeters.d/apator01946185.conf
(config) name=apator01946185
(config) type=apator162
(config) id=01946185
(config) key=<notprinted>
(wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log
```

So it says that there is one meter configured, but in log `/var/log/wmbusmeters/wmbusmeters.log` it says `No meters configured`:

```
(serial) received ascii "Started config rtlwmbus on stdin listening on any<0A>No meters configured. Printing id:s of all telegrams heard!<0A>"
[UNLOCKING] read_mutex receive (receive 69768)
[UNLOCKED]  read_mutex receive ( 0)
(rtlwmbus) checkRTLWMBusFrame "Started config rtlwmbus on stdin listening on any<0A>No meters configured. Printing id:s of all telegrams heard!<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "No meters configured. Printing id:s of all telegrams heard!<0A>"
(rtlwmbus) only text
```

```
(wmbusmeters) version: HEAD_1.9.0
(config) store meter files in: "/var/lib/wmbusmeters/meter_readings"
(config) using device: rtlwmbus:CMD(nc 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters stdin:rtlwmbus) 
(config) number of meters: 1
[LOCKING] bus_devices_mutex check_for_bus_devices ( 0)
[LOCKED]  bus_devices_mutex check_for_bus_devices (check_for_bus_devices 69768)
[MAIN] checking for dead wmbus devices...
[LOCKING] event_loop_mutex eventLoop ( 0)
(serial) expecting devices to work
[UNLOCKING] bus_devices_mutex check_for_bus_devices (check_for_bus_devices 69768)
[UNLOCKED]  bus_devices_mutex check_for_bus_devices ( 0)
[LOCKING] bus_devices_mutex check_for_bus_devices ( 0)
[LOCKED]  bus_devices_mutex check_for_bus_devices (check_for_bus_devices 69768)
[LOCKED]  event_loop_mutex eventLoop (eventLoop 69768)
[MAIN] checking for dead wmbus devices...
(serial) expecting devices to work
[LOCKING] serial_devices_mutex list_file_descriptiors_to_listen_to ( 0)
[UNLOCKING] bus_devices_mutex check_for_bus_devices (check_for_bus_devices 69768)
[UNLOCKED]  bus_devices_mutex check_for_bus_devices ( 0)
[LOCKED]  serial_devices_mutex list_file_descriptiors_to_listen_to (list_file_descriptiors_to_listen_to 69768)
[LOCKING] serial_devices_mutex lookup (list_file_descriptiors_to_listen_to 69768)
[UNLOCKING] serial_devices_mutex list_file_descriptiors_to_listen_to (list_file_descriptiors_to_listen_to 69768)
[UNLOCKED]  serial_devices_mutex list_file_descriptiors_to_listen_to ( 0)
[LOCKED]  serial_devices_mutex lookup (lookup 69768)
[SERIAL] select timeout 1 s
[UNLOCKING] serial_devices_mutex lookup (lookup 69768)
[UNLOCKED]  serial_devices_mutex lookup ( 0)
(lookup) with cmd "rtlwmbus:CMD(nc 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters stdin:rtlwmbus)"
[LOCKING] bus_devices_mutex open_bus_device ( 0)
[LOCKED]  bus_devices_mutex open_bus_device (open_bus_device 69768)
(main) opening rtlwmbus:CMD(nc 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters stdin:rtlwmbus)
Started config rtlwmbus listening on none using CMD(nc 192.168.1.2 1234 | rtl_wmbus -p s | wmbusmeters stdin:rtlwmbus)
[... 126 lines cut ...]
[LOCKING] serial_devices_mutex close_all_do_not_remove ( 0)
(serial) non working devices found, exiting.
[LOCKED]  serial_devices_mutex close_all_do_not_remove (close_all_do_not_remove 69768)
(serial) TIMER thread interrupted
(serial) event loop stopped!
[UNLOCKING] serial_devices_mutex close_all_do_not_remove (close_all_do_not_remove 69768)
[UNLOCKING] event_loop_mutex eventLoop (eventLoop 69768)
[UNLOCKED]  serial_devices_mutex close_all_do_not_remove ( 0)
[UNLOCKED]  event_loop_mutex eventLoop ( 0)
(wmbus) deleted rtlwmbus
[LOCKING] serial_devices_mutex tickle ( 0)
[LOCKED]  serial_devices_mutex tickle (tickle 69768)
[UNLOCKING] serial_devices_mutex tickle (tickle 69768)
[UNLOCKED]  serial_devices_mutex tickle ( 0)
(serialcmd) closed /bin/sh pid=69772 fd=-1 (rtlwmbus)
[LOCKING] serial_devices_mutex close_all_do_not_remove ( 0)
[LOCKED]  serial_devices_mutex close_all_do_not_remove (close_all_do_not_remove 69768)
[UNLOCKING] serial_devices_mutex close_all_do_not_remove (close_all_do_not_remove 69768)
[UNLOCKED]  serial_devices_mutex close_all_do_not_remove ( 0)
[LOCKING] serial_devices_mutex remove_non_working_serial_devices ( 0)
[LOCKED]  serial_devices_mutex remove_non_working_serial_devices (remove_non_working_serial_devices 69768)
(serial) no devices working emergency exit!
[UNLOCKING] serial_devices_mutex remove_non_working_serial_devices (remove_non_working_serial_devices 69768)
[UNLOCKED]  serial_devices_mutex remove_non_working_serial_devices ( 0)
```

chpego, 2023-08-25:
Hello @andriej 
Do you have some news ?